aaronwmorris/indi-allsky #1: File "./allsky.py", line 97, in <module>

ptson92, 2021-09-08:
Hi. 
I tried to install with QHY5 but got this error:
2021-09-08 18:02:40,568 [INFO] MainProcess _initialize() #255: Set BLOB mode
2021-09-08 18:02:40,570 [INFO] MainProcess newMessage() #129: new Message 2021-09-08T11:02:40: [ERROR] Connecting to camera failed (QHY5LII-M-6077e8602ac8994f3). 
2021-09-08 18:02:40,572 [INFO] MainProcess configureDevice() #234: Setting switch FRAME_FORMAT
2021-09-08 18:02:40,573 [INFO] MainProcess newSwitch() #116: new Switch CONNECTION for device QHY CCD QHY5LII-M-6077e
Traceback (most recent call last):
  File "./allsky.py", line 97, in <module>
    action_func(*args_list)
  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 393, in run
    self._initialize()
  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 258, in _initialize
    self.indiclient.configureDevice(self.ccdDevice, self.config['INDI_CONFIG_DEFAULTS'])
  File "/home/pi/indi-allsky/indi_allsky/indi.py", line 235, in configureDevice
    self.set_switch(device, k, on_switches=v['on'], off_switches=v.get('off', []))
  File "/home/pi/indi-allsky/indi_allsky/indi.py", line 403, in set_switch
    c = self.get_control(device, name, 'switch')
  File "/home/pi/indi-allsky/indi_allsky/indi.py", line 377, in get_control
    raise TimeOutException('Timeout finding control {0}'.format(name))
indi_allsky.exceptions.TimeOutException: Timeout finding control FRAME_FORMAT
2021-09-08 18:02:50,585 [INFO] MainProcess info() #54: process shutting down

aaronwmorris, 2021-09-10:
If you are willing to be a tester for me I can make sure QHY cameras work correctly.  Every vendor can use slightly different property names for things like gain.  I just do not have a QHY camera to test with.

I just need the output of **indi_getprop** with the camera connected and indiserver running.

aaronwmorris, 2021-09-10:
Also, can you send the contents of your config.json file?

aaronwmorris, 2021-09-23:
QHY support is merged.  Closing this issue.

closed 2021-09-23 by aaronwmorris