aaronwmorris/indi-allsky #4: Unknown distribution Raspbian 11 (armv6l)

samuelvoltarius, 2021-11-09:
if i run the setup.sh on my Pi i got

Unknown distribution Raspbian 11 (armv6l)

can you pls fix this ?

aaronwmorris, 2021-11-09:
I did some investigation and Raspbian 11 does not appear to be supported by Astroberry which is the repository I used to setup INDI.

I have updated my setup script to indicate this problem.  As soon as Astroberry support Raspbian 11, I will get the support enabled.

samuelvoltarius, 2021-11-09:
ok thx alot

samuelvoltarius, 2021-11-09:
I have a new problem. If i use Raspbian 10 i got a certificate verification failed at the Astroberry repo.

aaronwmorris, 2021-11-09:
Can you post the error here?

samuelvoltarius, 2021-11-09:
**** Installing packages... ****
Installing INDI via Astroberry repository
--2021-11-09 22:36:59--  https://www.astroberry.io/repo/key
Resolving www.astroberry.io (www.astroberry.io)... 87.98.235.184
Connecting to www.astroberry.io (www.astroberry.io)|87.98.235.184|:443... connec                                                                                                             ted.
ERROR: The certificate of ‘www.astroberry.io’ is not trusted.
ERROR: The certificate of ‘www.astroberry.io’ has expired.

after apt-get upgrade it is working but now i get 
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 92136E3D04A0FFB5
Reading package lists... Done
W: GPG error: https://www.astroberry.io/repo buster InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 92136E3D04A0FFB5
E: The repository 'https://www.astroberry.io/repo buster InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

aaronwmorris, 2021-11-09:
I wonder if you have an old `ca-certificates` package installed and it could not validate the certificate for www.astroberry.io.

What version of `ca-certificates` do you have installed?

`dpkg -l | grep ca-certificates`

samuelvoltarius, 2021-11-10:
apt-get upgrade fixed the CA-certificates problem.

But now i have a new one. 

if i try to run ./allsky.py -c config.json run

i got 

(indi-allsky) pi@raspberrypi:~/indi-allsky $ ./allsky.py -c config.json run
Traceback (most recent call last):
  File "./allsky.py", line 94, in <module>
    ia = indi_allsky.IndiAllSky(args.config)
  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 40, in __init__
    self.config = self._parseConfig(f_config_file.read())
  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 193, in _parseConfig
    c = json.loads(json_config)
  File "/usr/lib/python3.7/json/__init__.py", line 348, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.7/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.7/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 33 column 22 (char 841)
2021-11-10 12:01:53,837 [INFO] MainProcess info() #54: process shutting down

aaronwmorris, 2021-11-10:
I had an error in my template config file.  If you can pull down the latest code and recopy the template, you should be all set.

samuelvoltarius, 2021-11-10:
is running thx but i got e new problem (see the other topic pls)

aaronwmorris, 2021-11-11:
Closing this issue in favor of the other thread.

closed 2021-11-11 by aaronwmorris