aaronwmorris/indi-allsky #5: Touptek testing

samuelvoltarius, 2021-11-09:
Good evening,

I would like to test a Touptek cam but i dont get die config for it ?

aaronwmorris, 2021-11-09:
I would love to get Touptek supported added.  I just need to get the configuration parameters for the camera.  If you could hook up the camera and connect it.  My software will put the camera in a connected state or you can connect to it using Kstars/EKOS.

Then run `indi_getprop` from the command line and post the results here.  I can probably get the support setup in a few minutes.

samuelvoltarius, 2021-11-09:
stellarmate@stellarmate:~ $ indi_getprop
connect: Connection refused
stellarmate@stellarmate:~ $ indi_getprop
Toupcam GCMOS01200KMB.CONNECTION.CONNECT=On
Toupcam GCMOS01200KMB.CONNECTION.DISCONNECT=Off
Toupcam GCMOS01200KMB.DRIVER_INFO.DRIVER_NAME=Toupcam
Toupcam GCMOS01200KMB.DRIVER_INFO.DRIVER_EXEC=indi_toupcam_ccd
Toupcam GCMOS01200KMB.DRIVER_INFO.DRIVER_VERSION=0.5
Toupcam GCMOS01200KMB.DRIVER_INFO.DRIVER_INTERFACE=6
Toupcam GCMOS01200KMB.POLLING_PERIOD.PERIOD_MS=1000
Toupcam GCMOS01200KMB.DEBUG.ENABLE=Off
Toupcam GCMOS01200KMB.DEBUG.DISABLE=On
Toupcam GCMOS01200KMB.SIMULATION.ENABLE=Off
Toupcam GCMOS01200KMB.SIMULATION.DISABLE=On
Toupcam GCMOS01200KMB.CONFIG_PROCESS.CONFIG_LOAD=Off
Toupcam GCMOS01200KMB.CONFIG_PROCESS.CONFIG_SAVE=Off
Toupcam GCMOS01200KMB.CONFIG_PROCESS.CONFIG_DEFAULT=Off
Toupcam GCMOS01200KMB.CONFIG_PROCESS.CONFIG_PURGE=Off
Toupcam GCMOS01200KMB.ACTIVE_DEVICES.ACTIVE_TELESCOPE=EQMod Mount
Toupcam GCMOS01200KMB.ACTIVE_DEVICES.ACTIVE_ROTATOR=Rotator Simulator
Toupcam GCMOS01200KMB.ACTIVE_DEVICES.ACTIVE_FOCUSER=Focuser Simulator
Toupcam GCMOS01200KMB.ACTIVE_DEVICES.ACTIVE_FILTER=CCD Simulator
Toupcam GCMOS01200KMB.ACTIVE_DEVICES.ACTIVE_SKYQUALITY=SQM
Toupcam GCMOS01200KMB.CCD_EXPOSURE.CCD_EXPOSURE_VALUE=1
Toupcam GCMOS01200KMB.CCD_ABORT_EXPOSURE.ABORT=Off
Toupcam GCMOS01200KMB.CCD_FRAME.X=0
Toupcam GCMOS01200KMB.CCD_FRAME.Y=0
Toupcam GCMOS01200KMB.CCD_FRAME.WIDTH=1280
Toupcam GCMOS01200KMB.CCD_FRAME.HEIGHT=960
Toupcam GCMOS01200KMB.CCD_BINNING.HOR_BIN=2
Toupcam GCMOS01200KMB.CCD_BINNING.VER_BIN=2
Toupcam GCMOS01200KMB.FITS_HEADER.FITS_OBSERVER=Unknown
Toupcam GCMOS01200KMB.FITS_HEADER.FITS_OBJECT=Unknown
Toupcam GCMOS01200KMB.CCD_INFO.CCD_MAX_X=1280
Toupcam GCMOS01200KMB.CCD_INFO.CCD_MAX_Y=960
Toupcam GCMOS01200KMB.CCD_INFO.CCD_PIXEL_SIZE=3.75
Toupcam GCMOS01200KMB.CCD_INFO.CCD_PIXEL_SIZE_X=3.75
Toupcam GCMOS01200KMB.CCD_INFO.CCD_PIXEL_SIZE_Y=3.75
Toupcam GCMOS01200KMB.CCD_INFO.CCD_BITSPERPIXEL=16
Toupcam GCMOS01200KMB.CCD_COMPRESSION.CCD_COMPRESS=Off
Toupcam GCMOS01200KMB.CCD_COMPRESSION.CCD_RAW=On
Toupcam GCMOS01200KMB.TELESCOPE_TIMED_GUIDE_NS.TIMED_GUIDE_N=0
Toupcam GCMOS01200KMB.TELESCOPE_TIMED_GUIDE_NS.TIMED_GUIDE_S=0
Toupcam GCMOS01200KMB.TELESCOPE_TIMED_GUIDE_WE.TIMED_GUIDE_W=0
Toupcam GCMOS01200KMB.TELESCOPE_TIMED_GUIDE_WE.TIMED_GUIDE_E=0
Toupcam GCMOS01200KMB.CCD_FRAME_TYPE.FRAME_LIGHT=On
Toupcam GCMOS01200KMB.CCD_FRAME_TYPE.FRAME_BIAS=Off
Toupcam GCMOS01200KMB.CCD_FRAME_TYPE.FRAME_DARK=Off
Toupcam GCMOS01200KMB.CCD_FRAME_TYPE.FRAME_FLAT=Off
Toupcam GCMOS01200KMB.TELESCOPE_TYPE.TELESCOPE_PRIMARY=Off
Toupcam GCMOS01200KMB.TELESCOPE_TYPE.TELESCOPE_GUIDE=On
Toupcam GCMOS01200KMB.WCS_CONTROL.WCS_ENABLE=On
Toupcam GCMOS01200KMB.WCS_CONTROL.WCS_DISABLE=Off
Toupcam GCMOS01200KMB.UPLOAD_MODE.UPLOAD_CLIENT=On
Toupcam GCMOS01200KMB.UPLOAD_MODE.UPLOAD_LOCAL=Off
Toupcam GCMOS01200KMB.UPLOAD_MODE.UPLOAD_BOTH=Off
Toupcam GCMOS01200KMB.UPLOAD_SETTINGS.UPLOAD_DIR=/home/stellarmate
Toupcam GCMOS01200KMB.UPLOAD_SETTINGS.UPLOAD_PREFIX=IMAGE_XXX
Toupcam GCMOS01200KMB.CCD_VIDEO_STREAM.STREAM_ON=Off
Toupcam GCMOS01200KMB.CCD_VIDEO_STREAM.STREAM_OFF=On
Toupcam GCMOS01200KMB.STREAM_DELAY.STREAM_DELAY_TIME=0
Toupcam GCMOS01200KMB.STREAMING_EXPOSURE.STREAMING_EXPOSURE_VALUE=0.10000000000000000555
Toupcam GCMOS01200KMB.STREAMING_EXPOSURE.STREAMING_DIVISOR_VALUE=1
Toupcam GCMOS01200KMB.FPS.EST_FPS=30
Toupcam GCMOS01200KMB.FPS.AVG_FPS=30
Toupcam GCMOS01200KMB.RECORD_STREAM.RECORD_ON=Off
Toupcam GCMOS01200KMB.RECORD_STREAM.RECORD_DURATION_ON=Off
Toupcam GCMOS01200KMB.RECORD_STREAM.RECORD_FRAME_ON=Off
Toupcam GCMOS01200KMB.RECORD_STREAM.RECORD_OFF=On
Toupcam GCMOS01200KMB.RECORD_FILE.RECORD_FILE_DIR=/home/stellarmate/indi__D_
[... 56 lines cut ...]
Toupcam GCMOS01200KMB.CCD_VIDEO_STREAM.STREAM_OFF=On
Toupcam GCMOS01200KMB.STREAMING_EXPOSURE.STREAMING_EXPOSURE_VALUE=0.10000000000000000555
Toupcam GCMOS01200KMB.STREAMING_EXPOSURE.STREAMING_DIVISOR_VALUE=1
Toupcam GCMOS01200KMB.FPS.EST_FPS=30
Toupcam GCMOS01200KMB.FPS.AVG_FPS=30
Toupcam GCMOS01200KMB.RECORD_STREAM.RECORD_ON=Off
Toupcam GCMOS01200KMB.RECORD_STREAM.RECORD_DURATION_ON=Off
Toupcam GCMOS01200KMB.RECORD_STREAM.RECORD_FRAME_ON=Off
Toupcam GCMOS01200KMB.RECORD_STREAM.RECORD_OFF=On
Toupcam GCMOS01200KMB.RECORD_FILE.RECORD_FILE_DIR=/home/stellarmate/indi__D_
Toupcam GCMOS01200KMB.RECORD_FILE.RECORD_FILE_NAME=indi_record__T_
Toupcam GCMOS01200KMB.RECORD_OPTIONS.RECORD_DURATION=1
Toupcam GCMOS01200KMB.RECORD_OPTIONS.RECORD_FRAME_TOTAL=30
Toupcam GCMOS01200KMB.CCD_STREAM_FRAME.X=0
Toupcam GCMOS01200KMB.CCD_STREAM_FRAME.Y=0
Toupcam GCMOS01200KMB.CCD_STREAM_FRAME.WIDTH=1280
Toupcam GCMOS01200KMB.CCD_STREAM_FRAME.HEIGHT=960
Toupcam GCMOS01200KMB.CCD_STREAM_ENCODER.RAW=On
Toupcam GCMOS01200KMB.CCD_STREAM_ENCODER.MJPEG=Off
Toupcam GCMOS01200KMB.CCD_STREAM_RECORDER.SER=On
Toupcam GCMOS01200KMB.CCD_STREAM_RECORDER.OGV=Off
Toupcam GCMOS01200KMB.LIMITS.LIMITS_BUFFER_MAX=512
Toupcam GCMOS01200KMB.LIMITS.LIMITS_PREVIEW_FPS=10

Thx for your help

aaronwmorris, 2021-11-09:
I have added the necessary code to initialize Touptek cameras.  You should be able to use the standard template config file in the repository.

Let me know if that works for you!

samuelvoltarius, 2021-11-10:
./allsky.py -c config.json run
2021-11-10 21:18:26,016 [INFO] MainProcess info() #54: allocating a new mmap of length 4096
2021-11-10 21:18:26,272 [INFO] MainProcess __init__() #90: creating an instance of IndiClient
2021-11-10 21:18:26,275 [INFO] MainProcess _initialize() #243: Connecting to indiserver
INDI::BaseClient::connectServer: creating new connection...
2021-11-10 21:18:26,285 [INFO] MainProcess serverConnected() #178: Server connected (localhost:7624)
INDI::BaseClient::connectServer: Already connected.
2021-11-10 21:18:26,310 [INFO] MainProcess newDevice() #127: new device Telescope Simulator
2021-11-10 21:18:34,298 [INFO] MainProcess findCcds() #292: Found device Telescope Simulator
2021-11-10 21:18:34,302 [INFO] MainProcess findCcds() #297:  Detected telescope
2021-11-10 21:18:34,307 [INFO] MainProcess findCcds() #297:  Detected guider
2021-11-10 21:18:34,313 [ERROR] MainProcess _initialize() #256: No CCDs detected
2021-11-10 21:18:35,320 [INFO] MainProcess info() #54: process shutting down

aaronwmorris, 2021-11-10:
This can happen occasionally, especially when things are being changed a lot.

You may try rebooting or restarting the indiserver service with `sudo systemctl restart indiserver`

samuelvoltarius, 2021-11-10:
still the same after reboot and sudo systemctl restart indiserver

aaronwmorris, 2021-11-10:
When you ran the setup.sh script, did you select the `indi_toupcam_ccd` server?

samuelvoltarius, 2021-11-10:
yes

aaronwmorris, 2021-11-10:
Are you positive the camera is connected?  Does indi_getprop return data regarding the camera?

samuelvoltarius, 2021-11-10:
Telescope Simulator.CONNECTION.CONNECT=Off
Telescope Simulator.CONNECTION.DISCONNECT=On
Telescope Simulator.DRIVER_INFO.DRIVER_NAME=Telescope Simulator
Telescope Simulator.DRIVER_INFO.DRIVER_EXEC=indi_simulator_telescope
Telescope Simulator.DRIVER_INFO.DRIVER_VERSION=1.0
Telescope Simulator.DRIVER_INFO.DRIVER_INTERFACE=5
Telescope Simulator.POLLING_PERIOD.PERIOD_MS=250
Telescope Simulator.DEBUG.ENABLE=Off
Telescope Simulator.DEBUG.DISABLE=On
Telescope Simulator.CONFIG_PROCESS.CONFIG_LOAD=Off
Telescope Simulator.CONFIG_PROCESS.CONFIG_SAVE=Off
Telescope Simulator.CONFIG_PROCESS.CONFIG_DEFAULT=Off
Telescope Simulator.CONFIG_PROCESS.CONFIG_PURGE=Off
Telescope Simulator.CONNECTION_MODE.CONNECTION_SERIAL=On
Telescope Simulator.CONNECTION_MODE.CONNECTION_TCP=Off
Telescope Simulator.DEVICE_PORT.PORT=/dev/ttyUSB0
Telescope Simulator.DEVICE_BAUD_RATE.9600=On
Telescope Simulator.DEVICE_BAUD_RATE.19200=Off
Telescope Simulator.DEVICE_BAUD_RATE.38400=Off
Telescope Simulator.DEVICE_BAUD_RATE.57600=Off
Telescope Simulator.DEVICE_BAUD_RATE.115200=Off
Telescope Simulator.DEVICE_BAUD_RATE.230400=Off
Telescope Simulator.DEVICE_AUTO_SEARCH.INDI_ENABLED=Off
Telescope Simulator.DEVICE_AUTO_SEARCH.INDI_DISABLED=On
Telescope Simulator.DEVICE_PORT_SCAN.Scan Ports=Off
Telescope Simulator.ACTIVE_DEVICES.ACTIVE_GPS=GPS Simulator
Telescope Simulator.ACTIVE_DEVICES.ACTIVE_DOME=Dome Simulator
Telescope Simulator.DOME_POLICY.DOME_IGNORED=On
Telescope Simulator.DOME_POLICY.DOME_LOCKS=Off
Telescope Simulator.TELESCOPE_INFO.TELESCOPE_APERTURE=120
Telescope Simulator.TELESCOPE_INFO.TELESCOPE_FOCAL_LENGTH=900
Telescope Simulator.TELESCOPE_INFO.GUIDER_APERTURE=120
Telescope Simulator.TELESCOPE_INFO.GUIDER_FOCAL_LENGTH=900
Telescope Simulator.SCOPE_CONFIG_NAME.SCOPE_CONFIG_NAME=
Telescope Simulator.MOUNT_AXES.PRIMARY=0
Telescope Simulator.MOUNT_AXES.SECONDARY=0
Telescope Simulator.ACTIVE_DEVICES.ACTIVE_GPS=GPS Simulator
Telescope Simulator.ACTIVE_DEVICES.ACTIVE_DOME=Dome Simulator
Telescope Simulator.DOME_POLICY.DOME_IGNORED=On
Telescope Simulator.DOME_POLICY.DOME_LOCKS=Off
Telescope Simulator.TELESCOPE_INFO.TELESCOPE_APERTURE=120
Telescope Simulator.TELESCOPE_INFO.TELESCOPE_FOCAL_LENGTH=900
Telescope Simulator.TELESCOPE_INFO.GUIDER_APERTURE=120
Telescope Simulator.TELESCOPE_INFO.GUIDER_FOCAL_LENGTH=900
Telescope Simulator.SCOPE_CONFIG_NAME.SCOPE_CONFIG_NAME=
Telescope Simulator.MOUNT_AXES.PRIMARY=0
Telescope Simulator.MOUNT_AXES.SECONDARY=0

if think it is not finding it but i have it pluged in

aaronwmorris, 2021-11-10:
You may want to re-run the setup script and maybe try a different USB cable.  It definitely detected it previously.

samuelvoltarius, 2021-11-10:
i try a other linux pc and got this error now

sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) attempt to write a readonly database
[SQL: INSERT INTO camera (name) VALUES (?)]
[parameters: ('Toupcam GCMOS01200KMB',)]
(Background on this error at: https://sqlalche.me/e/14/e3q8)
2021-11-10 22:38:08,773 [INFO] MainProcess info() #54: process shutting down

aaronwmorris, 2021-11-10:
The camera needs to be connected before the indiserver starts.  The server is configured to auto-start at boot, so make sure the camera is connected at boot time.

Did you use a different user to run the setup script vs running the program?  The permission might not allow the user to update the DB.

You can try making the DB world writable.

`
chmod 777 /var/lib/indi-allsky
chmod 666 /var/lib/indi-allsky/indi-allsky.sqlite
`

samuelvoltarius, 2021-11-10:
chmod 777 /var/lib/indi-allsky
chmod 666 /var/lib/indi-allsky/indi-allsky.sqlite

worked but the next thing.

./allsky.py -c config.json run
2021-11-10 22:59:06,751 [INFO] MainProcess info() #54: allocating a new mmap of length 4096
2021-11-10 22:59:06,757 [INFO] MainProcess __init__() #90: creating an instance of IndiClient
2021-11-10 22:59:06,757 [INFO] MainProcess _initialize() #243: Connecting to indiserver
INDI::BaseClient::connectServer: creating new connection...
2021-11-10 22:59:06,757 [INFO] MainProcess serverConnected() #178: Server connected (localhost:7624)
INDI::BaseClient::connectServer: Already connected.
2021-11-10 22:59:06,759 [INFO] MainProcess newDevice() #127: new device Telescope Simulator
2021-11-10 22:59:06,759 [INFO] MainProcess newDevice() #127: new device Toupcam GCMOS01200KMB
2021-11-10 22:59:14,766 [INFO] MainProcess findCcds() #292: Found device Telescope Simulator
2021-11-10 22:59:14,766 [INFO] MainProcess findCcds() #297:  Detected telescope
2021-11-10 22:59:14,766 [INFO] MainProcess findCcds() #297:  Detected guider
2021-11-10 22:59:14,766 [INFO] MainProcess findCcds() #292: Found device Toupcam GCMOS01200KMB
2021-11-10 22:59:14,766 [INFO] MainProcess findCcds() #297:  Detected ccd
2021-11-10 22:59:14,766 [INFO] MainProcess findCcds() #297:  Detected guider
2021-11-10 22:59:14,766 [INFO] MainProcess _initialize() #260: Found 1 CCDs
2021-11-10 22:59:14,766 [WARNING] MainProcess _initialize() #263: Connecting to device Toupcam GCMOS01200KMB
2021-11-10 22:59:14,895 [INFO] MainProcess addCamera() #57: Camera DB ID: 1
2021-11-10 22:59:14,895 [INFO] MainProcess _initialize() #277: Set BLOB mode
2021-11-10 22:59:15,461 [INFO] MainProcess newText() #169: new Text DRIVER_INFO for device Toupcam GCMOS01200KMB
2021-11-10 22:59:15,463 [INFO] MainProcess newMessage() #175: new Message 2021-11-10T21:59:15: [INFO] Toupcam GCMOS01200KMB is online. Retrieving basic data. 
2021-11-10 22:59:15,463 [INFO] MainProcess newSwitch() #162: new Switch CONNECTION for device Toupcam GCMOS01200KMB
2021-11-10 22:59:15,897 [INFO] MainProcess configureDevice() #312: Setting switch CCD_FRAME_TYPE
2021-11-10 22:59:16,366 [INFO] MainProcess newSwitch() #162: new Switch CCD_FRAME_TYPE for device Toupcam GCMOS01200KMB
2021-11-10 22:59:16,953 [INFO] MainProcess getCcdInfo() #237: Detecting bayer pattern
2021-11-10 22:59:21,962 [WARNING] MainProcess getCcdInfo() #246: CCD_CFA fetch timeout, assuming monochrome camera
2021-11-10 22:59:21,984 [INFO] MainProcess _initialize() #292: Minimum CCD exposure: 0.00018000
2021-11-10 22:59:21,984 [INFO] MainProcess _initialize() #298: Default CCD exposure: 0.00010000
2021-11-10 22:59:21,984 [INFO] MainProcess _initialize() #305: CCD CFA: None
2021-11-10 22:59:21,984 [INFO] MainProcess _startImageWorker() #315: Starting ImageWorker process
2021-11-10 22:59:22,075 [INFO] MainProcess _startVideoWorker() #355: Starting VideoWorker process
2021-11-10 22:59:22,080 [INFO] ImageWorker001 info() #54: child process calling self.run()
2021-11-10 22:59:22,083 [INFO] MainProcess _startFileUploadWorker() #388: Starting FileUploader process 1
2021-11-10 22:59:22,085 [INFO] VideoWorker001 info() #54: child process calling self.run()
2021-11-10 22:59:22,089 [INFO] FileUploader001 info() #54: child process calling self.run()
2021-11-10 22:59:22,111 [INFO] MainProcess detectNight() #569: Sun altitude: 6:36:24.5
2021-11-10 22:59:22,112 [INFO] MainProcess detectMoonMode() #587: Moon altitide: 31:08:27.9, phase 43.4%
2021-11-10 22:59:22,112 [WARNING] MainProcess reconfigureCcd() #536: Change to day
2021-11-10 22:59:22,112 [WARNING] MainProcess setCcdGain() #373: Setting CCD gain to 0
2021-11-10 22:59:22,112 [INFO] MainProcess configureDevice() #307: Setting property CCD_CONTROLS
2021-11-10 22:59:22,123 [INFO] MainProcess newMessage() #175: new Message 2021-11-10T21:59:22: Error: Invalid range for Gain (Gain). Valid range is from 100 to 1000. Requested value is 0 
Traceback (most recent call last):
  File "./allsky.py", line 99, in <module>
    action_func(*args_list)
  File "/home/brutus/indi-allsky/indi_allsky/allsky.py", line 456, in run
    self.reconfigureCcd()
  File "/home/brutus/indi-allsky/indi_allsky/allsky.py", line 537, in reconfigureCcd
    self.indiclient.setCcdGain(self.ccdDevice, self.config['CCD_CONFIG']['DAY']['GAIN'])
  File "/home/brutus/indi-allsky/indi_allsky/indi.py", line 418, in setCcdGain
    self.configureDevice(ccdDevice, gain_config)
  File "/home/brutus/indi-allsky/indi_allsky/indi.py", line 308, in configureDevice
    self.set_number(device, k, v)
  File "/home/brutus/indi-allsky/indi_allsky/indi.py", line 508, in set_number
    self.__wait_for_ctl_statuses(c, timeout=timeout)
  File "/home/brutus/indi-allsky/indi_allsky/indi.py", line 576, in __wait_for_ctl_statuses
    raise RuntimeError('Error while changing property {0}'.format(ctl.getName()))
RuntimeError: Error while changing property CCD_CONTROLS
2021-11-10 22:59:22,172 [INFO] MainProcess info() #54: process shutting down
2021-11-10 22:59:22,172 [INFO] MainProcess info() #54: calling join() for process VideoWorker001
^C2021-11-10 22:59:41,549 [WARNING] VideoWorker001 sigint_handler() #179: Caught INT signal, shutting down
2021-11-10 22:59:41,549 [INFO] VideoWorker001 _stopVideoWorker() #376: Stopping VideoWorker process
2021-11-10 22:59:41,549 [WARNING] MainProcess sigint_handler() #179: Caught INT signal, shutting down
[... cut ...]
  File "/home/brutus/indi-allsky/indi_allsky/allsky.py", line 369, in _stopVideoWorker
    if not self.video_worker.is_alive():
  File "/usr/lib/python3.8/multiprocessing/process.py", line 160, in is_alive
    assert self._parent_pid == os.getpid(), 'can only test a child process'
2021-11-10 22:59:41,579 [INFO] ImageWorker001 info() #54: process exiting with exitcode 1
AssertionError: can only test a child process
2021-11-10 22:59:41,581 [INFO] FileUploader001 info() #54: process exiting with exitcode 1

and than it stops

aaronwmorris, 2021-11-10:
In the config.json file, increase the DAY and MOONMODE gains to above 100.  I would set DAY to 100.  I am not sure what good values would be for MOONMODE and NIGHT.  I would start at 200 and see how it goes.

aaronwmorris, 2021-11-13:
How did those settings work for you?  Were you able to get it working?

samuelvoltarius, 2021-11-27:
sorry i got a new pi the old was dmg.

now i geht this error 

2021-11-27 14:43:52,198 [INFO] MainProcess newMessage() #175: new Message 2021-11-27T13:43:52: [ERROR] Requested exposure value (0.0001) seconds out of bounds [0.00018,800].
Traceback (most recent call last):
  File "./allsky.py", line 99, in <module>
    action_func(*args_list)
  File "/home/pi/Desktop/indi-allsky/indi_allsky/allsky.py", line 461, in run
    self.shoot(self.exposure_v.value)
  File "/home/pi/Desktop/indi-allsky/indi_allsky/allsky.py", line 814, in shoot
    self.indiclient.setCcdExposure(self.ccdDevice, exposure, sync=sync, timeout=timeout)
  File "/home/pi/Desktop/indi-allsky/indi_allsky/indi.py", line 339, in setCcdExposure
    self.set_number(ccdDevice, 'CCD_EXPOSURE', {'CCD_EXPOSURE_VALUE': exposure}, sync=sync, timeout=timeout)
  File "/home/pi/Desktop/indi-allsky/indi_allsky/indi.py", line 513, in set_number
    self.__wait_for_ctl_statuses(c, timeout=timeout)
  File "/home/pi/Desktop/indi-allsky/indi_allsky/indi.py", line 581, in __wait_for_ctl_statuses
    raise RuntimeError('Error while changing property {0}'.format(ctl.getName()))
RuntimeError: Error while changing property CCD_EXPOSURE
2021-11-27 14:43:52,316 [INFO] MainProcess info() #54: process shutting down
2021-11-27 14:43:52,317 [INFO] MainProcess info() #54: calling join() for process ImageWorker001

samuelvoltarius, 2021-11-27:
Looks like

"CCD_EXPOSURE_MIN" : 0.0002,
    "CCD_EXPOSURE_DEF" : 0.0002

fixed it also

aaronwmorris, 2021-11-27:
I have made some updates to the minimum exposure auto-detection routine.  Hopefully, manual intervention for the minimum and default exposures will no longer be necessary.

aaronwmorris, 2022-02-01:
Closing this issue for now.

closed 2022-02-01 by aaronwmorris