aaronwmorris/indi-allsky #6: Install failed on Astroberry image

FlareP1, 2021-11-17:
I have tried to follow the instructions to install indi-allsky onto my astroberry image as it has my indo dev environment and other tools that I may want to use.  However there appers to be an conflict with Apache can you assist?  I assume this is becuase astroberry already has a web service.  I tried editing apache_indi-allsky.conf to move it to port 81 but this did not help?

Thanks

Looks a great project, I have an Altair GPCAM290 that works with indi so looking forwards to setting this up as an Allsky camera

````
Select an INDI driver: 1
**** Setting up indiserver service ****
**** Setting up indi-allsky service ****
**** Enabling services ****
**** Setup rsyslog logging ****
**** Start apache2 service ****
Module rewrite already enabled
Considering dependency setenvif for ssl:
Module setenvif already enabled
Considering dependency mime for ssl:
Module mime already enabled
Considering dependency socache_shmcb for ssl:
Module socache_shmcb already enabled
Module ssl already enabled
Site 000-default already disabled
Site default-ssl already disabled
Site indi-allsky already enabled
Synchronizing state of apache2.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable apache2
Job for apache2.service failed because the control process exited with error code.
See "systemctl status apache2.service" and "journalctl -xe" for details.
astroberry@astroberry:~/indi-allsky $ ^C
astroberry@astroberry:~/indi-allsky $ systemctl status apache2.service
● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2021-11-18 00:48:40 CET; 10s ago
     Docs: https://httpd.apache.org/docs/2.4/
  Process: 14680 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)

Nov 18 00:48:40 astroberry apachectl[14680]: AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 127.0
Nov 18 00:48:40 astroberry apachectl[14680]: (98)Address already in use: AH00072: make_sock: could not bind to address [::]:80
Nov 18 00:48:40 astroberry apachectl[14680]: (98)Address already in use: AH00072: make_sock: could not bind to address 0.0.0.0:80
Nov 18 00:48:40 astroberry apachectl[14680]: no listening sockets available, shutting down
Nov 18 00:48:40 astroberry apachectl[14680]: AH00015: Unable to open logs
Nov 18 00:48:40 astroberry apachectl[14680]: Action 'start' failed.
Nov 18 00:48:40 astroberry apachectl[14680]: The Apache error log may have more information.
Nov 18 00:48:40 astroberry systemd[1]: apache2.service: Control process exited, code=exited, status=1/FAILURE
Nov 18 00:48:40 astroberry systemd[1]: apache2.service: Failed with result 'exit-code'.
Nov 18 00:48:40 astroberry systemd[1]: Failed to start The Apache HTTP Server.
```

FlareP1, 2021-11-18:
I will answer my own problem.  It appears that astroberry uses nginx which causes a conflict with apache,  I disabled this as below and then ran ./setup.sh again as it does not complete if you get the apache error.

`sudo systemctl stop nginx`
`./setup.sh`

aaronwmorris, 2021-11-18:
Do you get the same error after stopping nginx?

aaronwmorris, 2021-11-18:
Did everything complete as normal once you disabled nginx?

I can look into creating a setup script that uses nginx instead of Apache.

FlareP1, 2021-11-18:
Yes the installation then completed correctly but there is still a conflict with the indi-webserver and indi-server so I also disabled that and then re-ran the setup to allow the correct indi-driver to be selected.  See below.  I am happy for you to close this but would appreciate some help with the Altiar config setup, will post a new issue for this

```
sudo systemctl stop nginx
sudo systemctl stop indiwebmanager
./setup.sh
```

aaronwmorris, 2021-11-20:
Closing this in favor of the other thread.

closed 2021-11-20 by aaronwmorris