aaronwmorris/indi-allsky #7: Altair GPCAM2 290M config

FlareP1, 2021-11-18:
Hi Arron, I would like some help setting up an Altair GPCAM2 290M to work with your allsky.  The camera works fine within Indi.  I believe it is a sub class of the toupek cameras so I tried to start with this.  See the output below, can you suggest how I go about getting the correct config.

These two header files from indi may help you

```
https://github.com/indilib/indi-3rdparty/blob/master/libaltaircam/altaircam.h
https://github.com/indilib/indi-3rdparty/blob/master/indi-toupbase/indi_toupbase.h
```

When I run manually with the altair_ccd driver selcted I get the following output

```
(indi-allsky) astroberry@astroberry:~/indi-allsky $ cp config.json_touptek config.json
(indi-allsky) astroberry@astroberry:~/indi-allsky $ ./allsky.py -c config.json run
2021-11-18 21:18:19,493 [INFO] MainProcess info() #54: allocating a new mmap of length 4096
2021-11-18 21:18:19,527 [INFO] MainProcess __init__() #90: creating an instance of IndiClient
2021-11-18 21:18:19,528 [INFO] MainProcess _initialize() #243: Connecting to indiserver
INDI::BaseClient::connectServer: creating new connection...
INDI::BaseClient::connectServer: Already connected.
2021-11-18 21:18:19,532 [INFO] MainProcess serverConnected() #178: Server connected (localhost:7624)
2021-11-18 21:18:19,533 [INFO] MainProcess newDevice() #127: new device Altair ALTAIRGP290M
2021-11-18 21:18:19,536 [INFO] MainProcess newDevice() #127: new device Telescope Simulator
2021-11-18 21:18:19,585 [INFO] MainProcess newText() #169: new Text ACTIVE_DEVICES for device Telescope Simulator
2021-11-18 21:18:19,586 [INFO] MainProcess newText() #169: new Text SCOPE_CONFIG_NAME for device Telescope Simulator
2021-11-18 21:18:19,586 [INFO] MainProcess newSwitch() #162: new Switch MOUNT_TYPE for device Telescope Simulator
2021-11-18 21:18:19,588 [INFO] MainProcess newSwitch() #162: new Switch SIM_PIER_SIDE for device Telescope Simulator
2021-11-18 21:18:27,541 [INFO] MainProcess findCcds() #292: Found device Altair ALTAIRGP290M
2021-11-18 21:18:27,542 [INFO] MainProcess findCcds() #297:  Detected ccd
2021-11-18 21:18:27,542 [INFO] MainProcess findCcds() #297:  Detected guider
2021-11-18 21:18:27,542 [INFO] MainProcess findCcds() #292: Found device Telescope Simulator
2021-11-18 21:18:27,543 [INFO] MainProcess findCcds() #297:  Detected telescope
2021-11-18 21:18:27,543 [INFO] MainProcess findCcds() #297:  Detected guider
2021-11-18 21:18:27,545 [INFO] MainProcess _initialize() #260: Found 1 CCDs
2021-11-18 21:18:27,546 [WARNING] MainProcess _initialize() #263: Connecting to device Altair ALTAIRGP290M
2021-11-18 21:18:27,569 [INFO] MainProcess newText() #169: new Text DRIVER_INFO for device Altair ALTAIRGP290M
2021-11-18 21:18:27,580 [INFO] MainProcess newMessage() #175: new Message 2021-11-18T20:18:27: [INFO] Altair ALTAIRGP290M is online. Retrieving basic data.
2021-11-18 21:18:27,591 [INFO] MainProcess newSwitch() #162: new Switch CONNECTION for device Altair ALTAIRGP290M
2021-11-18 21:18:27,626 [INFO] MainProcess addCamera() #57: Camera DB ID: 2
2021-11-18 21:18:27,626 [INFO] MainProcess _initialize() #277: Set BLOB mode
2021-11-18 21:18:27,762 [INFO] MainProcess newMessage() #175: new Message 2021-11-18T20:18:27: [INFO] Maximum gain considering dual gain is 22500.
Traceback (most recent call last):
  File "/home/astroberry/indi-allsky/virtualenv/indi-allsky/lib/python3.7/site-packages/PyIndi.py", line 1081, in <lambda>
    __getattr__ = lambda self, name: _swig_getattr(self, BaseClient, name)
  File "/home/astroberry/indi-allsky/virtualenv/indi-allsky/lib/python3.7/site-packages/PyIndi.py", line 80, in _swig_getattr
    raise AttributeError("'%s' object has no attribute '%s'" % (class_type.__name__, name))
AttributeError: 'BaseClient' object has no attribute 'removeDevice'
2021-11-18 21:18:27,778 [INFO] MainProcess info() #54: process shutting down
```

aaronwmorris, 2021-11-18:
That is an odd exception that I have never seen before.  What server are you using for your camera?  Are you using indi_toupcam_ccd or indi_altair_ccd ?

FlareP1, 2021-11-18:
Am using indi_Altair_ccd.

aaronwmorris, 2021-11-18:
I have posted a change that should properly set the gain and bin mode of cameras using the ccd_altair_ccd server.  Let me know if that helps.

FlareP1, 2021-11-18:
Thanks will give it a go in the next hour

Chris

FlareP1, 2021-11-18:
Tried your update there is a slight difference, this time the driver reported EOF.  I did run it previously and this did not report the EOF and the output was identical to my first post.  New output listed below.  
If I am interpreting the error correctly it is expecting a 'removeDevice' method in the BaseClass that it thinks is missing.  Maybe I should try a clean install from a standard OS ?  I also tried to use the indi_simulator_ccd and the appropriate json and this did work, so sounds like the base installation is good?

```
(indi-allsky) astroberry@astroberry:~/indi-allsky $ indiserver indi_altair_ccd &
[1] 5024
(indi-allsky) astroberry@astroberry:~/indi-allsky $ 2021-11-18T23:33:11: startup: indiserver indi_altair_ccd

(indi-allsky) astroberry@astroberry:~/indi-allsky $
(indi-allsky) astroberry@astroberry:~/indi-allsky $ cp ./examples/config.json_touptek ./config.json
(indi-allsky) astroberry@astroberry:~/indi-allsky $ ./allsky.py -c config.json run
2021-11-19 00:33:43,675 [INFO] MainProcess info() #54: allocating a new mmap of length 4096
2021-11-19 00:33:43,707 [INFO] MainProcess __init__() #90: creating an instance of IndiClient
2021-11-19 00:33:43,708 [INFO] MainProcess _initialize() #243: Connecting to indiserver
INDI::BaseClient::connectServer: creating new connection...
INDI::BaseClient::connectServer: Already connected.
2021-11-19 00:33:43,709 [INFO] MainProcess serverConnected() #178: Server connected (localhost:7624)
2021-11-19 00:33:43,710 [INFO] MainProcess newDevice() #127: new device Altair ALTAIRGP290M
2021-11-19 00:33:51,718 [INFO] MainProcess findCcds() #297: Found device Altair ALTAIRGP290M
2021-11-19 00:33:51,718 [INFO] MainProcess findCcds() #302:  Detected ccd
2021-11-19 00:33:51,719 [INFO] MainProcess findCcds() #302:  Detected guider
2021-11-19 00:33:51,719 [INFO] MainProcess _initialize() #260: Found 1 CCDs
2021-11-19 00:33:51,720 [WARNING] MainProcess _initialize() #263: Connecting to device Altair ALTAIRGP290M
2021-11-19 00:33:51,738 [INFO] MainProcess newText() #169: new Text DRIVER_INFO for device Altair ALTAIRGP290M
2021-11-19 00:33:51,749 [INFO] MainProcess newMessage() #175: new Message 2021-11-18T23:33:51: [INFO] Altair ALTAIRGP290M is online. Retrieving basic data.
2021-11-19 00:33:51,760 [INFO] MainProcess newSwitch() #162: new Switch CONNECTION for device Altair ALTAIRGP290M
2021-11-19 00:33:51,802 [INFO] MainProcess addCamera() #57: Camera DB ID: 2
2021-11-19 00:33:51,802 [INFO] MainProcess updateCcdBlobMode() #185: Set BLOB mode
2021-11-19 00:33:51,938 [INFO] MainProcess newMessage() #175: new Message 2021-11-18T23:33:51: [INFO] Maximum gain considering dual gain is 22500.
2021-11-18T23:33:51: Driver indi_altair_ccd: stderr EOF
<delProperty device="Altair ALTAIRGP290M"/>
2021-11-18T23:33:51: Driver indi_altair_ccd: restart #1
Child process 5025 died
Traceback (most recent call last):
  File "/home/astroberry/indi-allsky/virtualenv/indi-allsky/lib/python3.7/site-packages/PyIndi.py", line 1081, in <lambda>
    __getattr__ = lambda self, name: _swig_getattr(self, BaseClient, name)
  File "/home/astroberry/indi-allsky/virtualenv/indi-allsky/lib/python3.7/site-packages/PyIndi.py", line 80, in _swig_getattr
    raise AttributeError("'%s' object has no attribute '%s'" % (class_type.__name__, name))
AttributeError: 'BaseClient' object has no attribute 'removeDevice'
2021-11-19 00:33:51,960 [INFO] MainProcess info() #54: process shutting down
2021-11-18T23:33:52: Client 5: read: Connection reset by peer

```

aaronwmorris, 2021-11-18:
It looks like the indi server for the camera is crashing or disconnecting when setting the Blob Mode.

Can you try manually editing one of the files to see if it changes the behavior?

Change the prop from None to 'CCD1' in indi_allsky/indi.py on line 184.

`
def updateCcdBlobMode(self, ccd_device, blobmode=PyIndi.B_ALSO, prop=None):
        logger.info('Set BLOB mode')
        self.setBLOBMode(blobmode, ccd_device.getDeviceName(), prop)
`

to

`
def updateCcdBlobMode(self, ccd_device, blobmode=PyIndi.B_ALSO, prop='CCD1'):
        logger.info('Set BLOB mode')
        self.setBLOBMode(blobmode, ccd_device.getDeviceName(), prop)
`

FlareP1, 2021-11-19:
Sorry missed your comment, I was doing some debugging, will try your suggestions tomorrow it is late here.  

I had though that it was crashing on the next line when settings the default properties which are blank in the json.   But have not got my head around the message handlers for indi yet, so possibly it is a call back that happens asyncronously.  I added a coupe of lines of printing to find where it was in the code, see below.

```
        print('#CCDBLOBMODE ........')
        # set BLOB mode to BLOB_ALSO
        self.indiclient.updateCcdBlobMode(self.ccdDevice)

        print('#CCDCONFIGUREDEVICE ........')
        self.indiclient.configureDevice(self.ccdDevice, self.config['INDI_CONFIG_DEFAULTS'])
```

Output
```
#CCD_NAME ........
2021-11-19 01:13:08,718 [INFO] MainProcess newText() #169: new Text DRIVER_INFO for device Altair ALTAIRGP290M
2021-11-19 01:13:08,730 [INFO] MainProcess newMessage() #175: new Message 2021-11-19T00:13:08: [INFO] Altair ALTAIRGP290M is online. Retrieving basic data.
2021-11-19 01:13:08,741 [INFO] MainProcess newSwitch() #162: new Switch CONNECTION for device Altair ALTAIRGP290M
2021-11-19 01:13:08,786 [INFO] MainProcess addCamera() #57: Camera DB ID: 2
#CCDBLOBMODE ........
2021-11-19 01:13:08,786 [INFO] MainProcess updateCcdBlobMode() #185: Set BLOB mode
#CCDCONFIGUREDEVICE ........
2021-11-19 01:13:08,921 [INFO] MainProcess newMessage() #175: new Message 2021-11-19T00:13:08: [INFO] Maximum gain considering dual gain is 22500.
Child process 5453 died
2021-11-19T00:13:08: Driver indi_altair_ccd: stderr EOF
<delProperty device="Altair ALTAIRGP290M"/>
2021-11-19T00:13:08: Driver indi_altair_ccd: restart #7
Traceback (most recent call last):
  File "/home/astroberry/indi-allsky/virtualenv/indi-allsky/lib/python3.7/site-packages/PyIndi.py", line 1081, in <lambda>
    __getattr__ = lambda self, name: _swig_getattr(self, BaseClient, name)
  File "/home/astroberry/indi-allsky/virtualenv/indi-allsky/lib/python3.7/site-packages/PyIndi.py", line 80, in _swig_getattr
    raise AttributeError("'%s' object has no attribute '%s'" % (class_type.__name__, name))
AttributeError: 'BaseClient' object has no attribute 'removeDevice'
```

aaronwmorris, 2021-11-19:
What version of INDI is installed?

`dpkg -l | grep indi-bin`

FlareP1, 2021-11-19:
1.9.2-1 I believe that this was updated by the ./setup script

```
dpkg -l | grep indi-bin
ii  indi-bin                              1.9.2-1                              armhf        INDI server, drivers and tools
```
also
```
uname -ra
Linux astroberry 5.10.20-v7l+ #1404 SMP Thu Mar 4 19:44:07 GMT 2021 armv7l GNU/Linux
```

FlareP1, 2021-11-19:
No change by updating the blob mode to 'CCD1' see output below.  Before you made the latest change I was not getting the EOF error on the driver, see the first post.

```
2021-11-19 17:18:29,036 [INFO] MainProcess __init__() #90: creating an instance of IndiClient
2021-11-19 17:18:29,037 [INFO] MainProcess _initialize() #258: Connecting to indiserver
INDI::BaseClient::connectServer: creating new connection...
INDI::BaseClient::connectServer: Already connected.
2021-11-19 17:18:29,039 [INFO] MainProcess serverConnected() #178: Server connected (localhost:7624)
2021-11-19 17:18:29,042 [INFO] MainProcess newDevice() #127: new device Altair ALTAIRGP290M
2021-11-19 17:18:29,044 [INFO] MainProcess newDevice() #127: new device Telescope Simulator
2021-11-19 17:18:29,092 [INFO] MainProcess newText() #169: new Text ACTIVE_DEVICES for device Telescope Simulator
2021-11-19 17:18:29,093 [INFO] MainProcess newText() #169: new Text SCOPE_CONFIG_NAME for device Telescope Simulator
2021-11-19 17:18:29,093 [INFO] MainProcess newSwitch() #162: new Switch MOUNT_TYPE for device Telescope Simulator
2021-11-19 17:18:29,093 [INFO] MainProcess newSwitch() #162: new Switch SIM_PIER_SIDE for device Telescope Simulator
#CCDS ........
2021-11-19 17:18:37,047 [INFO] MainProcess findCcds() #298: Found device Altair ALTAIRGP290M
2021-11-19 17:18:37,048 [INFO] MainProcess findCcds() #303:  Detected ccd
2021-11-19 17:18:37,048 [INFO] MainProcess findCcds() #303:  Detected guider
2021-11-19 17:18:37,049 [INFO] MainProcess findCcds() #298: Found device Telescope Simulator
2021-11-19 17:18:37,049 [INFO] MainProcess findCcds() #303:  Detected telescope
2021-11-19 17:18:37,049 [INFO] MainProcess findCcds() #303:  Detected guider
2021-11-19 17:18:37,049 [INFO] MainProcess _initialize() #276: Found 1 CCDs
2021-11-19 17:18:37,050 [WARNING] MainProcess _initialize() #279: Connecting to device Altair ALTAIRGP290M
#CCD_NAME ........
2021-11-19 17:18:37,068 [INFO] MainProcess newText() #169: new Text DRIVER_INFO for device Altair ALTAIRGP290M
2021-11-19 17:18:37,083 [INFO] MainProcess newMessage() #175: new Message 2021-11-19T16:18:37: [INFO] Altair ALTAIRGP290M is online. Retrieving basic data.
2021-11-19 17:18:37,096 [INFO] MainProcess newSwitch() #162: new Switch CONNECTION for device Altair ALTAIRGP290M
2021-11-19 17:18:37,126 [INFO] MainProcess addCamera() #57: Camera DB ID: 2
#CCDBLOBMODE ........
2021-11-19 17:18:37,126 [INFO] MainProcess updateCcdBlobMode() #186: Set BLOB mode
#CCDCONFIGUREDEVICE ........
2021-11-19 17:18:37,281 [INFO] MainProcess newMessage() #175: new Message 2021-11-19T16:18:37: [INFO] Maximum gain considering dual gain is 22500.
2021-11-19T16:18:37: Driver indi_altair_ccd: stderr EOF
<delProperty device="Altair ALTAIRGP290M"/>
2021-11-19T16:18:37: Driver indi_altair_ccd: restart #1
Child process 1756 died
Traceback (most recent call last):
  File "/home/astroberry/indi-allsky/virtualenv/indi-allsky/lib/python3.7/site-packages/PyIndi.py", line 1081, in <lambda>
    __getattr__ = lambda self, name: _swig_getattr(self, BaseClient, name)
  File "/home/astroberry/indi-allsky/virtualenv/indi-allsky/lib/python3.7/site-packages/PyIndi.py", line 80, in _swig_getattr
    raise AttributeError("'%s' object has no attribute '%s'" % (class_type.__name__, name))
AttributeError: 'BaseClient' object has no attribute 'removeDevice'
2021-11-19 17:18:37,299 [INFO] MainProcess info() #54: process shutting down
2021-11-19T16:18:37: Client 0: read: Connection reset by peer
2021-11-19T16:18:37: Client 8: read: Connection reset by peer
```

FlareP1, 2021-11-19:
Progress, I think the crashing was due to the indi-3rdy party libraries being out of date with the indi installation that was updated with your setup script.  I did a full update

```
	sudo apt update
	sudo upgrade
```

Now I can progress pass that point.  Looks like is has good communication with the camera and can access its properties etc...  I will update the settings for the exposure limitation

```
2021-11-19 18:45:43,298 [INFO] MainProcess __init__() #90: creating an instance of IndiClient
2021-11-19 18:45:43,299 [INFO] MainProcess _initialize() #258: Connecting to indiserver
INDI::BaseClient::connectServer: creating new connection...
INDI::BaseClient::connectServer: Already connected.
2021-11-19 18:45:43,305 [INFO] MainProcess newDevice() #127: new device Altair ALTAIRGP290M
2021-11-19 18:45:43,306 [INFO] MainProcess newDevice() #127: new device Telescope Simulator
2021-11-19 18:45:43,309 [INFO] MainProcess serverConnected() #178: Server connected (localhost:7624)
2021-11-19 18:45:43,310 [INFO] MainProcess newText() #169: new Text ACTIVE_DEVICES for device Telescope Simulator
2021-11-19 18:45:43,310 [INFO] MainProcess newText() #169: new Text SCOPE_CONFIG_NAME for device Telescope Simulator
2021-11-19 18:45:43,311 [INFO] MainProcess newSwitch() #162: new Switch MOUNT_TYPE for device Telescope Simulator
2021-11-19 18:45:43,311 [INFO] MainProcess newSwitch() #162: new Switch SIM_PIER_SIDE for device Telescope Simulator
#CCDS ........
2021-11-19 18:45:51,312 [INFO] MainProcess findCcds() #298: Found device Altair ALTAIRGP290M
2021-11-19 18:45:51,312 [INFO] MainProcess findCcds() #303:  Detected ccd
2021-11-19 18:45:51,312 [INFO] MainProcess findCcds() #303:  Detected guider
2021-11-19 18:45:51,312 [INFO] MainProcess findCcds() #298: Found device Telescope Simulator
2021-11-19 18:45:51,313 [INFO] MainProcess findCcds() #303:  Detected telescope
2021-11-19 18:45:51,313 [INFO] MainProcess findCcds() #303:  Detected guider
2021-11-19 18:45:51,313 [INFO] MainProcess _initialize() #276: Found 1 CCDs
2021-11-19 18:45:51,313 [WARNING] MainProcess _initialize() #279: Connecting to device Altair ALTAIRGP290M
#CCD_NAME ........
2021-11-19 18:45:51,360 [INFO] MainProcess addCamera() #57: Camera DB ID: 2
#CCDBLOBMODE ........
2021-11-19 18:45:51,360 [INFO] MainProcess updateCcdBlobMode() #186: Set BLOB mode
2021-11-19 18:45:51,505 [INFO] MainProcess newText() #169: new Text DRIVER_INFO for device Altair ALTAIRGP290M
2021-11-19 18:45:51,506 [INFO] MainProcess newMessage() #175: new Message 2021-11-19T17:45:51: [INFO] Altair ALTAIRGP290M is online. Retrieving basic data.
2021-11-19 18:45:51,506 [INFO] MainProcess newSwitch() #162: new Switch CONNECTION for device Altair ALTAIRGP290M
2021-11-19 18:45:51,516 [INFO] MainProcess newMessage() #175: new Message 2021-11-19T17:45:51: [INFO] Maximum gain considering dual gain is 22500.
#CCDCONFIGUREDEVICE ........
#CCDSETFRAMETYPE ........
2021-11-19 18:46:02,372 [INFO] MainProcess configureDevice() #318: Setting switch CCD_FRAME_TYPE
2021-11-19 18:46:02,477 [INFO] MainProcess newSwitch() #162: new Switch CCD_FRAME_TYPE for device Altair ALTAIRGP290M
#CCDGETCCDINFO ........
2021-11-19 18:46:03,679 [INFO] MainProcess getCcdInfo() #243: Detecting bayer pattern
2021-11-19 18:46:08,689 [WARNING] MainProcess getCcdInfo() #252: CCD_CFA fetch timeout, assuming monochrome camera
2021-11-19 18:46:08,891 [INFO] MainProcess _initialize() #315: Minimum CCD exposure: 0.00010500
2021-11-19 18:46:08,892 [INFO] MainProcess _initialize() #321: Default CCD exposure: 0.00010000
2021-11-19 18:46:08,892 [INFO] MainProcess _initialize() #328: CCD CFA: None
2021-11-19 18:46:08,892 [INFO] MainProcess _startImageWorker() #338: Starting ImageWorker process
2021-11-19 18:46:08,961 [INFO] MainProcess _startVideoWorker() #378: Starting VideoWorker process
2021-11-19 18:46:08,973 [INFO] ImageWorker001 info() #54: child process calling self.run()
2021-11-19 18:46:09,003 [INFO] MainProcess _startFileUploadWorker() #411: Starting FileUploader process 1
2021-11-19 18:46:09,010 [INFO] VideoWorker001 info() #54: child process calling self.run()
2021-11-19 18:46:09,017 [INFO] MainProcess detectNight() #592: Sun altitude: 37:03:50.6
2021-11-19 18:46:09,020 [INFO] MainProcess detectMoonMode() #610: Moon altitide: -37:12:42.0, phase 99.9%
2021-11-19 18:46:09,020 [WARNING] MainProcess reconfigureCcd() #559: Change to day
2021-11-19 18:46:09,021 [WARNING] MainProcess setCcdGain() #379: Setting CCD gain to 100
2021-11-19 18:46:09,021 [INFO] MainProcess configureDevice() #313: Setting property CCD_CONTROLS
2021-11-19 18:46:09,022 [INFO] FileUploader001 info() #54: child process calling self.run()
2021-11-19 18:46:10,275 [WARNING] MainProcess setCcdBinning() #441: Setting CCD binning to (1, 1)
2021-11-19 18:46:10,276 [INFO] MainProcess configureDevice() #313: Setting property CCD_BINNING
2021-11-19 18:46:12,531 [INFO] MainProcess shoot() #837: Taking 0.00010000 s exposure (gain 100)
2021-11-19 18:46:12,683 [INFO] MainProcess newMessage() #175: new Message 2021-11-19T17:46:12: [ERROR] Requested exposure value (0.0001) seconds out of bounds [0.000105,1000].
Traceback (most recent call last):
  File "./allsky.py", line 99, in <module>
    action_func(*args_list)
  File "/home/astroberry/indi-allsky/indi_allsky/allsky.py", line 485, in run
    self.shoot(self.exposure_v.value)
  File "/home/astroberry/indi-allsky/indi_allsky/allsky.py", line 838, in shoot
[... cut ...]
  File "/home/astroberry/indi-allsky/indi_allsky/indi.py", line 514, in set_number
    self.__wait_for_ctl_statuses(c, timeout=timeout)
  File "/home/astroberry/indi-allsky/indi_allsky/indi.py", line 582, in __wait_for_ctl_statuses
    raise RuntimeError('Error while changing property {0}'.format(ctl.getName()))
RuntimeError: Error while changing property CCD_EXPOSURE
2021-11-19 18:46:12,795 [INFO] MainProcess info() #54: process shutting down
2021-11-19 18:46:12,796 [INFO] MainProcess info() #54: calling join() for process FileUploader001
```

FlareP1, 2021-11-19:
I have got it running now, needed to make the following additions/ changes to the config.json due to exposure limitations

```
    "CCD_EXPOSURE_MIN" : 0.00015,
    "CCD_EXPOSURE_DEF" : 0.00015,
```

Next to try and fine tune the settings

aaronwmorris, 2021-11-19:
Great news!  I probably do need to increase my default/minimum exposure times.  It is difficult to find settings that are compatible with all cameras.  :-)

FlareP1, 2021-11-19:
Thanks, note I needed to increase to 0.002 to make the capture reliable, edited my post above

aaronwmorris, 2021-11-19:
I will make an example config for Altair cameras with those settings.

aaronwmorris, 2021-11-19:
I forgot to ask you... can you run `indi_getprop` after connecting to the camera and send me the output?  I like to have all of the possible configuration parameters for different camera vendors available.

FlareP1, 2021-11-20:
arron, I think this is what you want.  See info below.  Can I add any of these settings to the config.json?   Also I think that the gain values will need some optimising.  Has been cloudy so far so have not changed from the defaults but think I may need a little more gain at night

Also have you considered allowing integration of multiple shorter exposures to improve dynamic range?  Especially for 8bit devices.

`indi_getprop -h 127.0.0.1 -p 7624 -t 10 "*.*.*"`
 
```
Altair ALTAIRGP290M.CONNECTION.CONNECT=On
Altair ALTAIRGP290M.CONNECTION.DISCONNECT=Off
Altair ALTAIRGP290M.DRIVER_INFO.DRIVER_NAME=Altair
Altair ALTAIRGP290M.DRIVER_INFO.DRIVER_EXEC=indi_altair_ccd
Altair ALTAIRGP290M.DRIVER_INFO.DRIVER_VERSION=0.7
Altair ALTAIRGP290M.DRIVER_INFO.DRIVER_INTERFACE=6
Altair ALTAIRGP290M.POLLING_PERIOD.PERIOD_MS=1000
Altair ALTAIRGP290M.DEBUG.ENABLE=Off
Altair ALTAIRGP290M.DEBUG.DISABLE=On
Altair ALTAIRGP290M.SIMULATION.ENABLE=Off
Altair ALTAIRGP290M.SIMULATION.DISABLE=On
Altair ALTAIRGP290M.CONFIG_PROCESS.CONFIG_LOAD=Off
Altair ALTAIRGP290M.CONFIG_PROCESS.CONFIG_SAVE=Off
Altair ALTAIRGP290M.CONFIG_PROCESS.CONFIG_DEFAULT=Off
Altair ALTAIRGP290M.CONFIG_PROCESS.CONFIG_PURGE=Off
Altair ALTAIRGP290M.ACTIVE_DEVICES.ACTIVE_TELESCOPE=Telescope Simulator
Altair ALTAIRGP290M.ACTIVE_DEVICES.ACTIVE_ROTATOR=Rotator Simulator
Altair ALTAIRGP290M.ACTIVE_DEVICES.ACTIVE_FOCUSER=Focuser Simulator
Altair ALTAIRGP290M.ACTIVE_DEVICES.ACTIVE_FILTER=CCD Simulator
Altair ALTAIRGP290M.ACTIVE_DEVICES.ACTIVE_SKYQUALITY=SQM
Altair ALTAIRGP290M.CCD_EXPOSURE.CCD_EXPOSURE_VALUE=0
Altair ALTAIRGP290M.CCD_ABORT_EXPOSURE.ABORT=Off
Altair ALTAIRGP290M.CCD_FRAME.X=0
Altair ALTAIRGP290M.CCD_FRAME.Y=0
Altair ALTAIRGP290M.CCD_FRAME.WIDTH=1920
Altair ALTAIRGP290M.CCD_FRAME.HEIGHT=1080
Altair ALTAIRGP290M.CCD_BINNING.HOR_BIN=1
Altair ALTAIRGP290M.CCD_BINNING.VER_BIN=1
Altair ALTAIRGP290M.FITS_HEADER.FITS_OBSERVER=Unknown
Altair ALTAIRGP290M.FITS_HEADER.FITS_OBJECT=Unknown
Altair ALTAIRGP290M.CCD_INFO.CCD_MAX_X=1920
Altair ALTAIRGP290M.CCD_INFO.CCD_MAX_Y=1080
Altair ALTAIRGP290M.CCD_INFO.CCD_PIXEL_SIZE=2.9000000953674316406
Altair ALTAIRGP290M.CCD_INFO.CCD_PIXEL_SIZE_X=2.9000000953674316406
Altair ALTAIRGP290M.CCD_INFO.CCD_PIXEL_SIZE_Y=2.9000000953674316406
Altair ALTAIRGP290M.CCD_INFO.CCD_BITSPERPIXEL=16
Altair ALTAIRGP290M.CCD_COMPRESSION.CCD_COMPRESS=Off
Altair ALTAIRGP290M.CCD_COMPRESSION.CCD_RAW=On
Altair ALTAIRGP290M.TELESCOPE_TIMED_GUIDE_NS.TIMED_GUIDE_N=0
Altair ALTAIRGP290M.TELESCOPE_TIMED_GUIDE_NS.TIMED_GUIDE_S=0
Altair ALTAIRGP290M.TELESCOPE_TIMED_GUIDE_WE.TIMED_GUIDE_W=0
Altair ALTAIRGP290M.TELESCOPE_TIMED_GUIDE_WE.TIMED_GUIDE_E=0
Altair ALTAIRGP290M.CCD_FRAME_TYPE.FRAME_LIGHT=On
Altair ALTAIRGP290M.CCD_FRAME_TYPE.FRAME_BIAS=Off
Altair ALTAIRGP290M.CCD_FRAME_TYPE.FRAME_DARK=Off
Altair ALTAIRGP290M.CCD_FRAME_TYPE.FRAME_FLAT=Off
Altair ALTAIRGP290M.TELESCOPE_TYPE.TELESCOPE_PRIMARY=On
Altair ALTAIRGP290M.TELESCOPE_TYPE.TELESCOPE_GUIDE=Off
Altair ALTAIRGP290M.WCS_CONTROL.WCS_ENABLE=Off
Altair ALTAIRGP290M.WCS_CONTROL.WCS_DISABLE=On
Altair ALTAIRGP290M.UPLOAD_MODE.UPLOAD_CLIENT=On
Altair ALTAIRGP290M.UPLOAD_MODE.UPLOAD_LOCAL=Off
Altair ALTAIRGP290M.UPLOAD_MODE.UPLOAD_BOTH=Off
Altair ALTAIRGP290M.UPLOAD_SETTINGS.UPLOAD_DIR=/home/astroberry
Altair ALTAIRGP290M.UPLOAD_SETTINGS.UPLOAD_PREFIX=IMAGE_XXX
Altair ALTAIRGP290M.CCD_VIDEO_STREAM.STREAM_ON=Off
Altair ALTAIRGP290M.CCD_VIDEO_STREAM.STREAM_OFF=On
Altair ALTAIRGP290M.STREAM_DELAY.STREAM_DELAY_TIME=0
Altair ALTAIRGP290M.STREAMING_EXPOSURE.STREAMING_EXPOSURE_VALUE=0.10000000000000000555
Altair ALTAIRGP290M.STREAMING_EXPOSURE.STREAMING_DIVISOR_VALUE=1
Altair ALTAIRGP290M.FPS.EST_FPS=30
Altair ALTAIRGP290M.FPS.AVG_FPS=30
Altair ALTAIRGP290M.RECORD_STREAM.RECORD_ON=Off
[... 64 lines cut ...]
Altair ALTAIRGP290M.CCD_VIDEO_STREAM.STREAM_ON=Off
Altair ALTAIRGP290M.CCD_VIDEO_STREAM.STREAM_OFF=On
Altair ALTAIRGP290M.STREAMING_EXPOSURE.STREAMING_EXPOSURE_VALUE=0.10000000000000000555
Altair ALTAIRGP290M.STREAMING_EXPOSURE.STREAMING_DIVISOR_VALUE=1
Altair ALTAIRGP290M.FPS.EST_FPS=30
Altair ALTAIRGP290M.FPS.AVG_FPS=30
Altair ALTAIRGP290M.RECORD_STREAM.RECORD_ON=Off
Altair ALTAIRGP290M.RECORD_STREAM.RECORD_DURATION_ON=Off
Altair ALTAIRGP290M.RECORD_STREAM.RECORD_FRAME_ON=Off
Altair ALTAIRGP290M.RECORD_STREAM.RECORD_OFF=On
Altair ALTAIRGP290M.RECORD_FILE.RECORD_FILE_DIR=/home/astroberry/indi__D_
Altair ALTAIRGP290M.RECORD_FILE.RECORD_FILE_NAME=indi_record__T_
Altair ALTAIRGP290M.RECORD_OPTIONS.RECORD_DURATION=1
Altair ALTAIRGP290M.RECORD_OPTIONS.RECORD_FRAME_TOTAL=30
Altair ALTAIRGP290M.CCD_STREAM_FRAME.X=0
Altair ALTAIRGP290M.CCD_STREAM_FRAME.Y=0
Altair ALTAIRGP290M.CCD_STREAM_FRAME.WIDTH=1920
Altair ALTAIRGP290M.CCD_STREAM_FRAME.HEIGHT=1080
Altair ALTAIRGP290M.CCD_STREAM_ENCODER.RAW=On
Altair ALTAIRGP290M.CCD_STREAM_ENCODER.MJPEG=Off
Altair ALTAIRGP290M.CCD_STREAM_RECORDER.SER=On
Altair ALTAIRGP290M.CCD_STREAM_RECORDER.OGV=Off
Altair ALTAIRGP290M.LIMITS.LIMITS_BUFFER_MAX=512
Altair ALTAIRGP290M.LIMITS.LIMITS_PREVIEW_FPS=10
```

aaronwmorris, 2021-11-20:
Thanks, that is exactly what I needed.

Yes, you can add arbitrary settings in the INDI_CONFIG_DEFAULTS section.  There are generally two types of settings "numbers" (which I call PROPERTIES) and "switches".  You can take a look at some of the examples I have to see how each is set.  Sometimes, you have to use Kstars/EKOS to figure out which are which.

I agree you will likely have to tune your gain settings.  I am not familiar with common settings for Altair.  Reading through the settings above and Altair's documentation, your camera appears to have a High Conversion Gain mode which you should probably use.  It looks like HCG does not kick in until a gain of 900.  You may want to consider running a night time gain of 1000-1500.

As for stacking short exposures, I am not sure it would have much benefit.  Astro camera's analog gain usually does a pretty good job of maximizing detail.  With an all sky camera, even with a wide field lens, there is too much movement to really stack pictures.  There is detectable movement even with 15 second exposures.

FlareP1, 2021-11-21:
Thanks have managed to get some of the indi settings to work had to fiddle around with the syntax, but I think the following is correct.  The settings do not seem to be controlled correctly.  I wanted to force the high conversion gain but at a lover gain value.  I think 900 is too much.  I need to spend more time debugging it.

```
    "INDI_CONFIG_DEFAULTS" : {
        "PROPERTIES" : {
			"TC_HGC_SET" : {"HCG Threshold" : 100},
			"TC_HGC_SET" : {"HCG/LCG gain ratio" : 4.5}
		},
        "SWITCHES" : {
            "TC_HCG_CONTROL" : {
                "off"	: ["GAIN_LOW"],
                "on"	: ["GAIN_HIGH"],
                "off"	: ["GAIN_HDR"]
            }
        }
```

Regarding the stacking the key here is to provide dynamic range extension.  I want to use shorter exposures, to prevent the saturation of the CCD, but then stack them in a tiff, so I should get the full 16bits, but all still within the 15sec.  Currently when I up the gain to be able to see more stars I massively saturate some of the brighter stars resulting in bloating.  If this is done in a stack this should be less of a problem?  Many be I just havent got my setup quite right yet.  Your image on the project home page looks much better.  What settings, camera, lens etc did you use for this image?

aaronwmorris, 2021-11-21:
Try this for your settings:

    "INDI_CONFIG_DEFAULTS" : {
        "PROPERTIES" : {
            "TC_HGC_SET" : {
                "HCG Threshold" : 100,
                "HCG/LCG gain ratio" : 4.5
            }
        },
        "SWITCHES" : {
            "TC_HCG_CONTROL" : {
                "on"  : ["GAIN_HIGH"],
                "off" : ["GAIN_LOW", "GAIN_HDR"]
            }
        }
    }

FlareP1, 2021-11-22:
Thanks yes that works.

I am interested in what lens you use on your camera I am getting good sensitivity but can only get focus in the centre portion of the lens

[image]

aaronwmorris, 2021-11-22:
On both of my cameras, I think I used a ZWO 2.1mm lens with my ASI290mm, and a ZWO 2.5mm on the Svbony sv305.  I just had these on hand from other purchases.

Achieving focus can be a challenge.  In the past, I have hooked my camera up to my laptop and run something like firecapture so I could run a real time video preview with shorter exposures like 0.25-0.5s with a very high gain.  You can use brighter stars using this method.  I have also found that using trees off in the distance that have a brighter background shining through them can make a good focus target.

aaronwmorris, 2022-02-01:
Closing this issue for now.

closed 2022-02-01 by aaronwmorris