aaronwmorris/indi-allsky #26: *Raspbian 11 is not yet support in the Astroberry repo*

AMoN-AMoK, 2022-01-16:
looks like is some unupdated libs on astroberry

pi@allsky:~/indi-allsky $ ./setup.sh
###############################################
### Welcome to the indi-allsky setup script ###
###############################################

Distribution: Raspbian
Release: 11

INDI_DRIVER_PATH: /usr/bin
INDISERVER_SERVICE_NAME: indiserver
ALLSKY_SERVICE_NAME: indi-allsky
HTDOCS_FOLDER: /var/www/html/allsky
DB_FOLDER: /var/lib/indi-allsky

Setup proceeding in 10 seconds... (control-c to cancel)

**** Installing packages... ****

Current default time zone: 'Europe/Madrid'
Local time is now:      Sun Jan 16 17:43:52 CET 2022.
Universal Time is now:  Sun Jan 16 16:43:52 UTC 2022.

Raspbian 11 is not yet support in the Astroberry repo

aaronwmorris, 2022-01-16:
This is a safety check in my setup script.  Unfortunately, the indi packages are not yet available for Raspbian 11.

You will have to run Raspbian 10 (Legacy) if you want to run indi-allsky or other indi components.

AMoN-AMoK, 2022-01-16:
Lets make a downgrade so.
Thanks Aaron!

aaronwmorris, 2022-01-16:
Sure, let me know if you have any problems!