aaronwmorris/indi-allsky #43: Installation Problem

xrb12250, 2022-01-30:
Hi Aaron,

the setup script fails to successfully install indi-allsky:
**** Python virtualenv setup ****
mkdir: cannot create directory ‘/home/pi/indi-allsky/virtualenv’: Permission denied

Trying to use sudo or root fails because of your checking routine... 

Any ideas?

System is a pi 4 with a clean install raspbian light (buster)

xrb12250, 2022-01-31:
Ok solved the problem with additional sudo's

`echo "**** Python virtualenv setup ****"
[[ ! -d "${ALLSKY_DIRECTORY}/virtualenv" ]] && sudo mkdir -m 755 "${ALLSKY_DIRECTORY}/virtualenv"
if [ ! -d "${ALLSKY_DIRECTORY}/virtualenv/indi-allsky" ]; then
    sudo virtualenv -p python3 ${ALLSKY_DIRECTORY}/virtualenv/indi-allsky
fi
`

xrb12250, 2022-01-31:
> Ok solved the problem with additional sudo's
>
> `echo "**** Python virtualenv setup ****" [[ ! -d "${ALLSKY_DIRECTORY}/virtualenv" ]] && sudo mkdir -m 755 "${ALLSKY_DIRECTORY}/virtualenv" if [ ! -d [...]

As expected, the whole installation is now somehow screwed up 😄 I'll give it another try

aaronwmorris, 2022-02-01:
Did you perform the git checkout as root or something?  You probably just need to change the ownership of the git checkout.

`sudo chown -R pi /home/pi/indi-allsky`

aaronwmorris, 2022-02-01:
I have updated the setup.sh script to fix the ownership of the git checkout if it is incorrect.

xrb12250, 2022-02-01:
This works now, but there are some new problems now, I'll open a new issue.

closed 2022-02-01 by xrb12250