aaronwmorris/indi-allsky #46: Debian 11 installation

Marmi86, 2022-02-01:
I want to install the allsky program on debian 11. 

During the installation I get the error message dbus python cannot be installed.

Do you have a solution for this?

Use it in a VM with Debian 11

aaronwmorris, 2022-02-01:
I booted up my Debian 11 VM and found I missed some development packages.  I have merged #48 and #49 so that the installation will complete.

Debian 11 might work, if you could test it I would appreciate it.  Debian 11 does not include all of the indi proprietary packages, but my script installs indi from the Debian 12 (bookworm) test repository so the individual proprietary packages will have to be manually installed and then run the setup.sh script again to setup the service.  Just pick the simulator server on the first run.

aaronwmorris, 2022-02-01:
I have merged #50 to install the ZWO ASI and a couple of other drivers.  It appears not all of the drivers are available in the repository.

Marmi86, 2022-02-02:
Which os are you using? have problems with all os whether ubuntu 20.04 debian 10 11

aaronwmorris, 2022-02-03:
All of my permanent cameras run under Raspbian 10 on a Raspberry PI, but I have done plenty of testing with Ubuntu 20.04 on the x86 platform, as well.

I just tested the deployment with a virgin Debian 11 and Ubuntu 20.04 VM images.  Both fully deployed the software.  Ubuntu 20.04 for x86 has better driver support (via PPA) than Debian 11 (but Debian 11 still works from my testing).

What kind of problems are you still seeing?

Marmi86, 2022-02-03:
I have found the error I use Proxmox and have tested there indi-allsky with container images (LXC). I have now installed a VM with Ubuntu Server and now it works :)

Now I am testing remote indi 

My hardware:
Intel Nuc I7 9 Gen
16gb Ram
Proxmox VM Software

aaronwmorris, 2022-02-03:
I have used Proxmox before, but I do not know what it would take to run indi-allsky under a container, or if it would even be possible.

I use KVM for my VM image testing and it works well enough, although, I generally only test the CCD simulator under a VM.  I have never tried passing a USB camera through to a guest.  Theoretically, it should work.

Marmi86, 2022-02-05:
Now it works with the remote device, I solved it a little differently:

[image]

This way I can run indi-allsky on my server and I have only  a Pi-Zero as a USB server.

aaronwmorris, 2022-02-06:
Just to let you know, indi-allsky can be run as an INDI client for a remote INDI server.  You would just need to run the indiserver on the Pi Zero and connect to the INDI service via the network.  There are command line arguments you can use to connect to the INDI server, instead of connecting to localhost (which is the default).

The command line argument is `-s` or `--server`

Marmi86, 2022-02-06:
Thank you for the advice 

The setup now looks like this:
Host Client: 
Proxmox Ubuntu 20.4 VM with indi-allsky

Remote Server
Pi Zero with RaspiOS "Buster" do not get indi installed with Bullseye
Software: Indi driver & Indiwebmanager

I am just starting indi-allsky in manual mode:

1. source virtualenv/indi-allsky/bin/activate
2. ./allsky.py run -s IP from Pi

Could I also set this to connect to the remote indiserver and control it from the webgui ?

Translated with www.DeepL.com/Translator (free version)

aaronwmorris, 2022-02-06:
Controlling the remote indi server would not be possible from the system view.

The green "RUNNING" status would work and and the active/inactive status of the indi-allsky service should work because those are local processes to where the web application is running.  However, a remote indiserver would always show as inactive since that is a remote process.

Marmi86, 2022-02-06:
So when I connect indi allsky to a remote indiserver, I can not run the indi allsky process via webgui.
I have to start and stop it manually via command line.

aaronwmorris, 2022-02-06:
Are there any exceptions recorded in ```/var/log/indi-allsky/webapp-indi-allsky.log``` ?

Marmi86, 2022-02-06:
My Log:

[webapp-indi-allsky.log](https://github.com/aaronwmorris/indi-allsky/files/8011104/webapp-indi-allsky.log)

aaronwmorris, 2022-02-06:
Nothing important in the log.  I will setup an environment and run some tests.

aaronwmorris, 2022-02-06:
Are you running indi-allsky from the command line or running via systemd?

Marmi86, 2022-02-06:
I use command line
I don't know what the option is for systemd

aaronwmorris, 2022-02-06:
I understand now.  The system page detects the service via dbus and systemd.  When you run from the command line, systemd is not aware of the processes, so it reports "inactive".

I have merged #59 which removes the command line option to specify the server on the command line and added the server and port information directly to the config file.  You can edit the hostname (and port) from the config page.

Once you do this, you can start the service from the system page.  You can monitor the output by watching the log.

```
tail -f /var/log/indi-allsky/indi-allsky.log
```

Marmi86, 2022-02-07:
Offtopic:
Can you recommend the svbony 305? would like to switch from the canon d550 to a smaller camera.

aaronwmorris, 2022-02-07:
The sv305 was the original camera I started with I still have it running in a permanent enclosure.  It has been in the enclosure for a year now.

Here are a couple of time lapses from my color sv305:
https://youtu.be/ClXrkSZQbmY
https://youtu.be/izlUewgaGAM

I also run my sv305 during the day, so I can upload a daytime timelapse if you are interested in seeing one.

The only problem I have run into with the sv305 is about 20% of the exposures take double the exposure time.  I have suspected this is a firmware bug, but I have not been able to find the cause yet.

Marmi86, 2022-02-07:
Nice Timelapse

aaronwmorris, 2022-02-07:
Thank you!  Other than the exposure issue I mentioned, the sv305 is a solid camera.  The driver is very stable too.  Weeks of uptime are not unusual and the camera is almost never the reason I have to reboot the system.

During the summer, temperatures reached 60-70c inside the enclosure and it is still operating fine.

Marmi86, 2022-02-07:
Yes i am still comparing the sv305 with the pi hq cam .
I have the pi zero ,but i don't know if that works so well pi zero with indi server remote and HD picture recording

aaronwmorris, 2022-02-07:
I also have a Raspberry Pi HQ camera I am planning on putting in a permanent enclosure.  The camera does work with indi and indi-allsky, but...

I will be honest, the Pi HQ camera is a bit of a mess.  The camera will work, but the experience is less than perfect.  I will stop just short of not recommending the camera, but I think you will have a better/easier experience with the sv305.

Marmi86, 2022-02-07:
Thank you 

I will probably buy the sv305 

I have now tested my pi zero with the remote indi server and with virtual here and I have come to the conclusion that for the pi zero the variant with virtual here performance runs much better.

Marmi86, 2022-02-08:
Hello 
what is this :) 
Feb  8 22:09:40 allsky-PC-VM [INFO] MainProcess allsky.detectMoonMode() #685: Moon Mode conditions detected
Feb  8 22:09:50 allsky-PC-VM [INFO] MainProcess allsky.detectNight() #663: Sun altitude: -42:32:10.9
Feb  8 22:09:50 allsky-PC-VM [INFO] MainProcess allsky.detectMoonMode() #681: Moon altitide: 35:18:30.0, phase 53.1%
Feb  8 22:09:50 allsky-PC-VM [INFO] MainProcess allsky.detectMoonMode() #685: Moon Mode conditions detected

Marmi86, 2022-02-08:
My cam makes no pictures

aaronwmorris, 2022-02-08:
Someone else mentioned a similar issue.  I think an unhandled exception is occurring which was not being logged.  An unhandled exception will cause the main program loop to stop.

I merged #67 so that unhandled exceptions are logged.

Marmi86, 2022-02-08:
I tested my lifecam today 
Hardware:
Pi with remote indi server and Microsoft lifecam usb and then I got this bug

I will tomorrow times connect it directly to the pc and see what happens then

Marmi86, 2022-02-09:
How can I switch on this resolution in the command line :

V4L2 CCD.V4L2_SIZE_DISCRETE.1920x1080=Off

gilesco, 2022-02-10:
The way I've been doing it is going to the top level indi-allsky, running "git pull" and then re-running setup.sh, then rebooting. It appears to retain your previous configuration.

aaronwmorris, 2022-02-10:
@gilesco is correct.  If there are any changes to the database or application configurations, re-running setup.sh is the way I intend to incorporate those changes.  I do not touch the indi-allsky config.json file, but some of the other config files (apache, gunicorn, etc) may be overwritten.  I do my best to minimize disruptive changes.

One of the last changes I made was to the gunicorn configuration to allow python reloading when the web application code changes.  Hopefully, this will minimize the 500 error codes when the web interface changes.

aaronwmorris, 2022-03-25:
Closing this issue for now.  Let me know if there is anything else I can help with.

closed 2022-03-25 by aaronwmorris