aaronwmorris/indi-allsky #47: Web Interface not working

xrb12250, 2022-02-01:
**So this is the new issue: The Webserver is working, but no web interface, only the Apache placeholder page is shown**

`pi@raspberrypi:~ $ cd indi-allsky
pi@raspberrypi:~/indi-allsky $ systemctl --user stop indi-allsky
pi@raspberrypi:~/indi-allsky $ source virtualenv/indi-allsky/bin/activate
(indi-allsky) pi@raspberrypi:~/indi-allsky $ ./allsky.py run
Traceback (most recent call last):
  File "./allsky.py", line 10, in <module>
    app = indi_allsky.flask.create_app()
  File "/home/pi/indi-allsky/indi_allsky/flask/__init__.py", line 66, in create_app
    app.config.from_file(p_flask_config, load=json.load)
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.7/site-packages/flask/config.py", line 196, in from_file
    with open(filename) as f:
FileNotFoundError: [Errno 2] Unable to load configuration file (No such file or directory): '/etc/indi-allsky/flask.json'`

**I don't know if it is important, but this error was also shown at the end of the installation:**

`**** Enabling services ****
Job for gunicorn-indi-allsky.socket failed.
See "systemctl --user status gunicorn-indi-allsky.socket" and "journalctl --user -xe" for details.
pi@raspberrypi:~/indi-allsky $ systemctl --user status gunicorn-indi-allsky.socket
● gunicorn-indi-allsky.socket - indi-allsky gunicorn socket
   Loaded: loaded (/home/pi/.config/systemd/user/gunicorn-indi-allsky.socket; enabled; vendo
   Active: failed (Result: resources)
   Listen: /var/lib/indi-allsky/gunicorn-indi-allsky.sock (Stream)

Feb 01 22:07:37 raspberrypi systemd[636]: gunicorn-indi-allsky.socket: Failed to create list
Feb 01 22:07:37 raspberrypi systemd[636]: gunicorn-indi-allsky.socket: Failed to listen on s
Feb 01 22:07:37 raspberrypi systemd[636]: gunicorn-indi-allsky.socket: Failed with result 'r
Feb 01 22:07:37 raspberrypi systemd[636]: Failed to listen on indi-allsky gunicorn socket.
Feb 01 22:09:32 raspberrypi systemd[636]: gunicorn-indi-allsky.socket: Failed to create list
Feb 01 22:09:32 raspberrypi systemd[636]: gunicorn-indi-allsky.socket: Failed to listen on s
Feb 01 22:09:32 raspberrypi systemd[636]: gunicorn-indi-allsky.socket: Failed with result 'r
Feb 01 22:09:32 raspberrypi systemd[636]: Failed to listen on indi-allsky gunicorn socket.
`

aaronwmorris, 2022-02-01:
I had a race condition in my setup.sh script that failed to create a folder to old the gunicorn socket file.  #49 

Pull down my latest code and run setup.sh again.

xrb12250, 2022-02-03:
Perfect, works!

closed 2022-02-03 by xrb12250