aaronwmorris/indi-allsky #55: Password and login

Frankastro85, 2022-02-05:
Hello,
My installation is successful (I think) but I can't access to the web page https://raspberrypi/indi-allsky/.
I don't know the login and pwd. Is-it the same of the Pi ?
Thank you

Frankastro85, 2022-02-05:
Hmmm, ok, I found it on the site of Thomas Jacquin :-)

aaronwmorris, 2022-02-05:
Excellent.  I used the same authentication credentials.

It is documented in the github readme:  https://github.com/aaronwmorris/indi-allsky#web-interface

Frankastro85, 2022-02-05:
Yes, perfect.
So now, on my Raspberry (new card), I've installed both Allsky -> Thomas Jacquin and You.
The Allsky run perfectly but I can't run your version.
How can I test your Indi-Allsky ?
Thank you.

aaronwmorris, 2022-02-05:
Both software will not run simultaneously.  I believe Thomas Jacquin's Allsky uses lighttpd web server where mine installs Apache.  If you re-run my setup.sh script, it should disable (not uninstall) Thomas Jacquin's Allsky components and services.

Frankastro85, 2022-02-05:
Ok, I'll try that.
Thank you.

ruirigel, 2022-02-20:
I've been looking for where to change the password but can't find it. Any suggestions?

aaronwmorris, 2022-02-20:
You can change the password using the following command:

`htpasswd /etc/indi-allsky/apache.passwd admin`

closed 2022-03-25 by aaronwmorris