aaronwmorris/indi-allsky #64: Starlight Xpress Oculus all-sky camera

bamiesking, 2022-02-07:
Hi,

I'd like to get indi-allsky working with a Starlight Xpress Oculus all-sky camera - is there a driver in indi-allsky that would work with this device? Any help greatly appreciated. 

Thanks.

aaronwmorris, 2022-02-14:
Just let me know if you are still interested.  I do not have a Starlight Xpress camera to test with and I am very interested in making this work.

It really does only take a few minutes to add the correct parameter to the code for the camera to function.

bamiesking, 2022-02-17:
Hi Aaron,

Yes I'm still interested. I'll get back to you on this next week. Sorry for the delay!

aaronwmorris, 2022-02-17:
No problem!  I just did not want to scare you off thinking this was a huge effort.

bamiesking, 2022-03-09:
Hi Aaron, just letting you know I am going to come back to this. Various issues (not related to INDI or indi-allsky) have got in the way. Feel free to close the issue and I will reopen it when I can give you the information you need. Thanks.

aaronwmorris, 2022-03-11:
No worries.  I will be here when you have time!

rkilgard, 2022-05-04:
Hi,

I'd love to reopen this issue. I'm running Astroberry on a Pi 3 and can take images with a Starlight Xpress Oculus through KStars/Ekos, so the Indi drivers are definitely working. Below is what I get from indi_getprop.

Cheers,
Roy

[indi_getprop.txt](https://github.com/aaronwmorris/indi-allsky/files/8626178/indi_getprop.txt)

aaronwmorris, 2022-05-04:
I have merged #189 .  Let me know if that works for you.

I did not see any gain settings reported in the indi_getprop output, does that sound correct to you?

rkilgard, 2022-05-04:
Yes, it doesn't seem to report gain. I'm not sure what that's about but INDI doesn't seem to care. The manufacturer says the gain is 0.3 e-/ADU.

The new version dies in a different place so that's progress. Here's what I get if I attempt to start indi-allsky manually. Maybe it needs the gain hard-coded?

(indi-allsky) astroberry@astroberry:~/indi-allsky $ systemctl --user start indiserver
(indi-allsky) astroberry@astroberry:~/indi-allsky $ source virtualenv/indi-allsky/bin/activate
(indi-allsky) astroberry@astroberry:~/indi-allsky $ ./allsky.py run
/home/astroberry/indi-allsky/virtualenv/indi-allsky/lib/python3.7/site-packages/paramiko/transport.py:236: CryptographyDeprecationWarning: Blowfish has been deprecated
  "class": algorithms.Blowfish,
2022-05-04 18:17:56,717 [INFO] MainProcess indi.__init__() #90: creating an instance of IndiClient
2022-05-04 18:17:56,717 [INFO] MainProcess allsky._initialize() #291: Connecting to indiserver
INDI::BaseClient::connectServer: creating new connection...
INDI::BaseClient::connectServer: Already connected.
2022-05-04 18:17:56,728 [INFO] MainProcess indi.newDevice() #119: new device SX CCD SuperStar
2022-05-04 18:17:56,720 [INFO] MainProcess indi.serverConnected() #172: Server connected (localhost:7624)
2022-05-04 18:17:56,732 [INFO] MainProcess indi.newDevice() #119: new device Telescope Simulator
2022-05-04 18:17:56,739 [INFO] MainProcess indi.newText() #163: new Text ACTIVE_DEVICES for device Telescope Simulator
2022-05-04 18:17:56,740 [INFO] MainProcess indi.newText() #163: new Text ACTIVE_DEVICES for device SX CCD SuperStar
2022-05-04 18:17:56,741 [INFO] MainProcess indi.newText() #163: new Text SCOPE_CONFIG_NAME for device Telescope Simulator
2022-05-04 18:17:56,741 [INFO] MainProcess indi.newSwitch() #156: new Switch MOUNT_TYPE for device Telescope Simulator
2022-05-04 18:17:56,772 [INFO] MainProcess indi.newSwitch() #156: new Switch SIM_PIER_SIDE for device Telescope Simulator
2022-05-04 18:18:04,739 [INFO] MainProcess indi.findCcds() #307: Found device SX CCD SuperStar
2022-05-04 18:18:04,740 [INFO] MainProcess indi.findCcds() #312:  Detected ccd
2022-05-04 18:18:04,740 [INFO] MainProcess indi.findCcds() #312:  Detected guider
2022-05-04 18:18:04,741 [INFO] MainProcess indi.findCcds() #307: Found device Telescope Simulator
2022-05-04 18:18:04,741 [INFO] MainProcess indi.findCcds() #312:  Detected telescope
2022-05-04 18:18:04,742 [INFO] MainProcess indi.findCcds() #312:  Detected guider
2022-05-04 18:18:04,742 [INFO] MainProcess allsky._initialize() #308: Found 1 CCDs
2022-05-04 18:18:04,743 [WARNING] MainProcess allsky._initialize() #311: Connecting to device SX CCD SuperStar
2022-05-04 18:18:04,756 [INFO] MainProcess indi.newText() #163: new Text DRIVER_INFO for device SX CCD SuperStar
2022-05-04 18:18:04,757 [INFO] MainProcess indi.newSwitch() #156: new Switch CONNECTION for device SX CCD SuperStar
2022-05-04 18:18:04,979 [INFO] MainProcess miscDb.addCamera() #43: Camera DB ID: 1
2022-05-04 18:18:04,979 [INFO] MainProcess indi.configureDevice() #327: Setting switch DEBUG
2022-05-04 18:18:05,082 [INFO] MainProcess indi.newSwitch() #156: new Switch DEBUG for device SX CCD SuperStar
2022-05-04 18:18:06,083 [INFO] MainProcess indi.updateCcdBlobMode() #179: Set BLOB mode
2022-05-04 18:18:07,085 [INFO] MainProcess indi.configureDevice() #327: Setting switch CCD_FRAME_TYPE
2022-05-04 18:18:07,188 [INFO] MainProcess indi.newSwitch() #156: new Switch CCD_FRAME_TYPE for device SX CCD SuperStar
2022-05-04 18:18:08,390 [INFO] MainProcess indi.getCcdInfo() #249: Detecting bayer pattern
2022-05-04 18:18:13,405 [WARNING] MainProcess indi.getCcdInfo() #258: CCD_CFA fetch timeout, assuming monochrome camera
2022-05-04 18:18:13,608 [WARNING] MainProcess indi.getCcdGain() #389: indi_sx_ccd does not support gain settings
2022-05-04 18:18:13,608 [INFO] MainProcess allsky._initialize() #356: Minimum CCD exposure: 0.00010001
2022-05-04 18:18:13,609 [INFO] MainProcess allsky._initialize() #371: Default CCD exposure: 0.00010001
2022-05-04 18:18:13,610 [INFO] MainProcess allsky._initialize() #378: CCD CFA: None
2022-05-04 18:18:13,610 [ERROR] MainProcess allsky.unhandled_exception() #36: An uncaught exception occurred:
2022-05-04 18:18:13,610 [ERROR] MainProcess allsky.unhandled_exception() #37: Type: <class 'KeyError'>
2022-05-04 18:18:13,611 [ERROR] MainProcess allsky.unhandled_exception() #38: Value: 'min'
2022-05-04 18:18:13,615 [ERROR] MainProcess allsky.unhandled_exception() #43: '  File "./allsky.py", line 128, in <module>\n    action_func(*args_list, **kwargs_dict)\n'
2022-05-04 18:18:13,616 [ERROR] MainProcess allsky.unhandled_exception() #43: '  File "/home/astroberry/indi-allsky/indi_allsky/allsky.py", line 552, in run\n    self._initialize()\n'
2022-05-04 18:18:13,616 [ERROR] MainProcess allsky.unhandled_exception() #43: '  File "/home/astroberry/indi-allsky/indi_allsky/allsky.py", line 382, in _initialize\n    ccd_min_gain = self.config[\'CCD_INFO\'][\'GAIN_INFO\'][\'min\']\n'

aaronwmorris, 2022-05-05:
Looks like I neglected the code paths for cameras that do not support gain.  Merged #190 to fix.

rkilgard, 2022-05-05:
Awesome! It's working! But all my images look like darks. It appears to be taking 0.0001 second images. Everything is set to the default values. I tried playing with the night gain, target ADU, and default and min exposure settings, but I can't get it to do anything different. I'm sure I'm missing something obvious here.

aaronwmorris, 2022-05-05:
You may have to give it a minute or two for the exposure algorithm to normalize.  It starts at the minimum exposure and increases from there.

If it seems like it gets stuck on the minimum exposure, taking darks can sometimes help with that situation.

rkilgard, 2022-05-05:
It did a great job and ran all night! It didn't generate any of the charts/timelapses though, and clicking on Images produces an Internal Server Error. Is there an action I need to take to enable those processes?

Thanks! This software is amazing. I've been fighting with USB/ethernet extenders for over a year now (and losing), but your software let me put a Pi near the camera and removed a workload from my telescope control computer.

aaronwmorris, 2022-05-05:
Interesting... check the log at `/var/log/indi-allsky/webapp-indi-allsky.log` and let me know what exceptions are generated for those views.

aaronwmorris, 2022-05-05:
Okay, this was from my last schema update for v2 where I added new column to the DB.

You just need to re-run the setup script to kick off the DB migration.  You can also manually run the migration if you are interested in doing that.

https://github.com/aaronwmorris/indi-allsky/wiki/Database-migrations

aaronwmorris, 2022-05-05:
Yeah, nuke it and delete the existing migrations and rerun the setup script.

```
rm -i /var/lib/indi-allsky/indi-allsky.sqlite

# from the indi-allsky folder
rm -i migrations/versions/*.py
```

rkilgard, 2022-05-05:
Alright, we're in business. We'll see how it does tonight. Thanks again!

aaronwmorris, 2022-05-09:
Did everything work out?

aaronwmorris, 2022-05-11:
The behavior you are seeing is expected for now.  I do not have the ability to display historical data currently.  I hope to get to this eventually.

fockez, 2022-10-04:
It seems that I still got 0.0001s exposure for a long time.

aaronwmorris, 2022-10-04:
Are you restarting (or reloading) the indi-allsky service when making the changes?

You may want to try setting a default exposure of something like 0.01 or 0.1 to see if you can get a good starting exposure.  Some cameras give the auto-exposure algorithm problems with extremely low exposures.  Sometimes dark frames can help with this.

fockez, 2022-10-05:
I set 0.1 and restart, now it keeps at 0.1.

aaronwmorris, 2022-10-05:
Can you post your config file here?

fockez, 2022-10-05:
I change to 30s and it successfully adjust to 60s. So, I believe 0.1 is not a good start. But what would be the best practice for this camera?

aaronwmorris, 2022-10-05:
Unfortunately, I do not have a SX camera so I have no real basis to give a recommendation.  My understanding is that the SX cameras have a fixed gain, so that is probably why the 0.1 default exposure is not working (other cameras can adjust to a higher gain level).

The default exposure just needs to be high enough to allow the exposure algorithm to start scaling properly.  You may want to try 1.0 or 5.0.

fockez, 2022-10-06:
If I change the default exposure, should I change the minimal exposure? During the daytime, the exposure will still get lower to the minimal exposure, after that could it possible to get back to the long exposure when getting dark?
Maybe I can provide a remote connection with SX camera, is that enough for you to test this camera?

fockez, 2022-10-06:
In fact, I find that indi driver of sx oculus will cause smear during short exposure, but indigo driver will not.

aaronwmorris, 2022-10-06:
The exposure algorithm will scale the exposure up and down automatically to keep the average brightness of the image at a certain level.

You do not need to change the minimum exposure when you change the default exposure.  The default exposure is just the initial starting exposure, the exposure algorithm will start adjusting the exposure up or down from that level to get an average brightness.

aaronwmorris, 2022-10-06:
If you want to grant remote SSH access or something, I would love to login and take a look.  Let me know.

aaronwmorris, 2022-12-07:
Closing this issue now now.  Reopen if there is anything else needed.

closed 2022-12-07 by aaronwmorris