aaronwmorris/indi-allsky #65: Camera Stopped Acquiring Images when moon mode was deactivated.

gilesco, 2022-02-08:
I'm still running everything in testing, but I noticed this morning that when the moon went below the horizon and my ASI120MC-S transitioned from moon mode to night mode I didn't receive any more images throughout the night.
[1.txt](https://github.com/aaronwmorris/indi-allsky/files/8022750/1.txt)

aaronwmorris, 2022-02-08:
An exception occurred and was not logged to syslog.  I have merged #67 so that unhandled exceptions will be logged to the logfile.

The two most likely causes of an exception here:
- Firmware bugs with camera.  Upgrading your INDI packages can resolve this if the firmware is old.
- The night time gain setting was out of range.

If it occurs again, the exception should be logged this time.

gilesco, 2022-02-09:
It happened again. I did a git pull and re-ran setup.sh when you did that merge.

I don't see anything in the syslog that looks pertinent.

indi-asi is already the newest version (2.0-3).

If I reboot the Pi then it sets the gain (to 29) as per the night setting, and appears to start working again.

Just looks like the transition from moon mode to night mode causes the issue.
[logs-moonmode-transition.txt](https://github.com/aaronwmorris/indi-allsky/files/8030206/logs-moonmode-transition.txt)
[working-after-reboot.txt](https://github.com/aaronwmorris/indi-allsky/files/8030208/working-after-reboot.txt)

gilesco, 2022-02-09:
I noticed there have been further merges since my last pull, will pull and start again, but probably won't be able to test this tonight.

aaronwmorris, 2022-02-09:
I have been looking at my code to see if I can figure out what the issue might be.  One thing that the program does is make sure the camera is ready to take exposures and is possible the change in the gain settings is causing the program to think the camera is not ready.

Merged #72 to log how much time has passed since the camera was ready.  Hopefully, this will produce some useful info.

gilesco, 2022-02-10:
OK, will hopefully be setting up again this afternoon, so will update and set it running and report back.

gilesco, 2022-02-11:
So after the last image before de-activating Moonmode we now get this:

```
Feb 11 04:22:55 pi-camera [INFO] FileUploader001 pycurl_sftp._put() #89: File transferred in 1.4312 s (109.20 kB/s)
Feb 11 04:22:55 pi-camera [INFO] FileUploader001 uploader.run() #103: Upload transaction completed in 1.4424 s
Feb 11 04:23:02 pi-camera [INFO] MainProcess allsky.run() #492: Camera last ready: 9.5s
Feb 11 04:23:02 pi-camera [INFO] MainProcess allsky.run() #493: Exposure state: BUSY
Feb 11 04:23:02 pi-camera [INFO] MainProcess allsky.detectNight() #677: Sun altitude: -30:35:23.6
Feb 11 04:23:02 pi-camera [INFO] MainProcess allsky.detectMoonMode() #695: Moon altitide: -0:00:41.1, phase 73.7%
Feb 11 04:23:02 pi-camera [WARNING] MainProcess allsky.reconfigureCcd() #640: Change to night (normal mode)
Feb 11 04:23:02 pi-camera [WARNING] MainProcess indi.setCcdGain() #401: Setting CCD gain to 29
Feb 11 04:23:02 pi-camera [INFO] MainProcess indi.configureDevice() #331: Setting property CCD_CONTROLS
Feb 11 04:23:03 pi-camera [WARNING] MainProcess indi.setCcdBinning() #463: Setting CCD binning to (1, 1)
Feb 11 04:23:03 pi-camera [INFO] MainProcess indi.configureDevice() #331: Setting property CCD_BINNING
Feb 11 04:23:04 pi-camera kernel: [43719.327464] usb 1-1.2: reset high-speed USB device number 3 using xhci_hcd
Feb 11 04:23:04 pi-camera [INFO] MainProcess allsky.reconfigureCcd() #660: Sensor temperature: 5.7
Feb 11 04:23:10 pi-camera telegraf[497]: 2022-02-11T04:23:10Z D! [outputs.influxdb] Wrote batch of 1 metrics in 22.404758ms
Feb 11 04:23:10 pi-camera telegraf[497]: 2022-02-11T04:23:10Z D! [outputs.influxdb] Buffer fullness: 0 / 10000 metrics
Feb 11 04:23:15 pi-camera [INFO] MainProcess allsky.run() #492: Camera last ready: 23.1s
Feb 11 04:23:15 pi-camera [INFO] MainProcess allsky.run() #493: Exposure state: BUSY
Feb 11 04:23:15 pi-camera [INFO] MainProcess allsky.detectNight() #677: Sun altitude: -30:33:18.7
Feb 11 04:23:15 pi-camera [INFO] MainProcess allsky.detectMoonMode() #695: Moon altitide: -0:01:57.9, phase 73.7%
Feb 11 04:23:15 pi-camera [INFO] MainProcess allsky.reconfigureCcd() #627: Sensor temperature: 5.7
Feb 11 04:23:25 pi-camera [INFO] MainProcess allsky.run() #492: Camera last ready: 33.1s
Feb 11 04:23:25 pi-camera [INFO] MainProcess allsky.run() #493: Exposure state: BUSY
Feb 11 04:23:25 pi-camera [INFO] MainProcess allsky.detectNight() #677: Sun altitude: -30:31:46.2
Feb 11 04:23:25 pi-camera [INFO] MainProcess allsky.detectMoonMode() #695: Moon altitide: -0:02:54.7, phase 73.7%
Feb 11 04:23:25 pi-camera [INFO] MainProcess allsky.reconfigureCcd() #627: Sensor temperature: 5.7
Feb 11 04:23:35 pi-camera [INFO] MainProcess allsky.run() #492: Camera last ready: 43.2s
Feb 11 04:23:35 pi-camera [INFO] MainProcess allsky.run() #493: Exposure state: BUSY
Feb 11 04:23:35 pi-camera [INFO] MainProcess allsky.detectNight() #677: Sun altitude: -30:30:13.7
Feb 11 04:23:35 pi-camera [INFO] MainProcess allsky.detectMoonMode() #695: Moon altitide: -0:03:51.4, phase 73.7%
Feb 11 04:23:35 pi-camera [INFO] MainProcess allsky.reconfigureCcd() #627: Sensor temperature: 5.7
```

Exposure State remains BUSY until the Pi is rebooted.

aaronwmorris, 2022-02-11:
Well, my guess about the exposure control being busy (or in an error state) was correct.  I hope the firmware update will correct this behavior.

The bit depth detection is really just a guess based on the maximum value returned by the image data.  It is common when the gain changes for the number to fluctuate until the exposure stabilizes.  The darks should be in whatever the original bit depth is returned from the camera.  ZWO cameras that return greater than 8 bit data, usually auto-scale their data from [for example] 12 bits to 16 bits.  Other vendors will return the original 12 bit data (in a 16 bit space).

You can view the darks metadata info in a hidden view:  https://hostname/indi-allsky/darks

aaronwmorris, 2022-02-11:
I added the moon mode disable option just for you.  :-)

I would not worry about the bit depth detection.  The only reason the code does that is it needs to know the max value to properly downsample the images to 8-bit jpeg files.  The bit detection code does not affect darks.  The bit depth detection for the darks is a different process (from the messages you see).

You can also run your camera in exclusively 8-bit mode, but you will need to take a new set of 8-bit darks.  Add the following to the indi custom properties.  Swap the on/off values to force 16-bit mode.

```
{
        "PROPERTIES" : {},
        "SWITCHES" : {
            "CCD_VIDEO_FORMAT" : {
                "on"  : ["ASI_IMG_RAW8"],
                "off" : ["ASI_IMG_RAW16"]
        }
}
```

Note:  These are the options I used in my ASI290mm.  The options might be different with your model.

gilesco, 2022-02-11:
Cheers, I was staring at that part of the config and wondering if there was something I could do with that!

We have rain and bad conditions forecast, so the box is indoors for now.

Will provide more feedback which will probably be on the next moon cycle. With a large amount of ongoing commits, do you want to close this issue and we re open again in the future if necessary?

aaronwmorris, 2022-02-25:
I think I know what caused this problem now.  The project has some defaults set for gain, specifically:

- Night: 250
- Moon Mode: 75
- Day: 0

Day and Moon mode are within the limits of the AS120, which has a max gain of 100.  When the camera tried to switch over to night mode, it tried to set gain 250 and failed.

Because the software supports so many different cameras and each camera can have a different gain range, it is difficult to come up with sane defaults.

I have merged #101 which will verify the gain settings are within the limits of the camera.

gilesco, 2022-02-26:
Not sure whether that could be the cause, I had set the gain modes for day, night and moon on the config page to 0, 29 & 20 respectively. Unless there was something hardcoded to try a setting of 250 when going to night mode.
Moon is not currently an issue, and I have set the moon altitude to 91 as per your earlier change.

The ASI120MC is not the best for the AllSky, it is just a camera that I happened to have lying around, If budget allows I will look to use a ASI178MM in the future.

aaronwmorris, 2022-02-28:
ASI120MC-S firmware v3.0 does not appear to fix the issue.  When switching from day (gain 0) to night (gain 100) it stopped responding again.

Although, I did find something else odd... in the morning, shortly after changing to day mode (gain 100 -> 0) the camera change from 16-bit mode to 8-bit mode.  Nothing in my code would have caused this change.  

```
Feb 27 06:41:38 allskypi [INFO] MainProcess allsky.run() #550: Camera last ready: 4.1s
Feb 27 06:41:38 allskypi [INFO] MainProcess allsky.run() #551: Exposure state: BUSY
Feb 27 06:41:38 allskypi [INFO] MainProcess allsky.detectNight() #785: Sun altitude: -5:46:07.1
Feb 27 06:41:38 allskypi [INFO] MainProcess allsky.detectMoonMode() #803: Moon altitide: 13:52:04.0, phase 13.5%
Feb 27 06:41:38 allskypi [INFO] MainProcess allsky.reconfigureCcd() #735: Sensor temperature: 19.2
Feb 27 06:41:48 allskypi [INFO] MainProcess allsky.run() #550: Camera last ready: 14.1s
Feb 27 06:41:48 allskypi [INFO] MainProcess allsky.run() #551: Exposure state: BUSY
Feb 27 06:41:48 allskypi [INFO] MainProcess allsky.detectNight() #785: Sun altitude: -5:43:34.5
Feb 27 06:41:48 allskypi [INFO] MainProcess allsky.detectMoonMode() #803: Moon altitide: 13:53:31.6, phase 13.5%
Feb 27 06:41:48 allskypi [INFO] MainProcess allsky.reconfigureCcd() #735: Sensor temperature: 19.2
Feb 27 06:41:49 allskypi [INFO] MainProcess indi.newMessage() #181: new Message 2022-02-27T11:41:49: [INFO] Switching to 8-bit video.
Feb 27 06:41:49 allskypi [INFO] MainProcess indi.newSwitch() #168: new Switch CCD_VIDEO_FORMAT for device ZWO CCD ASI120MC-S
Feb 27 06:41:49 allskypi [INFO] MainProcess indi.newMessage() #181: new Message 2022-02-27T11:41:49: [INFO] Taking a 15 seconds frame...
Feb 27 06:41:58 allskypi [INFO] MainProcess allsky.run() #550: Camera last ready: 24.2s
Feb 27 06:41:58 allskypi [INFO] MainProcess allsky.run() #551: Exposure state: BUSY
Feb 27 06:41:58 allskypi [INFO] MainProcess allsky.detectNight() #785: Sun altitude: -5:41:00.8
Feb 27 06:41:58 allskypi [INFO] MainProcess allsky.detectMoonMode() #803: Moon altitide: 13:54:59.3, phase 13.5%
Feb 27 06:41:58 allskypi [INFO] MainProcess allsky.reconfigureCcd() #735: Sensor temperature: 20.0
Feb 27 06:42:04 allskypi [INFO] MainProcess indi.newMessage() #181: new Message 2022-02-27T11:42:04: [INFO] Exposure done, downloading image...
Feb 27 06:42:04 allskypi [INFO] MainProcess indi.newText() #175: new Text DRIVER_INFO for device ZWO CCD ASI120MC-S
Feb 27 06:42:04 allskypi [INFO] MainProcess indi.newMessage() #181: new Message 2022-02-27T11:42:04: [INFO] Download complete.
Feb 27 06:42:04 allskypi [INFO] MainProcess indi.newBLOB() #139: new BLOB CCD1
Feb 27 06:42:04 allskypi [INFO] MainProcess allsky.run() #617: Exposure received in 30.4388 s (15.4388)
```

There is apparently not an unheard of issue:  https://indilib.org/forum/ccds-dslrs/6867-guide-camera-unexpectedly-switches-to-8-bit-video-mode.html?start=0 

The interesting thing here is the camera switches to 8-bit mode after failing to take a frame.

I see I have some XML files in my .indi folder.  I am going to delete these and run again.

gilesco, 2022-02-28:
Yes, I've noted the 8-bit / 16-bit toggle.

My camera has quite a bit of noise at higher gains, so generally I have stuck with Gains less than 50, but am starting to experiment with higher gains again.

An annoying thing about the 8/16 toggle is that the darks, be they 8-bit or 16-bit become unusable.

gilesco, 2022-02-28:
Yes, but sometimes I run "allsky.py darks" and I get 8-bit darks, and other times I run "allsky.py darks" and I get 16-bit darks. All a bit random, and difficult to control, and I don't know whether when the allsky.py runs normally whether my images are going to be 8-bit or 16-bit.
It appears it is certainly a camera issue - when budget allows I'll certainly try a different model, this is an old camera that I had unused, so seemed fitting to try to build the allsky with it.

aaronwmorris, 2022-02-28:
Yes, this is definitely a camera issue combined with an INDI specific driver hack.

As far as 16 vs 8-bit, you can force the camera into a certain mode by using the indi custom options.  https://github.com/aaronwmorris/indi-allsky/wiki/INDI-custom-config#zwo-8-bit-mode (just swap the on/off for 16 bit)

I am considering adding some additional code to periodically set the custom INDI options.  This might set the camera back to the proper mode (eventually).  It might be enough for the camera to run perpetually.

gilesco, 2022-03-01:
That's an interesting find, am currently offline due to the weather, but will update and try again soon.

aaronwmorris, 2022-03-01:
Unfortunately, this did not help.  My camera changed bit depth automatically this morning.

I may play with the libasi source code to see if there is something I can do there...

aaronwmorris, 2022-03-03:
I tried some changes to the indi-asi source code, but ultimately ran into other problems with the camera.

I have had some success with setting the camera back to 16-bit mode by periodically re-applying the `INDI_CONFIG_DEFAULTS`.  I am going let this code run for a while longer before merging these updates.

aaronwmorris, 2022-03-03:
Looks like the changes are working.  I have merged PR #103 .

Here is the specific information for the ASI120:  https://github.com/aaronwmorris/indi-allsky/wiki/ASI120MC-S-Camera-Issues

aaronwmorris, 2022-03-06:
I noticed in your log messages you use telegraf.  I actually have some Grafana/Influx/Telegraf integration with indi-allsky

https://github.com/aaronwmorris/indi-allsky/wiki/Grafana-Telegraf-Influx-Integration

gilesco, 2022-03-06:
Yes, I found the telegraf json so created a dashboard. Will checkout your example dashboard!

aaronwmorris, 2022-03-25:
I have had my camera running continuously for over a week and it was still taking pictures without issue.  I think I can declare the issue with the ASI120 to be mitigated.

aaronwmorris, 2022-05-09:
Going to close this issue for now.  Let me know if there is anything else I can do.

closed 2022-05-09 by aaronwmorris