aaronwmorris/indi-allsky #90: Timelapse

ruirigel, 2022-02-21:
it seems ok to disable capture and timelapse in day mode, but in night mode it is not generating timelapse, is that normal?

aaronwmorris, 2022-02-21:
What kind of camera are you running?  You may have to post some lines from your log file so I can see what is going on.

`/var/log/indi-allsky/indi-allsky.log`

ruirigel, 2022-02-21:
qhy ccd
the ccd started capturing now, when it switched to night mode

```
eb 21 18:51:08 raspberrypi [WARNING] ImageWorker001 image.run() #285: Image uploading disabled
Feb 21 18:51:16 raspberrypi [INFO] MainProcess allsky.run() #492: Camera last ready: 10.1s
Feb 21 18:51:16 raspberrypi [INFO] MainProcess allsky.run() #493: Exposure state: OK
Feb 21 18:51:16 raspberrypi [INFO] MainProcess allsky.detectNight() #677: Sun altitude: -7:27:20.6
Feb 21 18:51:16 raspberrypi [INFO] MainProcess allsky.detectMoonMode() #695: Moon altitide: -51:16:02.7, phase 73.3%
Feb 21 18:51:26 raspberrypi [INFO] MainProcess allsky.run() #492: Camera last ready: 20.2s
Feb 21 18:51:26 raspberrypi [INFO] MainProcess allsky.run() #493: Exposure state: OK
Feb 21 18:51:26 raspberrypi [INFO] MainProcess allsky.detectNight() #677: Sun altitude: -7:29:24.0
Feb 21 18:51:26 raspberrypi [INFO] MainProcess allsky.detectMoonMode() #695: Moon altitide: -51:14:29.5, phase 73.3%
Feb 21 18:51:36 raspberrypi [INFO] MainProcess allsky.run() #492: Camera last ready: 30.3s
Feb 21 18:51:36 raspberrypi [INFO] MainProcess allsky.run() #493: Exposure state: OK
Feb 21 18:51:36 raspberrypi [INFO] MainProcess allsky.detectNight() #677: Sun altitude: -7:31:27.4
Feb 21 18:51:36 raspberrypi [INFO] MainProcess allsky.detectMoonMode() #695: Moon altitide: -51:12:56.1, phase 73.3%
Feb 21 18:51:43 raspberrypi [INFO] MainProcess allsky.shoot() #986: Taking 2.95733142 s exposure (gain 250)
Feb 21 18:51:43 raspberrypi [INFO] MainProcess allsky.run() #595: Total time since last exposure 40.0008 s
Feb 21 18:51:46 raspberrypi [INFO] MainProcess indi.newBLOB() #139: new BLOB CCD1
Feb 21 18:51:46 raspberrypi [INFO] MainProcess allsky.run() #559: Exposure received in 3.2790 s (0.3216)
Feb 21 18:51:46 raspberrypi [INFO] ImageWorker001 image.run() #132: Detected image type: Light Frame, bits: 8
Feb 21 18:51:46 raspberrypi [INFO] ImageWorker001 image.detectBitDepth() #309: Image max value: 255
Feb 21 18:51:46 raspberrypi [INFO] ImageWorker001 image.detectBitDepth() #330: Detected bit depth: 8
Feb 21 18:51:46 raspberrypi [INFO] ImageWorker001 image.run() #143: Image: 1280 x 960
Feb 21 18:51:46 raspberrypi [WARNING] ImageWorker001 image.calibrate() #554: Dark not found: ccd1 8bit 2.9573314s gain 250 bin 1
Feb 21 18:51:46 raspberrypi [INFO] ImageWorker001 sqm.calculate() #19: Exposure: 2.957331, gain: 250
Feb 21 18:51:46 raspberrypi [WARNING] ImageWorker001 sqm.getRoi() #44: Using central 20% ROI for SQM calculations
Feb 21 18:51:46 raspberrypi [INFO] ImageWorker001 sqm.calculate() #25: Raw SQM average: 35.61
Feb 21 18:51:46 raspberrypi [INFO] ImageWorker001 sqm.calculate() #30: Weighted SQM average: 167.53
Feb 21 18:51:46 raspberrypi [WARNING] ImageWorker001 image.calculate_histogram() #853: Using central ROI for ADU calculations
Feb 21 18:51:46 raspberrypi [INFO] ImageWorker001 image.calculate_histogram() #878: Greyscale mean: 44.04
Feb 21 18:51:46 raspberrypi [INFO] ImageWorker001 image.calculate_histogram() #889: Brightness average: 44.04
Feb 21 18:51:46 raspberrypi [INFO] ImageWorker001 image.calculate_histogram() #917: Current target ADU: 69.42 (59.42/79.42)
Feb 21 18:51:46 raspberrypi [INFO] ImageWorker001 image.calculate_histogram() #918: Current ADU history: (4) [62.24, 55.42, 49.09, 44.04]
Feb 21 18:51:46 raspberrypi [INFO] ImageWorker001 image.calculate_histogram() #922: ADU average: 52.70
Feb 21 18:51:46 raspberrypi [WARNING] ImageWorker001 stars.detectObjects() #60: Using central ROI for blob calculations
Feb 21 18:51:46 raspberrypi [INFO] MainProcess allsky.run() #492: Camera last ready: 0.5s
Feb 21 18:51:46 raspberrypi [INFO] MainProcess allsky.run() #493: Exposure state: OK
Feb 21 18:51:46 raspberrypi [INFO] MainProcess allsky.detectNight() #677: Sun altitude: -7:33:32.0
Feb 21 18:51:46 raspberrypi [INFO] MainProcess allsky.detectMoonMode() #695: Moon altitide: -51:11:21.8, phase 73.3%
Feb 21 18:51:47 raspberrypi [INFO] ImageWorker001 stars.detectObjects() #104: Star detection in 0.2349 s
Feb 21 18:51:47 raspberrypi [INFO] ImageWorker001 stars.detectObjects() #106: Found 11 objects
Feb 21 18:51:47 raspberrypi [INFO] ImageWorker001 image.getOrbXY() #1146: Sun hour angle: 91.57 @ 2022/2/21 18:51:47
Feb 21 18:51:47 raspberrypi [INFO] ImageWorker001 image.getOrbXY() #1146: Moon hour angle: -147.55 @ 2022/2/21 18:51:47
Feb 21 18:51:47 raspberrypi [INFO] ImageWorker001 image.getOrbXY() #1146: Sun hour angle: -89.87 @ 2022/2/22 06:46:00
Feb 21 18:51:47 raspberrypi [INFO] ImageWorker001 image.getOrbXY() #1146: Sun hour angle: -105.22 @ 2022/2/23 05:44:28
Feb 21 18:51:47 raspberrypi [INFO] ImageWorker001 image.getOrbXY() #1146: Sun hour angle: 90.30 @ 2022/2/23 18:46:27
Feb 21 18:51:47 raspberrypi [INFO] ImageWorker001 image.getOrbXY() #1146: Sun hour angle: 105.39 @ 2022/2/23 19:46:48
Feb 21 18:51:47 raspberrypi [INFO] ImageWorker001 image.run() #257: Image processed in 0.5628 s
Feb 21 18:51:47 raspberrypi [INFO] ImageWorker001 image.write_img() #469: Image filename: /var/www/html/allsky/images/20220221/night/21_18/ccd1_20220221_185146.jpg
Feb 21 18:51:47 raspberrypi [INFO] ImageWorker001 image.write_img() #482: Finished writing files
Feb 21 18:51:49 raspberrypi [WARNING] ImageWorker001 image.run() #285: Image uploading disabled
Feb 21 18:51:57 raspberrypi [INFO] MainProcess allsky.run() #492: Camera last ready: 10.5s
Feb 21 18:51:57 raspberrypi [INFO] MainProcess allsky.run() #493: Exposure state: OK
Feb 21 18:51:57 raspberrypi [INFO] MainProcess allsky.detectNight() #677: Sun altitude: -7:35:35.2
Feb 21 18:51:57 raspberrypi [INFO] MainProcess allsky.detectMoonMode() #695: Moon altitide: -51:09:48.3, phase 73.3%
```

aaronwmorris, 2022-02-21:
Everything looks good there.  Just keep in mind that [by default] "night mode" does not start until the sun is -6 degrees below the horizon.  This is calculated based on having the correct latitude, longitude, and timezone configured.

ruirigel, 2022-02-21:
> Everything looks good there. Just keep in mind that [by default] "night mode" does not start until the sun is -6 degrees below the horizon. This is calculated based on having the correct latitude, longitude, and timezone configured.

yes I am aware of that. Tonight I will leave allsky on all night to see what gives, so far nothing generated.

ruirigel, 2022-02-25:
Four days later what results I got.
When daytime capture and daytime timelapse are turned on simultaneously, day and night timelapses are generated. 
Suggestion: It would be interesting to be able to choose day or night to generate or create timelapse.
Maybe a issue: the star trail of timelapse, the sum of all the frames is only generated during the day, at night it doesn't appear.

aaronwmorris, 2022-02-25:
> Maybe a issue: the star trail of timelapse, the sum of all frames is only generated during the day, at night it doesn't appear.

Do you mean the star trails are only generated from daytime images?

ruirigel, 2022-02-25:
> > Maybe a issue: the star trail of timelapse, the sum of all frames is only generated during the day, at night it doesn't appear.
> 
> Do you mean the star trails are only generated from daytime images?

yes. please see the pic: https://i.imgur.com/MOaLJUw.png

aaronwmorris, 2022-03-25:
Closing this issue for now.  Let me know if you need anymore help.

closed 2022-03-25 by aaronwmorris