aaronwmorris/indi-allsky #92: Browser Access to USB Drive

HxPI, 2022-02-22:
Please provide support to remedy issues with 403 errors related to USB drive access. Thanks.

aaronwmorris, 2022-02-22:
If you could run the following commands and provide the output.  I think the problem is the permissions on /media/astroberry

```
ls -ld /media
ls -ld /media/astroberry
ls -ld /media/astroberry/ALLSKY
```

HxPI, 2022-02-22:
drwxr-xr-x 3 root 4096 Jun 9 2020 /media
drwxr-x—-+ 3 root 4096 Feb 22 18:31 /media/astroberry 
drwxr-xr-x 9 astroberry astroberry 65536 Dec 31 1969 /media/astroberry/ALLSKY

aaronwmorris, 2022-02-23:
I am investigating how Astroberry/Linux decides what permissions to set.

aaronwmorris, 2022-02-23:
Try adding the www-data user to the admin group and restarting nginx

```
sudo usermod -a -G admin www-data

sudo systemctl restart nginx
```

HxPI, 2022-02-23:
Received message "Usermod: group 'admin' does not exist"

HxPI, 2022-02-23:
There is an "adm" group according to groups command. Perhaps that's it?

aaronwmorris, 2022-02-23:
You are right, try adm

HxPI, 2022-02-23:
Still 403 Forbidden

aaronwmorris, 2022-02-23:
You may also try adding to the plugdev group and restart nginx.

```
sudo usermod -a -G plugdev www-data
```

HxPI, 2022-02-23:
Same 403 forbidden

aaronwmorris, 2022-02-23:
Okay, I am working remotely on my test astroberry system so I cannot plug in a USB stick right now.  Tomorrow, I can explore this in more detail when I am next to my raspberry pi.  I'll figure out the way to get this working.

HxPI, 2022-02-23:
Sounds good. Thanks.

aaronwmorris, 2022-02-23:
Well, I feel kind of silly now.  I do not think this was due to any default mount options.  The /media/astroberry folder is just a static folder.

Adding the execute bit for other on the /media/astroberry folder should be good enough to fix this.

```
sudo chmod o+x /media/astroberry
```

HxPI, 2022-02-23:
Everything looks good now except Latest tab is looking for latest.jpg and I have the format set to png. Can you take a look at that?

HxPI, 2022-02-23:
Well now I have an issue with the astroberry wifi hotspot. I am unable to connect to it. Any suggestions how to fix that? Otherwise will need to reinstall astroberry, then indi-allksy I suppose.

HxPI, 2022-02-23:
Now SQM tab shows nothing! It used to show an image but not nothing and web developer tools shows no errors, just loading next images.

HxPI, 2022-02-23:
Also the last image is way behind the current time, like 8 minutes and seems to fall further behind. Is this normal or is the RPi3 with Astroberry too slow to handle this?

HxPI, 2022-02-23:
So a reset brought the SQM back but there seems to be an issue with large sudden changes in brightness. When I turned the lights off, the charts stopped receiving updates and the chart data was alternating display data points order low/high and high/low. The SQM tab says problem loading page.

HxPI, 2022-02-23:
I'm going to say there is still more work to do with getting this to work with astroberry. I'll try the rasbian load and see if it fairs better. Otherwise, I'll just keep astroberry and fire capture going until this all gets sorted out. Best wishes and I look forward to seeing this working well once all the bugs are sorted out. Thanks again.

aaronwmorris, 2022-02-23:
I have merged PR #95 so that the latest image is correctly loaded based on the image type selected in the config.

Running on Astroberry with a Rasberry PI 3 is likely going to be problematic due to the constrained memory.  You could get around some of this by disabling the GUI, but there are still a lot of background services running on Astroberry.

Building the timelapse video with ffmpeg is also a bit of a memory hog.  Building a 1K video usually requires about 512MB of available memory.  Even my Raspberry PI 3 cameras occasionally run out of memory doing this on vanilla Raspbian installs.

It really is better to run on a plain Raspbian (non-desktop) minimal install.

Disable GUI
```
sudo systemctl set-default multi-user.target
# reboot
```

Re-enable GUI
```
sudo systemctl set-default graphical.target
# reboot
```

aaronwmorris, 2022-02-23:
I really appreciate the feedback.  All of your suggestions and findings have been very helpful.

HxPI, 2022-02-23:
No problem. I’m loading Rasbian 10 32bit and will load indi-allsky to see if it works out the box.

aaronwmorris, 2022-02-24:
I am hunting down any other bugs there may be with the PNG setting.  I merged #96 so that star trails and keograms are generated with the correct image type.

HxPI, 2022-02-24:
With Raspberry Pi OS installed, how is the hotspot setup? I don’t see a hotspot and cannot connect to indi-allsky without RPi hotspot.

aaronwmorris, 2022-02-24:
You just know how to find gaps.  :-)

I have not yet had a need to setup a hotspot, but I can work on that next.

HxPI, 2022-02-24:
Yeah hopefully that’s the last one. Wi-Fi access will allow VNC and Web interface access.

aaronwmorris, 2022-03-11:
I have completed my hotspot setup script if you would like to try it.  It is at `misc/setup_hotspot`.

I have tested it on Raspbian 10 and 11.

HxPI, 2022-03-12:
I am getting an error when running ./setup.sh:

###############################################
### Welcome to the indi-allsky setup script ###
###############################################

Distribution: Raspbian
Release: 11
Arch: armv7l

INDI_DRIVER_PATH: /usr/bin
INDISERVER_SERVICE_NAME: indiserver
ALLSKY_SERVICE_NAME: indi-allsky
GUNICORN_SERVICE_NAME: gunicorn-indi-allsky
ALLSKY_ETC: /etc/indi-allsky
HTDOCS_FOLDER: /var/www/html/allsky
DB_FOLDER: /var/lib/indi-allsky
INSTALL_INDI: true

Setup proceeding in 10 seconds... (control-c to cancel)

Fixing git checkout ownership
**** Installing packages... ****

There are not prebuilt indi packages for this distribution
Please run ./build_indi.sh before running setup.sh

Unable to continue with indi-allsky setup. Please advise.

HxPI, 2022-03-12:
I’m using another script which is building the indi packages. It also setup a hotspot so that may be good enough. Install and testing in progress.

HxPI, 2022-03-12:
indi packages are built. Now getting this error when running ./setup.sh

E: Unable to locate package indi-full
E: Unable to locate package indi-rpicam

HxPI, 2022-03-12:
Reinstalled Raspbian 10 and indi-allsky. After running hotspot setup, my iPad will not connect to the hotspot. It says the security is weak and just sits there spinning. How do I access the SSID settings on Raspbian? Is there any way to set the security to at least PSK2?

HxPI, 2022-03-12:
Was able to download Network Manager and verify hotspot settings. Able to connect to hotspot now, for some unknown reason it just started working. There is an issue with images not saving in PNG. I have FITs   saved because RAW was also selected. Latest image is in PNG but no other images. Of course all the tabs that depend on the PNG are not working either.

HxPI, 2022-03-12:
Changed to JPEG no difference. The latest image seems to be working though and does save in the selected format and is kept up to date. However, there are no JPEG or PNG images being saved to the applicable folder.

aaronwmorris, 2022-03-12:
> indi packages are built. Now getting this error when running ./setup.sh
> 
> E: Unable to locate package indi-full
> E: Unable to locate package indi-rpicam
> ...
> Not sure what is going on. Install needs to be easier than this!

I think what happened here is your system was detected as Debian 11 instead of Raspbian 11 after you ran the build script for INDI.  There was a bit of code left over in the Debian 11 section that tried to install indi packages that should not have been there.  I fixed that.

Something in the package dependencies used to compile INDI is causing the system to suddenly report as Debian 11.  I have seen this before, but I do not know what to do to fix this.  The Debian 11 detection also causes issues with the python modules since there are no pre-built python wheels, the python modules with C dependencies have to be built from source.

Trying to "do the right thing" is a challenge when supporting multiple distributions and multiple architectures with custom software that has to be built from source.

> Changed to JPEG no difference. The latest image seems to be working though and does save in the selected format and is kept up to date. However, there are no JPEG or PNG images being saved to the applicable folder. I already set the execute permission on /media/pi. However, that folder won't let me change the settings to allow anyone to access. Probably not the issue if latest image is being saved there.

I am not sure what is happening with your images.  I am currently chasing down a bug in the PNG code where it is taking up to 30 seconds to compress a PNG image.  You might be hitting this issue.  Hopefully the indi-allsky log can show you error messages.

https://github.com/aaronwmorris/indi-allsky/wiki/PNG-Performance

However, my test code at `testing/imageCompressTest.py` always shows sub-second compression times for PNG.

In the meantime, I would recommend sticking with JPEG.

HxPI, 2022-03-12:
My RPi3 keeps crashing so I think it’s not going to work for this. I ordered a RPi4 4GB and will try that. What OS do you recommend for running indi-allsky on this?

aaronwmorris, 2022-03-12:
Raspbian 10 (Legacy) still has the easiest experience.

All of my permanent camera installations are RPi3s and I have trouble with them crashing from time to time.  I implemented a watchdog so they automatically reboot if they lock up.  Ever since implementing the watchdog I have not had to manually intervene.  https://github.com/aaronwmorris/indi-allsky/wiki/Raspberry-Pi-Watchdog

I will go investigate getting the automount support setup for the USB drives.  I am also still looking into the PNG issue.

HxPI, 2022-03-13:
Can indi-allsky be setup on Ubuntu Mate?

aaronwmorris, 2022-03-13:
I will go download that and see if there are any necessary modifications to the setup script.

aaronwmorris, 2022-03-14:
I have merged PR #118 for the Ubuntu MATE RPi support.  You will have to run the `./misc/build_indi.sh` script to build INDI from source since there are no binaries available.

If it is Ubuntu MATE on x86_64 (intel) then that should work with the existing setup.

HxPI, 2022-03-15:
For dark frames, are they created for various temperature ranges? For example, operating in 32F vs 80F will be a large difference in camera temp and resulting dark frame. Seems there should be various dark frames depending on camera temperature. Perhaps every 10 degrees is sufficient?

gilesco, 2022-03-16:
Not too many people have a temperature controlled camera as their allsky camera, so capturing dark frames at various temperatures would be difficult to achieve.

HxPI, 2022-03-16:
Agreed. One way would be to put the camera in the freezer then take darks and monitor camera temp. Collect dark frames in groups as temperature rise, say every 10 degrees. Perhaps enough frames would be capture in each group for the entire range of temperatures to make a reasonable sized dark frame library.

aaronwmorris, 2022-03-16:
My current theory behind dark frames is time based (as opposed to temperature based).  Exposures are taken at 5s intervals at the day, night, and moonmode gains.  Temperature is also recorded from the camera, although, I do not currently utilize this data point (not all cameras have a temperature sensor).  All darks are stored as the original FITS files from INDI.

If you have captured darks using indi-allsky, you can view your dark frame info in a hidden view:  https://hostname/indi-allsky/darks  The files are normally stored in /var/www/html/allsky/images/darks

The use of the SQLite database makes this easy as I store all of the metadata in the `darkframe` table.  Dark frame model:  https://github.com/aaronwmorris/indi-allsky/blob/master/indi_allsky/flask/models.py#L86  I give the files descriptive names, but the file names do not matter since I query the metadata from the database.  SQL is much more efficient than searching file names.

I can investigate adding a manual dark frame upload.  I can also look at updating the dark frame matching algorithm to include temperature in the matching.

HxPI, 2022-03-16:
Normally, darks are taken to match temperature AND exposure time per given gain. I normally expose for 30 sec at gain 250. Depending on time of year, the temperature can vary. I’m not sure 5 second darks are going to properly calibrate with my given exposure settings. I suppose worse case I can just save raw FITs and make my own dark library. Anything you can implement to assist with this is greatly appreciated. Thanks.

aaronwmorris, 2022-03-16:
Sorry, I was not clear....  The darks are taken in 5 second increments, 1, 5, 10, 15, 20, etc until the max exposure time.  The dark is matched at the next highest exposure found.  ie a 21 second exposure will have the 25 second dark applied to it.

I have just not found temperature matched darks to be critical for all sky images.  I know they are very important with deep sky images, but the signal is so low with deep sky, temperature matched darks are necessary to extract as much signal as possible.  All sky images are usually fully exposed and are rarely stretched.

But... I am not here to tell people how to use their cameras.  :-)   I am looking at how to implement darks with temperature matching.

HxPI, 2022-03-16:
I see what you mean now. You are correct all sky images should not be linear like deep sky. My primary concern is amp glow and hot pixels. If your method captures those artifacts then that should work. I’ll take a look at this when I get my new RPi4 up and running. Thanks.

aaronwmorris, 2022-03-25:
I have not really found the issue with my long compression times with PNG.  However, if you reduce the compression level for PNG to 5 or 6 the compression times are reasonable.  This does not cause a decrease in quality since the compression is lossless, but the files are slightly larger.

I have also completed a new script to setup automounting of USB drives when there is no GUI running.  The setup script is at `./misc/setup_usb_automount.sh`.  The automount uses the same subsystem as Astroberry, so the mount is setup the same exact way.  I will eventually get around to writing a wiki page for it.

I am still working on the temperature matched darks...

HxPI, 2022-03-25:
I reinstalled with Raspbian 10 on RPi4. A couple issues. First the wifi isn't working. the hotspot is active but I am unable to connect. I wanted Network Manager installed so I installed it but not sure if that would affect wifi hotspot. Also, can you add Network Manager to the install script? It's the only way I know to change the wifi password and other network management tasks. So currently unable to test with wifi.

HxPI, 2022-03-25:
Reinstalled everything default and so far it’s working as it should. I’ll let it run for a while and see what it does for day and night Timelapse’s. Next step will be run usb script and change storage to usb.

HxPI, 2022-03-25:
Can the SQM sky brightness – mags/arcsec² be added to right side of jSQM chart? I have no idea what the left side number is but sky brightness is easier to understand. Thanks

http://unihedron.com/projects/darksky/magconv.php

HxPI, 2022-03-25:
Another suggestion is have gain ramp up/down during th transition from day/night and night/day and also when changing to/from moon gain. That way there isn’t an abrupt exposure change. This is similar to holy grail exposure ramping used in photography.

HxPI, 2022-03-25:
Another question is TIFF compression config setting but TIFF is not an option for file format. Can this be made an available file format? Also, is there a way to set 16bit mode for color camera capture debayered image files?

HxPI, 2022-03-26:
Trying to do dark frames. Getting error when activating python virtual environment: bash

virtualenv/indi-allsky/bin/activate: No such file or directory

Typed exactly as stated in Manual operation.

aaronwmorris, 2022-03-26:
Network manager is what is used in the script to setup the hotspot.  I have tested multiple times with the hotspot setup and it always works.  I see you are using an iPad, I will have to test with an iPad since I was testing with an Android phone.  I will run more tests.

I have not enabled the ability to configure TIFF files in the config due to TIFF files not being supported by all browsers.  indi-allsky can produce TIFFs, but they will never be displayed [in the browser].  It would probably be best to create PNG files and perform a mass conversion to TIFF.

As for the sky brightness.... this is considerably more difficult than I originally envisioned.  In order to use a camera to detect sky brightness, you have to account for the QE and spectrum detected by the camera.  If your camera has a UV/IR cut filter, then most of the light detected is visual.  If you do not, then some of the brightness detected is outside the visual spectrum.  It may take someone with a lot more experience in this area to come up with a way to do the conversion.

For the darks, you may have to check the log to make sure the darks are matching.  There will be a message about which dark was matched to the frame.  I have not had an issue with persistent noise (hot pixels, etc) in longer exposures (+5s) but I have seen issues with shorter exposures.  Some cameras return inconsistent noise at exposures shorter than 1-2 seconds.

HxPI, 2022-03-26:
I tried an Android device also and wasn’t able to connect. I could have introduced errors when trying to setup network manager separately.

Since the reinstall and keeping everything default, there have been no major issues. I will try the hotspot setup and see if it works with the default password. If it does, then all I really need to do is change the password. Can that be done by editing the script?

aaronwmorris, 2022-03-26:
Yes, you can just edit the script with the new password.

HxPI, 2022-03-26:
Ok. Since network manager is being used, is there a way to access and/or install a gui for it?

aaronwmorris, 2022-03-26:
I can pretty easily add a desktop environment that will only be started if you connect via remote desktop (RDP).  Meaning it would not run in the console via HDMI.  This could have a network manager widget to edit the settings.

HxPI, 2022-03-26:
Well I ran the hotspot script and now I don't see the IndiAllsky SSID on any device. I'll try installing again.

aaronwmorris, 2022-03-26:
Have you tried rebooting to see if the SSID shows up?

[77 more replies]

closed 2022-05-01 by HxPI