aaronwmorris/indi-allsky #112: setup failed

tlkw79, 2022-03-13:
Hi,

I tried to setup indi-allsky but failed.

Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libcurl4-gnutls-dev : Depends: libcurl3-gnutls (= 7.68.0-1ubuntu2.7) but 7.68.0-1ubuntu2.8 is to be installed
 libgnutls28-dev : Depends: libidn2-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

aaronwmorris, 2022-03-13:
What distribution are you using?

tlkw79, 2022-03-14:
NAME="Ubuntu"
VERSION="20.04.4 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.4 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal

aaronwmorris, 2022-03-14:
You may try to do a full package upgrade to see if it will fix the dependency issue.

```
sudo apt-get update
sudo apt-get dist-upgrade
```

If you are running Ubuntu MATE on a Raspberry PI, I just added the arm support today.  #118

aaronwmorris, 2022-04-10:
I am closing this for now.  Let me know if there is anything else I can do for you.

closed 2022-04-10 by aaronwmorris