aaronwmorris/indi-allsky #153: Temperature

bsbachert, 2022-04-09:
Is it possible to add Fahrenheit as a temperature option, or an option to display both. 

Thanks, love the software.

Brian

aaronwmorris, 2022-04-09:
Absolutely!  I will work on it.

bsbachert, 2022-04-09:
Thank you Aaron.

aaronwmorris, 2022-04-10:
I have merged #155 to support displaying temperature in Fahrenheit and Kelvin (because why not :-) )

bsbachert, 2022-04-10:
Because you are awesome I’m thinking. 😃

Thank you so much. I’m Celsius illiterate most days.

Brian

aaronwmorris, 2022-04-27:
Closing this for now.  Let me know if there is anything else I can do for you.

bsbachert, 2022-04-28:
Well, now that you mention it. 😋 I like to do color for day, and gray at
night. Is there a way to incorporate that into settings? Like a gray in
night mode setting? Color in day mode setting?

Thanks for all you do. Do you have a donate link?

Brian

aaronwmorris, 2022-04-28:
For something like theming, I would probably need some help in that area.  I am just not a very good front-end developer.  :-)

bsbachert, 2022-04-28:
Sorry, I was meaning camera settings. I use gray scale in night mode, then
shut it off for day mode time lapses. It would be cool if it could do it
automatically.

aaronwmorris, 2022-04-28:
Ah, I see.  Let me think about that.

bsbachert, 2022-04-28:
Ok, thank you.

aaronwmorris, 2022-04-29:
Merged #183 for separate night and day grayscale options.  I renamed the config variable, so you will have to setup the option again in the configuration page.

bsbachert, 2022-04-29:
Updated the software, and switched on gray scale at night. Waited until
night mode started. I can confirm it works. Camera switched to gray scale
instantly when night mode began.

Thank you so much!!!!

aaronwmorris, 2022-05-13:
Closing this for now.  Let me know if there is anything else I can do!

bsbachert, 2022-05-15:
Hi Aaron,

I see you added a feature where we could add our own data via a txt file.
Would it be possible to export the SQM data to a txt file, so I could show
that data on my website?

Thanks again for you time. Where can I donate to you, and buy you a coffee
or a beer?

Thank you, for all your help.

Brian

aaronwmorris, 2022-05-15:
Sure, all of the data is stored in the database.

bsbachert, 2022-05-15:
Yes. That would show the number of stars, max, avg., and min jsqm correct?

bsbachert, 2022-05-15:
[image]

bsbachert, 2022-05-15:
This data..

aaronwmorris, 2022-05-15:
Cool... I will add this to my todo list

bsbachert, 2022-05-15:
Thanks again.

aaronwmorris, 2022-05-25:
I have merged #202 and #203 .  This functionality will upload image metadata as JSON to the remote server.

bsbachert, 2022-05-26:
Thank you Aaron. Now I can import the data into my webpage. Yay, so happy.

Thanks again.
Brian

bsbachert, 2022-05-29:
Hello Aaron,

Up and running. I am downloading it from my website every 10 seconds 
using Watch, and a small wget script, then adding the json file to the 
add text option in the config dialog. Works perfect.

Thanks again.

Brian Bachert

aaronwmorris, 2022-06-27:
Closing this issue for now.  Let me know if there is anything else I can do.

closed 2022-06-27 by aaronwmorris

bsbachert, 2022-10-11:
Thanks Aaron.