aaronwmorris/indi-allsky #156: Gain set to max when changing from Day to Night or Day to Moon

herseyc, 2022-04-10:
Great work on this project, but I am having a weird issue (as discussed on CloudyNights). I have the gain configured to 110 for day, 300 for night, and 200 for moon.  When it changes from day to night or day to moon it is setting the gain to 720 (which is the max for the SV305).

Apr  9 19:59:43 allskypi [WARNING] MainProcess allsky.reconfigureCcd() #767: Change to night (moon mode)
Apr  9 19:59:43 allskypi [WARNING] MainProcess indi.setCcdGain() #412: Setting CCD gain to 720
Apr  9 19:59:43 allskypi [INFO] MainProcess indi.configureDevice() #332: Setting property CCD_GAIN
Apr  9 19:59:43 allskypi [INFO] MainProcess indi.newMessage() #179: new Message 2022-04-09T23:59:43: [INFO] Camera control 0 to 720
Apr  9 19:59:44 allskypi [WARNING] MainProcess indi.setCcdBinning() #474: Setting CCD binning to (1, 1)
Apr  9 19:59:44 allskypi [INFO] MainProcess indi.configureDevice() #332: Setting property CCD_BINNING
Apr  9 19:59:44 allskypi [INFO] MainProcess indi.newMessage() #179: new Message 2022-04-09T23:59:44: [INFO] Binning changed
Apr  9 19:59:58 allskypi [INFO] MainProcess allsky.shoot() #1160: Taking 1.32142818 s exposure (gain 720)

I am also seeing this in the logs:

Apr  9 08:23:33 allskypi [ERROR] MainProcess allsky.sighup_handler() #183: CCD night gain above maximum, changing to 720
Apr  9 08:23:33 allskypi [ERROR] MainProcess allsky.sighup_handler() #190: CCD moon mode gain above maximum, changing to 720
Apr  9 08:23:33 allskypi [ERROR] MainProcess allsky.sighup_handler() #197: CCD day gain above maximum, changing to 720

I am not sure if HUPing the indi-allsky fixes it. Tonight I stopped indi-allsky, stopped indi server, then started indi server, and then started indi-allsky.  If the issue happens again tomorrow night I will try to just HUP indi-allsky and report back if that fixes it.

I can submit a full copy of the logs if needed, just let me know.

Thanks
Hersey

aaronwmorris, 2022-04-10:
I have seen this in my own logs, but I have yet tracked down the cause.  It only seems to happen when the process is HUPed.

herseyc, 2022-04-10:
So it happened again last night when it changed from Moon mode to Night mode after the moon set.  I'll reboot this morning and won't make any change to see if it repeats tonight.

Thanks
Hersey

aaronwmorris, 2022-04-10:
I found a bug #159 in the code related to HUP that caused the gain to be increased incorrectly.

herseyc, 2022-04-11:
Updated yesterday afternoon.  No issues after the update.

Thanks
Hersey

herseyc, 2022-04-13:
Have not had any additional issues with the gain changing to max.  Closing this out.  Thanks!

closed 2022-04-13 by herseyc