aaronwmorris/indi-allsky #172: 16bit TIFF RGB Format Save Option

HxPI, 2022-04-24:
Would it be possible to make 16bit TIFF format a separate selectable save option, similar to RAW? Since TIFFs don't work in some browsers, it would not be idea as a primary format like JPG or PNG. However it is useful as a proxy file format. In my particular case, I'd like to save as PNG for normal views and also save as TIFF to process with time-lapse software. Being able to save the full 12bit color from my ASI camera will allow high quality time-lapse. Also, if TIFF can have dark frame applied that will speed up workflow even more. Thanks.

aaronwmorris, 2022-04-25:
This does not sound unreasonable.  I have some ideas and I will work on this next.

aaronwmorris, 2022-04-26:
Okay, I have merged #175 .  This allows saving the original 16bit data to TIFF and PNG formats.  Make sure you have plenty of space.  A 1920x1080 image will be 4-5MB.

Also, FYI, I renamed the option to save fits in #174

HxPI, 2022-04-26:
Updated to the latest version. When selecting TIFF raw output, no images are saved to the folder for normal or raw format. When setting raw format to Disabled, then normal images will save to the folder. System works fine when PNG raw format is selected.

aaronwmorris, 2022-04-26:
I found a typo in the code related to the raw TIFF files.  #177

HxPI, 2022-04-26:
Ok. Are dark frames applied to the raw png/tiff files? If not, can that be added as an option?

aaronwmorris, 2022-04-26:
Yes, the only processing I am doing on the raw files is dark frame subtraction and debayering for CFA sensors.

HxPI, 2022-04-29:
After comparing the 8bit PNG to 16bit TIFF, I see absolutely no difference between the two. That seems odd there should be at least some more color detail in the 16bit TIFF if it is saving the full 12bit data from the camera. I'm using an ASI290MC. Can you check to verify the raw TIFF/PNG formats are saving the full bit depth from the camera. Thanks.

aaronwmorris, 2022-04-29:
I generated some new files from my ASI385 (I do not have a 290MC) and all indications are the data is represented as 16-bits.

The camera is supposed to start in a 16-bit mode, but there is a chance the camera might be in 8-bit mode.  I would explicitly add the INDI custom options to ensure the camera is running in 16/12 bit mode.

```
{
    "PROPERTIES" : {},
    "SWITCHES" : {
        "CCD_VIDEO_FORMAT" : {
            "off"  : ["ASI_IMG_RAW8"],
            "on" : ["ASI_IMG_RAW16"]
        }
    }
} 
```

HxPI, 2022-04-30:
Still see no difference. The raw TIFF is literally a copy of the normal 8bit PNG. Do you see any noticeable difference between the 8bit file and 16bit file from your ASI385?

HxPI, 2022-04-30:
> I generated some new files from my ASI385 (I do not have a 290MC) and all indications are the data is represented as 16-bits.
> 
> The camera is supposed to start in a 16-bit mode, but there is a chance the camera might be in 8-bit mode. I would explicitly add the INDI custom options to ensure the camera is running in 16/12 bit mode.
> 
> You will have to restart indi-allsky as a HUP is not sufficient.
> 
> ```
> {
>     "PROPERTIES" : {},
>     "SWITCHES" : {
>         "CCD_VIDEO_FORMAT" : {
>             "off"  : ["ASI_IMG_RAW8"],
>             "on" : ["ASI_IMG_RAW16"]
>         }
>     }
> } 
> ```

aaronwmorris, 2022-04-30:
Yes, that is still correct.  The python json module just reformats the text.

My raw TIFF and PNG files are reported as 16-bit unsigned integer format.  The files are also much larger than their 8-bit counterparts.

I do not think you will be able to see a visual difference between a 12-bit file and a 8-bit file.  Most monitors these days are only capable of displaying 8-bit (24-bit RGB) color.  Even the new Apple Pro Display only has a 10-bit color gamut.

HxPI, 2022-04-30:
Well I’m going to do some more testing. I believe there should be at least a data difference between 8bit and 12bit. Im after better color gradients in the capture images. If it’s not possible, I have a Plan B. One must always have a backup plan! Thanks again.

aaronwmorris, 2022-04-30:
> Is there an ASI_IMG_RGB48 option?

ZWO color options for color cameras are usually:

ASI_IMG_RAW8  (8 bit raw)
ASI_IMG_RAW16  (16 bit raw)
ASI_IMG_RGB24  (24 bit RGB/8 bit per channel debayered)
ASI_IMG_Y8 (8 bit color YUV *I think*)

Monochrome cameras only have ASI_IMG_RAW8 and ASI_IMG_RAW16

HxPI, 2022-04-30:
> > Is there an ASI_IMG_RGB48 option?
> 
> ZWO color options for color cameras are usually:
> 
> ASI_IMG_RAW8 (8 bit raw) ASI_IMG_RAW16 (16 bit raw) ASI_IMG_RGB24 (24 bit RGB/8 bit per channel debayered) ASI_IMG_Y8 (8 bit color YUV _I think_)
> 
> Monochrome cameras only have ASI_IMG_RAW8 and ASI_IMG_RAW16

All of this is confusing! I’ll have to do some more research and get to the bottom of what I can get out of my ASI290MC. I’ll post my findings.

HxPI, 2022-04-30:
> > > Is there an ASI_IMG_RGB48 option?
> > 
> > 
> > ZWO color options for color cameras are usually:
> > ASI_IMG_RAW8 (8 bit raw)
> > ASI_IMG_RAW16 (16 bit raw) ASI_IMG_RGB24 (24 bit RGB/8 bit per channel debayered) ASI_IMG_Y8 (8 bit color YUV _I think_)
> > Monochrome cameras only have ASI_IMG_RAW8 and ASI_IMG_RAW16
> 
> All of this is confusing! I’ll have to do some more research between debayered RAW16 and RGB24 with the ASI290MC. I’ll post my findings.

aaronwmorris, 2022-04-30:
> > > Is there an ASI_IMG_RGB48 option?
> > 
> > 
> > ZWO color options for color cameras are usually:
> > ASI_IMG_RAW8 (8 bit raw)
> > ASI_IMG_RAW16 (16 bit raw)
> > ASI_IMG_RGB24 (24 bit RGB/8 bit per channel debayered) ASI_IMG_Y8 (8 bit color YUV _I think_)
> > Monochrome cameras only have ASI_IMG_RAW8 and ASI_IMG_RAW16
> 
> All of this is confusing! I’ll have to do some more research between debayered RAW16 and RGB24 with the ASI290MC. There may be some issues with reduced dynamic range at higher gains. I’ll post my findings.

I had not considered the gain aspect.  You are correct!  At gain 250 the dynamic range drops to about 9.5 stops.  At 300, it is about 9 stops.

HxPI, 2022-04-30:
Yeah I saw no difference between debayered RAW16 and RGB24 using Pixinsight. I guess RAW16 will only be useful for day images, which is fine. I think this thread is done. Thanks.

closed 2022-05-01 by HxPI