aaronwmorris/indi-allsky #176: On Demand Timelapse, Star Trail, and Keogram Generation

HxPI, 2022-04-26:
Would it be possible to add an option in System page to manually invoke Time-lapse, Star Trail, and Keogram generation? It seems the only time these are generated is at sunrise and sunset. However, sometimes the device needs to be turned off before these times and the files are just sitting in the folder. An example is when I use the device during outreach events that only lasts 4-5 hours or an imaging session where I pack up before sunrise. Rather than leave the device on and wait for sunrise/sunset to process the files, an option to manually activate for the current files would be useful. 

So if the process is manually invoked, whatever files are available for a given day/night would be processed. It should be possible that multiple products could be produced manual generation is invoked multiple times, using whatever files are available for the given day/night time period. Of course, the normal function of processing all day/night images at sunrise/sunset should still function without interaction.

aaronwmorris, 2022-04-26:
Unfortunately, this functionality is not easy to implement.

The primary challenge is these are long running tasks, usually 10 to 30 minutes in duration.  Processes in the web server generally have a time limit of 3 minutes (180 seconds) and this is usually limited by the browser (not the server).  Once a request takes longer than 180 seconds, the connection is killed and the server process terminated.

The way around this limit is implementing an asynchronous task queue on the server.  The browser sends a request to the server, the server adds the task to the queue, and an independent process on the server picks up the task and executes it.

I would like to get to a point where I can do this, but it could be a while.

aaronwmorris, 2022-04-30:
I am starting to make some moves toward the goal of being able to submit tasks from the web server process.  I have to change the entire inter-process communication architecture to achieve this.

My first objective was to ensure that the sqlite database is going to be robust enough to handle the increase in queries.  I am using the database a lot more than even I originally envisioned.  Sqlite is not exactly a high concurrency database.  In #179 I moved to using the WAL journal which should dramatically improve the database performance.

aaronwmorris, 2022-05-05:
I am making some progress on my task_queue branch.  Using the database for task management seems stable enough.  I am letting one of my virtual cameras run to shake things down.

HxPI, 2022-05-05:
Great. I’m wondering how much overhead does this add to power and performance? Hopefully it doesn’t take o much away from the already great performance and battery life.

aaronwmorris, 2022-05-06:
I doubt there will be a significant impact.  The changes to support this amount to reading and writing a few kilobytes of data to the database every 15-30 seconds.  There is probably more disk I/O generated from logging than this change.

aaronwmorris, 2022-05-16:
This has been the impact so far.  About twice as much disk i/o.  Not too bad.

[image]

aaronwmorris, 2022-06-18:
I have merged #223 which includes a new web interface for generating timelapses, keograms, and star trails on demand.

HxPI, 2022-06-18:
Getting an error during setup.sh update. Says release files are not valid yet. Is this normal?

aaronwmorris, 2022-06-18:
Yes, this just means the time is off.

You can re-enable time sync with the following
```
sudo timedatectl set-ntp true
```

HxPI, 2022-06-18:
Setup worked after NTP was enabled. Recommend adding this to the Indi-allsky update instructions. Thanks.

HxPI, 2022-06-18:
There are two confirm checkboxes for system reboot and shutdown system. Is there only supposed to be one and/or which one is supposed to be selected to confirm?

aaronwmorris, 2022-06-18:
Both need to be checked to flush.

HxPI, 2022-06-18:
Oh ok. Thanks.

HxPI, 2022-06-19:
Tried the update but there was an issue. Day images saved but as soon as night began imaging stopped. There was only one image saved for night and no further images were taken, including the latest image. Something has caused the system to stop taking night images when going from day to night. Please look into this. Thanks.

HxPI, 2022-06-28:
Still need to verify this feature. Will report back the results.

StarGeezerPhil, 2022-07-23:
Great work and pretty excited looking at all your recent changes. @aaronwmorris just a quick question as I've had confusing results with regenerating star trails.

I'm trying to use the regenerate feature alongside a night of reasonable captures to tune my ADU / max star trail ADU / star mask threshold values to get the best results for my sky/light pollution (bortle 8-9).

- Does the whole system need to be rebooted for the regenerate feature to use new config?
- Does the regenerate feature use current config settings or the settings when the files were captured?

I've been regenerating some recently, but don't seem to be getting different results.

Are you also able to explain the star mask threshold value and how it affects the image?

I presume that ST max ADU simply excludes any images brighter than the defined target ADU.

aaronwmorris, 2022-07-23:
> * Does the whole system need to be rebooted for the regenerate feature to use new config?
> 
> * Does the regenerate feature use current config settings or the settings when the files were captured?

The whole system does not have to be rebooted, but you do have to at least HUP the service so the config file is re-read and applied.

> I presume that ST max ADU simply excludes any images brighter than the defined target ADU.

That is correct.  Images where the average ADU exceeds the max ADU setting are excluded.  I made a change recently to only calculate the average ADU using the central 2/3 of the image.  Many people have trees and these will bring down the overall average ADU (assuming they are dark).

I have also added the ability to build a custom mask so that you can explicitly include/exclude areas of the image for the calculation.  https://github.com/aaronwmorris/indi-allsky/wiki/Detection-Masks

> Are you also able to explain the star mask threshold value and how it affects the image?

There were some cases I found where just using the average ADU value was not sufficient to exclude "brighter" images from the star trails.  The `Star Trails Mask Threshold` is an ADU value.  If the total number of individual pixels above this ADU value exceed `Star Trails Pixel Threshold` (default 1%) then the image is excluded.

StarGeezerPhil, 2022-07-23:
Aaron, as always - thank you for the response and explanation.

Brill - clarifies a lot of my queries! Really appreciate that.

So to check/confirm: 
- Change of config - HUP required to apply settings (not familiar with "HUP" - is this like reloading the service w/out restart)
- Star trail calculation - doesn't use "SQM / Star Region of Interest", but uses the central 2/3 of the image - OR - a star detection mask (if configured) - is that right? I guess it's best to also avoid highly light polluted regions in such a mask (I have 3 cities on my horizon - with a lot of bright glow above each)
- Star Mask is a pixel count trigger from the (entire or masked?) image at which point to reject images, used alongside average ADU value. (Higher value will require more bright pixels to trigger).

Thanks again!

aaronwmorris, 2022-07-23:
> Change of config - HUP required to apply settings (not familiar with "HUP" - is this like reloading the service w/out restart)

HUP is an old unix term relating to serial terminals, but these days it is an OS signal that is commonly used in software to reconfigure itself without performing a full process stop and start.  A graceful restart.

> Star trail calculation - doesn't use "SQM / Star Region of Interest", but uses the central 2/3 of the image - OR - a star detection mask (if configured) - is that right? I guess it's best to also [...]

I just submitted #274 so that the SQM Region of Interest is used.  (The central 2/3 of the image is still used if SQM_ROI is not defined).  The detection mask will be used in its place, if defined.

You can definitely play with it to see if excluding the light domes makes things better.  This is one of the more difficult areas I deal with:  picking sane defaults.  I tried to make everything as "turn key" as possible, but picking the correct images for star trails is a very subjective.

> Star Mask is a pixel count trigger from the (entire or masked?) image at which point to reject images, used alongside average ADU value. (Higher value will require more bright pixels to trigger).

Confirmed.

StarGeezerPhil, 2022-07-23:
Brill - thanks again. Sorry - couple more.

Just re: Star Trails Mask Threshold - mine limits to a max value of 255, and was this from the entire image or the ADU ROI/star detection mask? Also what's the Pixel Threshold?

aaronwmorris, 2022-07-23:
> Just re: Star Trails Mask Threshold - mine limits to a max value of 255, and was this from the entire image or the ADU ROI/star detection mask?

Its only from the detection area.  The max value is 255 because the calculation is always done on the downscaled, 8-bit data.

> Also what's the Pixel Threshold?

The Pixel Threshold is the percentage of pixels in the detection region that exceed the Mask Threshold.

StarGeezerPhil, 2022-07-23:
Thanks again - apologies I'd misread your earlier post, but get it now 👍

How do we set a mask image in config/command line and does it need to be a specific file format?

aaronwmorris, 2022-07-23:
> How do we set a mask image in config/command line and does it need to be a specific file format?

The option in the config form is `Detection Mask`.  You just specify the path to the mask file on the filesystem.  The file needs to be in PNG format.

aaronwmorris, 2022-07-24:
I am planning on writing a guide on how to make a mask file.  It is just going to require a lot of screenshots.

StarGeezerPhil, 2022-07-24:
Brill, thanks Aaron, but not an option I have on mine - have I missed another update? 🤣

No trouble making the mask, just how to set it. I'll check for updates.

aaronwmorris, 2022-07-24:
Yes, this is a relatively new feature, so you will have to update to the latest code.

aaronwmorris, 2022-08-21:
I am going to close this issue for now.  Let me know if I can help with anything else.

StarGeezerPhil, 2022-08-26:
@aaronwmorris I'm trying to generate timelapse videos to optimise my scaling and bitrate settings for the videos.

Can I presume that once the image is created it is too late for the "resize" option to apply to the video creation?

Would it be possible for this resize (or a second "Video resolution" setting) to be applied during the timelapse generation? I'd rather have high resolution images, but (where necessary) reduce the video resolution only.

I'm getting failures generating the videos.

aaronwmorris, 2022-08-26:
I will investigate options for ffmpeg to change the video timelapse generation.

StarGeezerPhil, 2022-08-29:
Hi Aaron, sorry - you might want to close this particular issue. I've commented on the IMX378 thread separately, but trying to find workarounds myself for the max resolution limitation.

Commandline for ffmped is in the format "-s [X]x[Y]" for absolute, or I believe you can use "-vf scale=[X]:-1" to maintain ratio.

Would it be possible in the script to check the resolution and IF(X)>4,096 OR (Y)>2,304 add a scale factor to the ffmpeg command-line?

aaronwmorris, 2022-08-29:
I'll go ahead and close this for now.  We can talk about the resizing in the other thread.

closed 2022-08-29 by aaronwmorris

StarGeezerPhil, 2022-08-31:
@aaronwmorris really sorry to ask another question on this (I'm sure you've answered it somewhere before, but gh search isn't great for comments within issues - could you open a discussion area here as I feel bad extending or raising "issues" that are more feedback/usage related).

**Star Mask** - should this be from the original (uncropped) image, or do we use an image as-is from INDI-AllSky as a reference to start the mask from?

Checked image order of operations, as below, so I presume I need to de-crop an as-is image to use the correct resolution for the mask?

8. Calculate ADU
9. Detect stars
10. Line detection
11. Crop

aaronwmorris, 2022-09-01:
I enabled the "discussions" feature, we will see how that goes.

As for the mask, the mask should be the original image size, pre-cropped.  I just picked one of he images and drew on top of the image.

Bra1nsen, 2025-03-29:
I'm really impressed by the circular keogram timelapse design. Is there an existing Python library or module that can generate this type of visualization out of the box?