aaronwmorris/indi-allsky #184: Install on orange pi

pierrebilb, 2022-04-29:
Hi ,
i tried to install indi-allsky on orange pi +2e with armbian  debian 11 but i get errors with built wheel of cffi and ansible..
Any ideas why these errors?
Thanks
Pierre

aaronwmorris, 2022-04-29:
Can you post the errors?  Do you know if you are running `armhf` (32bit) or `aarch64` (64bit)?

aaronwmorris, 2022-04-30:
Okay, I found at couple of packages necessary for building some of the python modules from source.  #185 & #186 

The pre-built python wheels are a really nice feature, but they can be a bit of moving target.

pierrebilb, 2022-04-30:
I'm running armv7l; i will poste the errors next week. I'm not home thiss week end.
Thanks a lot for your help!

pierrebilb, 2022-05-01:
Here is the error i encounter  when buildind indi, even with last github(#185 and #186):
`Python: 3.9.2
          platform: Linux-5.15.25-sunxi-armv7l-with-glibc2.31
          pip: 20.3.4
          setuptools: 52.0.0
          setuptools_rust: 1.3.0
          =============================DEBUG ASSISTANCE=============================

      Traceback (most recent call last):
        File "/home/pierre/indi-allsky/virtualenv/ansible/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
          main()
        File "/home/pierre/indi-allsky/virtualenv/ansible/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/home/pierre/indi-allsky/virtualenv/ansible/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 130, in get_requires_for_build_wheel
          return hook(config_settings)
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 149, in get_requires_for_build_wheel
          return self._get_build_requires(
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 130, in _get_build_requires
          self.run_setup()
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 145, in run_setup
          exec(compile(code, __file__, 'exec'), locals())
        File "setup.py", line 39, in <module>
          setup(
        File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 153, in setup
          return distutils.core.setup(**attrs)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/core.py", line 108, in setup
          _setup_distribution = dist = klass(attrs)
        File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 432, in __init__
          _Distribution.__init__(self, {
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/dist.py", line 293, in __init__
          self.finalize_options()
        File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 708, in finalize_options
          ep(self)
        File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 714, in _finalize_setup_keywords
          ep.require(installer=self.fetch_build_egg)
        File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2472, in require
          items = working_set.resolve(reqs, env, installer, extras=self.extras)
        File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 777, in resolve
          raise VersionConflict(dist, req).with_context(dependent_req)
      pkg_resources.VersionConflict: (setuptools 52.0.0 (/usr/lib/python3/dist-packages), Requirement.parse('setuptools>=58.0'))
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
`
Pierre

aaronwmorris, 2022-05-02:
That is really bizarre.  Have you tried performing a full package upgrade first (apt-get update; apt-get upgrade)?

You may also want to completely remove the virtualenvs under the virtualenv/ folder and start over from scratch.

pierrebilb, 2022-05-02:
Hi,
i started again from scratch, writing the sd card with armbian debian 11 , performed packages update (apt-get update, apt-get -y dist-upgrade) and got the exact same error!
Pierre

pierrebilb, 2022-05-02:
I also tried with armbian ubuntu 20.04, almost the same error:
`Python: 3.8.10
          platform: Linux-5.15.25-sunxi-armv7l-with-glibc2.29
          pip: 20.0.2
          setuptools: 45.2.0
          setuptools_rust: 1.3.0
          =============================DEBUG ASSISTANCE=============================

      Traceback (most recent call last):
        File "/home/pierre/indi-allsky/virtualenv/ansible/lib/python3.8/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
          main()
        File "/home/pierre/indi-allsky/virtualenv/ansible/lib/python3.8/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/home/pierre/indi-allsky/virtualenv/ansible/lib/python3.8/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 130, in get_requires_for_build_wheel
          return hook(config_settings)
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 147, in get_requires_for_build_wheel
          return self._get_build_requires(
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 128, in _get_build_requires
          self.run_setup()
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 143, in run_setup
          exec(compile(code, __file__, 'exec'), locals())
        File "setup.py", line 39, in <module>
          setup(
        File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 144, in setup
          return distutils.core.setup(**attrs)
        File "/usr/lib/python3.8/distutils/core.py", line 108, in setup
          _setup_distribution = dist = klass(attrs)
        File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 447, in __init__
          _Distribution.__init__(self, {
        File "/usr/lib/python3.8/distutils/dist.py", line 292, in __init__
          self.finalize_options()
        File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 740, in finalize_options
          ep.load()(self)
        File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 746, in _finalize_setup_keywords
          ep.require(installer=self.fetch_build_egg)
        File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2467, in require
          items = working_set.resolve(reqs, env, installer, extras=self.extras)
        File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 792, in resolve
          raise VersionConflict(dist, req).with_context(dependent_req)
      pkg_resources.VersionConflict: (setuptools 45.2.0 (/usr/lib/python3/dist-packages), Requirement.parse('setuptools>=58.0'))
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error
`

aaronwmorris, 2022-05-03:
I am trying to find an OS image I can test on my hardware.

I think what is happening is one of the packages that normally gets installed from a prebuilt wheel... the wheel is not available for your version of python and architecture combination.  Therefore your system is trying to build it from source.  There is a conflict between requirements of some modules.

Give me a couple of days on this...

pierrebilb, 2022-05-03:
I searched a bit on internet and found out like you that armbian python should generate conflict..
Thanks a lot for helping me!
best regards
Pierre

aaronwmorris, 2022-05-03:
I do not think it will help, but try to pull the latest code.  I upgraded a couple of modules in #187

pierrebilb, 2022-05-04:
Hi,
unfortunately, it didn't make the trick; still the same error..

aaronwmorris, 2022-05-05:
You may want to try again with Armbian Ubuntu 20.04.  I have changed Ubuntu 20.04 to use python 3.9 in #188 (instead of the default 3.8).  The python wheels site maintains better support for pre-compiled wheels for 3.9 on armhf (armv7l).

aaronwmorris, 2022-05-06:
According to the log, it is still trying to use Python 3.8.  Can you post the information regarding the distribution and platform that is printed when you first run the script?

aaronwmorris, 2022-05-06:
Try installing the system version of the python cryptography module.

```
sudo apt-get install python3-cryptography
```

Remove the ansible virtualenv in the indi-allsky install and start again.

```
rm -fR virtualenv/ansible
```

pierrebilb, 2022-05-07:
Ok,
i'll try tonight..

pierrebilb, 2022-05-08:
Indi installed allright but some problems occurred after in the script:
  Building wheel for cmake (pyproject.toml): still running...
        Building wheel for cmake (pyproject.toml): still running...
        Building wheel for cmake (pyproject.toml): still running...
        Building wheel for cmake (pyproject.toml): still running...
        Building wheel for cmake (pyproject.toml): still running...
        Building wheel for cmake (pyproject.toml): still running...
        Building wheel for cmake (pyproject.toml): still running...
        Building wheel for cmake (pyproject.toml): still running...
        Building wheel for cmake (pyproject.toml): finished with status 'error'
        error: subprocess-exited-with-error

        × Building wheel for cmake (pyproject.toml) did not run successfully.
        │ exit code: 1
        ╰─> [4499 lines of output]
            Not searching for unused variables given on the command line.
            CMake Error: CMake was unable to find a build program corresponding to "Ninja".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
            -- Configuring incomplete, errors occurred!
            See also "/tmp/pip-install-r3q16ae7/cmake_b4435414f403460095e93296fa35b4ef/_cmake_test_compile/build/CMakeFiles/CMakeOutput.log".
            Not searching for unused variables given on the command line.
            -- The C compiler identification is GNU 9.4.0
            -- Check for working C compiler: /usr/bin/cc
            -- Check for working C compiler: /usr/bin/cc -- works
            -- Detecting C compiler ABI info
            -- Detecting C compiler ABI info - done
            -- Detecting C compile features

aaronwmorris, 2022-05-08:
Well, I am really at a loss.  It seems like something is really old or broken in the images you are using.

I am still investigating.

pierrebilb, 2022-05-12:
I tried to manually install opencv  on armbian ubuntu and didn't success... It doesn't seems to be easy..

aaronwmorris, 2022-05-12:
I merged #196 yesterday.  This upgrades the version of Python on Ubuntu 20.04.  You may try deleting your virtualenvs and trying again.

pierrebilb, 2022-05-14:
Hi,
still the same error when building opencv..

aaronwmorris, 2022-05-14:
The real problem here is it should be be building opencv.  It should be using a pre-compiled wheel.  For some reason, your system's python is not finding the available pre-compiled wheels.

RCompassi, 2022-08-01:
I am having the same issue with building the python wheel.

I'm using Orange pi PC plus and armbian 22.05 bullseye

>  writing manifest file 'src/cryptography.egg-info/SOURCES.txt'

      /tmp/pip-build-env-satzdsvg/overlay/lib/python3.9/site-packages/setuptools/command/build_py.py:153: SetuptoolsDeprecationWarning:     Installing 'cryptography.hazmat.bindings._rust' as data is deprecated, please list it in `packages`.
          !!
 ############################
          # Package would be ignored #
          ############################
          Python recognizes 'cryptography.hazmat.bindings._rust' as an importable package,
          but it is not listed in the `packages` configuration of setuptools.

          'cryptography.hazmat.bindings._rust' has been automatically added to the distribution only
          because it may contain data files, but this behavior is likely to change
          in future versions of setuptools (and therefore is considered deprecated).

          Please make sure that 'cryptography.hazmat.bindings._rust' is included as a package by using
          the `packages` configuration field or the proper discovery methods
          (for example by using `find_namespace_packages(...)`/`find_namespace:`
          instead of `find_packages(...)`/`find:`).

          You can read more about "package discovery" and "data files" on setuptools
          documentation page.

      !!

        check.warn(importable)
      copying src/cryptography/py.typed -> build/lib.linux-armv7l-cpython-39/cryptography
      creating build/lib.linux-armv7l-cpython-39/cryptography/hazmat/bindings/_rust
      copying src/cryptography/hazmat/bindings/_rust/__init__.pyi -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/bindings/_rust
      copying src/cryptography/hazmat/bindings/_rust/asn1.pyi -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/bindings/_rust
      copying src/cryptography/hazmat/bindings/_rust/ocsp.pyi -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/bindings/_rust
      copying src/cryptography/hazmat/bindings/_rust/x509.pyi -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/bindings/_rust
      running build_ext
      running build_rust

          =============================DEBUG ASSISTANCE=============================
          If you are seeing a compilation error please try the following steps to
          successfully install cryptography:
          1) Upgrade to the latest pip and try again. This will fix errors for most
             users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip
          2) Read https://cryptography.io/en/latest/installation/ for specific
             instructions for your platform.
          3) Check our frequently asked questions for more information:
             https://cryptography.io/en/latest/faq/
          4) Ensure you have a recent Rust toolchain installed:
             https://cryptography.io/en/latest/installation/#rust

          Python: 3.9.2
          platform: Linux-5.15.48-sunxi-armv7l-with-glibc2.31
          pip: n/a
          setuptools: 63.3.0
          setuptools_rust: 1.4.1
          =============================DEBUG ASSISTANCE=============================

      error: can't find Rust compiler

      If you are using an outdated pip version, it is possible a prebuilt wheel is available for this package but pip is not able to install from it. Installing from the wheel would avoid the need for a Rust compiler.

      To update pip, run:

          pip install --upgrade pip

      and then retry package installation.

      If you did intend to build this package from source, try installing a Rust compiler from your system package manager and ensure it is on the PATH during installation. Alternatively, rustup (available at https://rustup.rs) is the recommended way to download and update the Rust compiler toolchain.

      This package requires Rust >=1.41.0.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for cryptography
Successfully built python-apt
Failed to build cryptography
ERROR: Could not build wheels for cryptography, which is required to install pyproject.toml-based projects

aaronwmorris, 2022-08-01:
I will see what packages are necessary for the rust compiler and add that to the setup script.

RCompassi, 2022-08-01:
> something

I installed manually the rust compiler, as instructed on their site, and got the same error. Perhaps I missed a configuration somewhere?

aaronwmorris, 2022-08-01:
It could be another missing development related package.  This can get frustrating.

RCompassi, 2022-08-01:
You have to state the path to rustc somewhere I guess

RCompassi, 2022-08-01:
I installed Rust again, configured it's path, the shell respond with the correct version when you type rustc --version
Same error. Needs to install pip and no rust found, but both are installed correctly

aaronwmorris, 2022-08-01:
I was able to build cryptography by just installing `rustc` but I could only build the module from source.  If I tried to build from pip, it always fails.

However, none of this should be necessary.  There are prebuilt wheels for your python version and platform.

I have also merged #279 to add rust and opencv packages in case they are needed.  I do not think this will help you, but I added them by default.

RCompassi, 2022-08-01:
> 

But how do we use those prebuilt versions and stop them from being built?

aaronwmorris, 2022-08-02:
> But how do we use those prebuilt versions and stop them from being built?

This is what does not make any sense.  The prebuilt versions should automatically be detected and used.  I can only assume that something in the python distribution for your SoC has something that is causing the wheels to not be detected.  Maybe it is a really early version of Python in the sub-release.

RCompassi, 2022-08-02:
I think I got it, but i had to instal a LOT of dependencies, and I don't know wich one did the trick! lol

`sudo apt-get install build-essential libssl-dev libffi-dev python-dev`
`pip install --upgrade pip`
`pip install cryptography`
`pip install pyopenssl`
`pip install setuptools-rust`
`sudo apt-get install build-dep python3-lxml`
`sudo apt-get install libxml2-dev libxslt-dev`
`python3 -m pip install --upgrade pip`

Now it moves forward until:

`TASK [build_indi : Setup indi_core build] **************************************
changed: [localhost]

TASK [build_indi : Build indi_core] ********************************************`

And stays there. Do it really takes a VERY long time to build indi_core?

aaronwmorris, 2022-08-02:
Building indi can take 2-3+ hours, perhaps even longer depending on how many cores the compiler is able to use.  The number of cores used depends on the amount of memory in your system.

RCompassi, 2022-08-02:
Yeah, it took a very long time but ended with a success!

RCompassi, 2022-08-02:
[ 46%] No patch step for 'CMakeProject-build'
            [ 50%] Performing configure step for 'CMakeProject-build'
            loading initial cache file /tmp/pip-install-4b6f05mz/cmake_630cd9bbb7d241e48b8e342fcfdef1ba/_skbuild/linux-armv7l-3.9/cmake-build/initial-cache.txt
            loading initial cache file /tmp/pip-install-4b6f05mz/cmake_630cd9bbb7d241e48b8e342fcfdef1ba/_skbuild/linux-armv7l-3.9/cmake-build/CMakeProject-build-prefix/tmp/CMakeProject-build-cache-Release.cmake
            CMake Error at /usr/share/cmake-3.18/Modules/CMakeDetermineCompilerId.cmake:223 (file):
              file write failed (No space left on device):

                /tmp/pip-install-4b6f05mz/cmake_630cd9bbb7d241e48b8e342fcfdef1ba/_skbuild/linux-armv7l-3.9/cmake-build/CMakeProject-build/CMakeFiles/3.18.4/CompilerIdC/CMakeCCompilerId.c
            Call Stack (most recent call first):
              /usr/share/cmake-3.18/Modules/CMakeDetermineCompilerId.cmake:234 (CMAKE_DETERMINE_COMPILER_ID_WRITE)
              /usr/share/cmake-3.18/Modules/CMakeDetermineCompilerId.cmake:32 (CMAKE_DETERMINE_COMPILER_ID_BUILD)
              /usr/share/cmake-3.18/Modules/CMakeDetermineCCompiler.cmake:116 (CMAKE_DETERMINE_COMPILER_ID)
              CMakeLists.txt:12 (project)

            CMake Error at /usr/share/cmake-3.18/Modules/CMakeDetermineCompilerId.cmake:223 (file):
              file write failed (No space left on device):

There is 29gb free on the SD.....

I'm giving up, I really like to try, but is too much effort

aaronwmorris, 2022-08-02:
If you are using Armbian, there are some "optimizations" you may need to disable....

https://github.com/aaronwmorris/indi-allsky/wiki/Armbian-Tuning

aaronwmorris, 2022-08-02:
Merged #280 to add additional packages dependencies and python modules (findings above)

RCompassi, 2022-08-02:
Did the tunning, there is new eerors after 4h of compiling..

Collecting scipy
  Downloading scipy-1.9.0.tar.gz (42.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 42.0/42.0 MB 835.3 kB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [61 lines of output]
      The Meson build system
      Version: 0.62.2
      Source dir: /tmp/pip-install-_2ecn7pj/scipy_ba1b53a422f04116bd2431410bc2f1b9
      Build dir: /tmp/pip-install-_2ecn7pj/scipy_ba1b53a422f04116bd2431410bc2f1b9/.mesonpy-bzhucuv3/build
      Build type: native build
      Project name: SciPy
      Project version: 1.9.0
      C compiler for the host machine: cc (gcc 10.2.1 "cc (Debian 10.2.1-6) 10.2.1 20210110")
      C linker for the host machine: cc ld.bfd 2.35.2
      C++ compiler for the host machine: c++ (gcc 10.2.1 "c++ (Debian 10.2.1-6) 10.2.1 20210110")
      C++ linker for the host machine: c++ ld.bfd 2.35.2
      Host machine cpu family: arm
      Host machine cpu: armv7l
      Compiler for C supports arguments -Wno-unused-but-set-variable: YES
      Library m found: YES

      ../../meson.build:41:0: ERROR: Unknown compiler(s): [['gfortran'], ['flang'], ['nvfortran'], ['pgfortran'], ['ifort'], ['g95']]
      The following exception(s) were encountered:
      Running "gfortran --version" gave "[Errno 2] No such file or directory: 'gfortran'"
      Running "gfortran -V" gave "[Errno 2] No such file or directory: 'gfortran'"
      Running "flang --version" gave "[Errno 2] No such file or directory: 'flang'"
      Running "flang -V" gave "[Errno 2] No such file or directory: 'flang'"
      Running "nvfortran --version" gave "[Errno 2] No such file or directory: 'nvfortran'"
      Running "nvfortran -V" gave "[Errno 2] No such file or directory: 'nvfortran'"
      Running "pgfortran --version" gave "[Errno 2] No such file or directory: 'pgfortran'"
      Running "pgfortran -V" gave "[Errno 2] No such file or directory: 'pgfortran'"
      Running "ifort --version" gave "[Errno 2] No such file or directory: 'ifort'"
      Running "ifort -V" gave "[Errno 2] No such file or directory: 'ifort'"
      Running "g95 --version" gave "[Errno 2] No such file or directory: 'g95'"
      Running "g95 -V" gave "[Errno 2] No such file or directory: 'g95'"

      A full log can be found at /tmp/pip-install-_2ecn7pj/scipy_ba1b53a422f04116bd2431410bc2f1b9/.mesonpy-bzhucuv3/build/meson-logs/meson-log.txt
      + meson setup --native-file=/tmp/pip-install-_2ecn7pj/scipy_ba1b53a422f04116bd2431410bc2f1b9/.mesonpy-native-file.ini -Ddebug=false -Doptimization=2 --prefix=/usr /tmp/pip-install-_2ecn7pj/scipy_ba1b53a422f04116bd2431410bc2f1b9 /tmp/pip-install-_2ecn7pj/scipy_ba1b53a422f04116bd2431410bc2f1b9/.mesonpy-bzhucuv3/build
      Traceback (most recent call last):
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
          main()
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 130, in get_requires_for_build_wheel
          return hook(config_settings)
        File "/tmp/pip-build-env-vqylade5/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 909, in get_requires_for_build_wheel
          with _project(config_settings) as project:
        File "/usr/lib/python3.9/contextlib.py", line 117, in __enter__
          return next(self.gen)
        File "/tmp/pip-build-env-vqylade5/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 888, in _project
          with Project.with_temp_working_dir(
        File "/usr/lib/python3.9/contextlib.py", line 117, in __enter__
          return next(self.gen)
        File "/tmp/pip-build-env-vqylade5/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 547, in with_temp_working_dir
          yield cls(source_dir, tmpdir, build_dir)
        File "/tmp/pip-build-env-vqylade5/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 463, in __init__
          self._configure(reconfigure=bool(build_dir) and not native_file_mismatch)
        File "/tmp/pip-build-env-vqylade5/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 494, in _configure
          self._meson(
        File "/tmp/pip-build-env-vqylade5/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 477, in _meson
          return self._proc('meson', *args)
        File "/tmp/pip-build-env-vqylade5/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 472, in _proc
          subprocess.check_call(list(args))
        File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['meson', 'setup', '--native-file=/tmp/pip-install-_2ecn7pj/scipy_ba1b53a422f04116bd2431410bc2f1b9/.mesonpy-native-file.ini', '-Ddebug=false', '-Doptimization=2', '--prefix=/usr', '/tmp/pip-install-_2ecn7pj/scipy_ba1b53a422f04116bd2431410bc2f1b9', '/tmp/pip-install-_2ecn7pj/scipy_ba1b53a422f04116bd2431410bc2f1b9/.mesonpy-bzhucuv3/build']' returned non-zero exit status 1.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

The script exited abnormally, please try to run again...

aaronwmorris, 2022-08-02:
Looks like scipy requires a fortran compiler.

Try again after installing gfortran
```
apt-get install gfortran
```

RCompassi, 2022-08-03:
Besides fortran, needs openblas:

` Found CMake: /usr/bin/cmake (3.18.4)
      Run-time dependency openblas found: NO (tried pkgconfig and cmake)
`

Installing and recompiling....

git clone https://github.com/xianyi/OpenBLAS.git
cd OpenBLAS
make -j4
sudo make install

RCompassi, 2022-08-03:
Even after installing openblass, I cant get it to work

      Found CMake: /usr/bin/cmake (3.18.4)
      Run-time dependency openblas found: NO (tried pkgconfig and cmake)

aaronwmorris, 2022-08-03:
Merged #283 with some of the additional dependencies.

RCompassi, 2022-08-03:
No luck yet!

 [311/1568] Compiling C object scipy/special/cython_special.cpython-39-arm-linux-gnueabihf.so.p/meson-generated_cython_special.c.o
      FAILED: scipy/special/cython_special.cpython-39-arm-linux-gnueabihf.so.p/meson-generated_cython_special.c.o
      cc -Iscipy/special/cython_special.cpython-39-arm-linux-gnueabihf.so.p -Iscipy/special -I../../scipy/special -I/tmp/pip-build-env-ulz5v95_/overlay/lib/python3.9/site-packages/numpy/core/include -Iscipy/_lib -I../../scipy/_lib -I../../scipy/_build_utils/src -I../../scipy/special/cephes -I/usr/include/python3.9 -I/usr/include/arm-linux-gnueabihf/python3.9 -I/usr/include/arm-linux-gnueabihf/openblas-pthread/ -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=c99 -O2 -Wno-unused-but-set-variable -Wno-unused-function -Wno-conversion -Wno-misleading-indentation -Wno-incompatible-pointer-types -fPIC -Wno-cpp -Wno-maybe-uninitialized -MD -MQ scipy/special/cython_special.cpython-39-arm-linux-gnueabihf.so.p/meson-generated_cython_special.c.o -MF scipy/special/cython_special.cpython-39-arm-linux-gnueabihf.so.p/meson-generated_cython_special.c.o.d -o scipy/special/cython_special.cpython-39-arm-linux-gnueabihf.so.p/meson-generated_cython_special.c.o -c scipy/special/cython_special.cpython-39-arm-linux-gnueabihf.so.p/cython_special.c
      cc: fatal error: Killed signal terminated program cc1
      compilation terminated.
      [312/1568] Generating 'scipy/linalg/cython_lapack.cpython-39-arm-linux-gnueabihf.so.p/cython_lapack.c'.
      [313/1568] Compiling C object scipy/linalg/_decomp_update.cpython-39-arm-linux-gnueabihf.so.p/meson-generated__decomp_update.c.o
      [314/1568] Compiling C object scipy/linalg/_cythonized_array_utils.cpython-39-arm-linux-gnueabihf.so.p/meson-generated__cythonized_array_utils.c.o
      [315/1568] Compiling C object scipy/linalg/_matfuncs_expm.cpython-39-arm-linux-gnueabihf.so.p/meson-generated__matfuncs_expm.c.o
      [316/1568] Compiling C object scipy/linalg/_flapack.cpython-39-arm-linux-gnueabihf.so.p/meson-generated_..__flapackmodule.c.o
      ninja: build stopped: subcommand failed.
      + meson setup --native-file=/tmp/pip-install-a0qmpgaz/scipy_1a5cab3d25964b0ebd4e0567748b8b4c/.mesonpy-native-file.ini -Ddebug=false -Doptimization=2 --prefix=/usr /tmp/pip-install-a0qmpgaz/scipy_1a5cab3d25964b0ebd4e0567748b8b4c /tmp/pip-install-a0qmpgaz/scipy_1a5cab3d25964b0ebd4e0567748b8b4c/.mesonpy-yzkiptza/build
      + meson compile
      Traceback (most recent call last):
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 156, in prepare_metadata_for_build_wheel
          hook = backend.prepare_metadata_for_build_wheel
      AttributeError: module 'mesonpy' has no attribute 'prepare_metadata_for_build_wheel'

      During handling of the above exception, another exception occurred:

      Traceback (most recent call last):
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
          main()
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 160, in prepare_metadata_for_build_wheel
          whl_basename = backend.build_wheel(metadata_directory, config_settings)
        File "/tmp/pip-build-env-ulz5v95_/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 927, in build_wheel
          return project.wheel(out).name
        File "/tmp/pip-build-env-ulz5v95_/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 875, in wheel
          wheel = _WheelBuilder(self).build(self._install_plan, self._copy_files, self._build_dir)
        File "/tmp/pip-build-env-ulz5v95_/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 354, in build
          self._project.build()  # ensure project is built
        File "/tmp/pip-build-env-ulz5v95_/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 535, in build
          self._meson('compile')
        File "/tmp/pip-build-env-ulz5v95_/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 477, in _meson
          return self._proc('meson', *args)
        File "/tmp/pip-build-env-ulz5v95_/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 472, in _proc
          subprocess.check_call(list(args))
        File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['meson', 'compile']' returned non-zero exit status 1.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

The script exited abnormally, please try to run again...

aaronwmorris, 2022-08-03:
> cc: fatal error: Killed signal terminated program cc1

Unless I am misunderstanding, it looks like a signal killed the compiler.  I wonder if you ran out of memory during the compile.

RCompassi, 2022-08-03:
Looks like it ran out of memoty. I increased the swap file and tmp and will try again

RCompassi, 2022-08-03:
No go.

Any idea?

 cc -Iscipy/special/cython_special.cpython-39-arm-linux-gnueabihf.so.p -Iscipy/special -I../../scipy/special -I/tmp/pip-build-env-j1jjtqsr/overlay/lib/python3.9/site-packages/numpy/core/include -Iscipy/_lib -I../../scipy/_lib -I../../scipy/_build_utils/src -I../../scipy/special/cephes -I/usr/include/python3.9 -I/usr/include/arm-linux-gnueabihf/python3.9 -I/usr/include/arm-linux-gnueabihf/openblas-pthread/ -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=c99 -O2 -Wno-unused-but-set-variable -Wno-unused-function -Wno-conversion -Wno-misleading-indentation -Wno-incompatible-pointer-types -fPIC -Wno-cpp -Wno-maybe-uninitialized -MD -MQ scipy/special/cython_special.cpython-39-arm-linux-gnueabihf.so.p/meson-generated_cython_special.c.o -MF scipy/special/cython_special.cpython-39-arm-linux-gnueabihf.so.p/meson-generated_cython_special.c.o.d -o scipy/special/cython_special.cpython-39-arm-linux-gnueabihf.so.p/meson-generated_cython_special.c.o -c scipy/special/cython_special.cpython-39-arm-linux-gnueabihf.so.p/cython_special.c
      cc: fatal error: Killed signal terminated program cc1
      compilation terminated.
      [312/1568] Compiling C object scipy/linalg/_decomp_update.cpython-39-arm-linux-gnueabihf.so.p/meson-generated__decomp_update.c.o
      [313/1568] Generating 'scipy/linalg/cython_lapack.cpython-39-arm-linux-gnueabihf.so.p/cython_lapack.c'.
      [314/1568] Compiling C object scipy/linalg/_cythonized_array_utils.cpython-39-arm-linux-gnueabihf.so.p/meson-generated__cythonized_array_utils.c.o
      [315/1568] Compiling C object scipy/linalg/_matfuncs_expm.cpython-39-arm-linux-gnueabihf.so.p/meson-generated__matfuncs_expm.c.o
      [316/1568] Compiling C object scipy/linalg/_flapack.cpython-39-arm-linux-gnueabihf.so.p/meson-generated_..__flapackmodule.c.o
      ninja: build stopped: subcommand failed.
      + meson setup --native-file=/tmp/pip-install-9jj7mj0h/scipy_8087a34f07c04e66ae2a048262da7efc/.mesonpy-native-file.ini -Ddebug=false -Doptimization=2 --prefix=/usr /tmp/pip-install-9jj7mj0h/scipy_8087a34f07c04e66ae2a048262da7efc /tmp/pip-install-9jj7mj0h/scipy_8087a34f07c04e66ae2a048262da7efc/.mesonpy-6tb47zux/build
      + meson compile
      Traceback (most recent call last):
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 156, in prepare_metadata_for_build_wheel
          hook = backend.prepare_metadata_for_build_wheel
      AttributeError: module 'mesonpy' has no attribute 'prepare_metadata_for_build_wheel'

      During handling of the above exception, another exception occurred:

      Traceback (most recent call last):
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
          main()
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 160, in prepare_metadata_for_build_wheel
          whl_basename = backend.build_wheel(metadata_directory, config_settings)
        File "/tmp/pip-build-env-j1jjtqsr/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 927, in build_wheel
          return project.wheel(out).name
        File "/tmp/pip-build-env-j1jjtqsr/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 875, in wheel
          wheel = _WheelBuilder(self).build(self._install_plan, self._copy_files, self._build_dir)
        File "/tmp/pip-build-env-j1jjtqsr/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 354, in build
          self._project.build()  # ensure project is built
        File "/tmp/pip-build-env-j1jjtqsr/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 535, in build
          self._meson('compile')
        File "/tmp/pip-build-env-j1jjtqsr/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 477, in _meson
          return self._proc('meson', *args)
        File "/tmp/pip-build-env-j1jjtqsr/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 472, in _proc
          subprocess.check_call(list(args))
        File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['meson', 'compile']' returned non-zero exit status 1.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

The script exited abnormally, please try to run again...

aaronwmorris, 2022-08-03:
Looks like you ran out of memory again.

What I really need to get to the bottom is why the pre-compiled wheels are not being used.  I just do not have a system that exhibits this issue.

RCompassi, 2022-08-03:
I have 1gb ram, have disabled zram, ramlog and tmpfs as per your armbian tuning.

Any sugestions on what should I do to increase the memory?

aaronwmorris, 2022-08-04:
Zram can help if you can enable the zram swap support without enabling the zram /tmp support.

RCompassi, 2022-08-04:
So, what should I modify in the /etc/default/armbian-zram-config

Only set zram as true?

what about the other variables?

RCompassi, 2022-08-04:
Even with huge amount of space:

Filesystem      Size  Used Avail Use% Mounted on
udev            431M     0  431M   0% /dev
tmpfs           5.0G  820K  5.0G   1% /run
/dev/mmcblk0p1   28G  7.5G   21G  27% /
tmpfs           5.0G     0  5.0G   0% /dev/shm
tmpfs           5.0G  4.0K  5.0G   1% /run/lock
/dev/zram1      981M   72K  915M   1% /tmp
tmpfs           5.0G     0  5.0G   0% /run/user/1000

I'm getting:

        File "/tmp/pip-build-env-fxo6ppxa/overlay/lib/python3.9/site-packages/Cython/StringIOTree.py", line 69, in copyto
          target.write(stream_content)
        File "/usr/lib/python3.9/codecs.py", line 721, in write
          return self.writer.write(data)
        File "/usr/lib/python3.9/codecs.py", line 378, in write
          self.stream.write(data)
      OSError: [Errno 28] No space left on device

aaronwmorris, 2022-08-04:
Yeah, compiling the python modules requires a huge amount of space.  With the zram enabled /tmp you do not have enough.  I have not figured out a way of disabling the /tmp with the zram swap enabled in Armbian.

RCompassi, 2022-08-04:
Ah, too bad then

aaronwmorris, 2022-08-04:
A couple of months ago, I purchased an Orange Pi Zero 2 for testing and I did not have any problems.

I am thinking about buying an Orange Pi PC Plus to see if I can get to the bottom of this.

You can see what I have tested here:  https://github.com/aaronwmorris/indi-allsky/wiki/Supported-SoC-Hardware

aaronwmorris, 2022-08-27:
My Orange Pi PC Plus just arrived today.  I will be doing testing over the next few days.

aaronwmorris, 2022-08-28:
I have found couple of extra packages that need to be installed and locked down a python module, but....

The problem appears to be the armv7l (32-bit) platform and newer versions of Python (3.8+).  The pre-built python wheels are just not available.  The industry appears to be standardizing on aarch64 (64-bit) and that is leaving older SoCs behind.

Also, 1GB of memory is just not enough to compile the opencv-python-headless module.

I am thinking of writing a script to compile python 3.7 to see if it may be possible to use the wheels for that version of python.

aaronwmorris, 2022-08-28:
After disabling zram swap and installing dphys-swapfile (with a 2GB swap) everything was able to build.  It took about 3 hours to bulid indi and another 3-4 hours to build all of the python modules, but I am fully up and running on my Orange Pi PC Plus on Armbian 22.05.3 (Bullseye).

pierrebilb, 2022-09-03:
Good news, i'm going to try later this month to install on opi pc +!
Thank you very much

aaronwmorris, 2022-09-21:
Merged #357 to remove the rawpy requirements on 32-bit arm platforms.

pierrebilb, 2022-09-23:
It's working!
Thanks a lot
Best regards
Pierre