aaronwmorris/indi-allsky #191: Altair 224C

sharpie78, 2022-05-09:
Hello, I've got this software up and running and really like it, the timelapse creation etc are a great time saver. Thankyou.
I do have a couple of minor issues... hopefully you'll be able to help... :-)

Moon mode doesn't seem to work. The moon was above the horizon last night but it never switched to moon mode.  

I seem unable to capture in 16bit. I'm probably missing something obvious. 

I can log in to the camera using kstars (while indi-allsky isn't running but the indiserver is) and set the camera up as I want it but when I log out of it from kstars and restart indi-allsky, it goes back to 8bits per pixel. 

I've tried various settings, read around the subject and other issues people have had but I'm not a linux/indi power user. I think a similar work around to the ASI120MC S camera issue might work but I do not know how to add the parameters in INDI configuration.

Happy to send logs and do any testing you want for further info on this camera but not being a power user, I may need some direction on that.

Thanks in advance
Jack

sharpie78, 2022-05-09:
After doing some testing etc I'm pretty sure this can be solved by editing the INDI configuration. The camera seems to default to "video capture mode" when using AltairCapture and rgb mode (8bit) auto exposure mode on connection in INDI. Adding a configuration to load on startup would logically sort this out... to me it makes sense anyway. 

When connected with INDI I'm obviously able to set it as I like (including HCG/LCG settings like gpcam290) and it works fine in kstars. 

I've attached 2 files... the properties of the camera at the required settings (lines 34 & 42 & 109 in particular are relevant I think) and the properties of the camera on connection with indiserver/indi-allsky (default settings) (lines 56 & 65 & 154)

[224c default settings.txt](https://github.com/aaronwmorris/indi-allsky/files/8651864/224c.default.settings.txt)
[224c required settings.txt](https://github.com/aaronwmorris/indi-allsky/files/8651865/224c.required.settings.txt)

Hopefully those settings will help with development of camera support for you too.
Cheers
Jack

aaronwmorris, 2022-05-09:
Okay, I am going to try to deal with the 16 bit mode first.

Try the following settings and fully restart indi-allsky...

```
{
    "PROPERTIES" : {},
    "SWITCHES" : {
        "CCD_VIDEO_FORMAT" : {
            "on"  : ["TC_VIDEO_COLOR_RAW"],
            "off" : ["TC_VIDEO_COLOR_RGB"]
        },
        "TC_HCG_CONTROL" : {
            "on"  : ["GAIN_HIGH"],
            "off" : ["GAIN_LOW", "GAIN_HDR"]
        }
    }
}
```

sharpie78, 2022-05-09:
I have copied and pasted that into INDI configuration on the webpage and rebooted the whole rpi. running getprop again while indi-allsky is running confirms those settings have been applied, the camera and software are still working fine but still in 8bit.
does it make a difference if i paste it there or into the .json file in /etc ?

sharpie78, 2022-05-09:
deleted the previous config and pasted that in... still 8 bit and interestingly when running getprop they're both off.

Altair ALTAIR224C3.FITS_HEADER.FITS_OBJECT=Unknown
Altair ALTAIR224C3.CCD_CAPTURE_FORMAT.INDI_RAW=Off
Altair ALTAIR224C3.CCD_CAPTURE_FORMAT.INDI_RGB=Off
Altair ALTAIR224C3.CCD_TRANSFER_FORMAT.FORMAT_FITS=On
Altair ALTAIR224C3.CCD_TRANSFER_FORMAT.FORMAT_NATIVE=Off

when i check in kstars the format is  set to Raw16 though. bit of progress. :-) screenshot attached
[image]

sharpie78, 2022-05-09:
[image]

aaronwmorris, 2022-05-09:
Let's try to combine settings....

```
{
    "PROPERTIES" : {},
    "SWITCHES" : {
        "CCD_CAPTURE_FORMAT" : {
            "on"  : ["INDI_RAW"],
            "off" : ["INDI_RGB"]
        },
        "CCD_VIDEO_FORMAT" : {
            "on"  : ["TC_VIDEO_COLOR_RAW"],
            "off" : ["TC_VIDEO_COLOR_RGB"]
        }
    }
}
```

sharpie78, 2022-05-09:
seems hopeful. think it worked. just doing a cold boot to make sure.

sharpie78, 2022-05-09:
sort of..... it's the right switches.... but the other way around. raw off. rgb on. it seems counter-intuitive to me but a getprop confirms it and kstars does too. 16bits per pixel. happy days. thankyou very much for helping me figure that one out. :-)

presumably if I want to add any other switches I just add them in too ? starting to understand the syntax etc a bit now 

any ideas on the moon mode issue too please ?

aaronwmorris, 2022-05-09:
Great!  I will update my wiki as well regarding the bit mode.

As for moon mode, do you have the correct latitude and longitude set in the config?

sharpie78, 2022-05-09:
I think so. what format should it be in ? decimal ? i.e 51.477318, -0.000825 for greenwich observatory

aaronwmorris, 2022-05-09:
Yes, decimal/floating point numbers.  Those look correct.

I might have a bug I recently introduced, let me review the code.

aaronwmorris, 2022-05-09:
I think I found the bug with moonmode.  I am working on that.

Can you post the settings you are using for the 16-bits above.  After reading your post, I'm not sure I understand what you meant :-)

sharpie78, 2022-05-09:
this is what I have in INDI config. 
{
    "PROPERTIES": {},
    "SWITCHES": {
        "CCD_CAPTURE_FORMAT": {
            "on": ["INDI_RGB"],
            "off": ["INDI_RAW"]
        },
        "CCD_VIDEO_FORMAT": {
            "on": ["TC_VIDEO_COLOR_RAW"],
            "off": ["TC_VIDEO_COLOR_RGB"]
        },
        "TC_HCG_CONTROL": {
            "on": ["GAIN_LOW"],
            "off": ["GAIN_HIGH" , "GAIN_HDR"]
        }
    }
}

sharpie78, 2022-05-09:
a small issue has arisen aaron.... indi-allsky fails when I restart the software... It does start successfully when you start it a 2nd time. The same behaviour happens on bootup. 

not a massive issue obviously, more of a minor inconvenience.

aaronwmorris, 2022-05-09:
Are any exceptions logged?

sharpie78, 2022-05-09:
not sure how to find them if there are any to be honest.

aaronwmorris, 2022-05-09:
The logs are in `/var/log/indi-allsky/indi-allsky.log`

sharpie78, 2022-05-09:
looking through it now

sharpie78, 2022-05-09:
what should I be looking for

sharpie78, 2022-05-09:
can i delete the log and then reboot ? it's too long to look through atm. lol

sharpie78, 2022-05-09:
think this is what you need :-)
[exceptions log.txt](https://github.com/aaronwmorris/indi-allsky/files/8654802/exceptions.log.txt)

aaronwmorris, 2022-05-09:
That does not make a lot of sense...

Try changing the order of the directives in your config file....

```
{
    "PROPERTIES" : {},
    "SWITCHES" : {
        "CCD_VIDEO_FORMAT" : {
            "on"  : ["TC_VIDEO_COLOR_RAW"],
            "off" : ["TC_VIDEO_COLOR_RGB"]
        },
        "CCD_CAPTURE_FORMAT" : {
            "on"  : ["INDI_RAW"],
            "off" : ["INDI_RGB"]
        },
	"TC_HCG_CONTROL": {
            "on": ["GAIN_LOW"],
            "off": ["GAIN_HIGH" , "GAIN_HDR"]
        }
    }
}
```

sharpie78, 2022-05-09:
will do. I'm just running it through a transition between day/night to see how it behaves with the gain change. get back to ya in 15
mins or so

aaronwmorris, 2022-05-09:
Well... that was embarrassing #194 

Moon mode should be fixed.

sharpie78, 2022-05-09:
LCG/HCG switch worked fine. stayed in 16bit mode. just need to adjust the daytime/night-time gain a touch to balance out the transition between day/night so I don't end up with an over exposed frame or 2 in the middle of the sequence. 
changing the order did not resolve the indi-allsky crash on first startup but it did revert the camera back to 8bit mode. I've  switched it back to the first order and it's all good again. back to 16 bit and working fine.

sharpie78, 2022-05-09:
how do I resolve the moon mode issue please ?
sorry I'm basically a noob

aaronwmorris, 2022-05-09:
Just pull down the latest code and restart.

```
git pull origin master
```

sharpie78, 2022-05-09:
ok cool. do I run that from jack@allskypi:~$ or do I need to cd into another directory ? I'm very new to git, thanks for your patience

aaronwmorris, 2022-05-09:
No problem, you need to do this from within the indi-allsky folder.

https://github.com/aaronwmorris/indi-allsky/wiki/Updating-indi-allsky  (re-running setup.sh is not necessary for this small of a change)

sharpie78, 2022-05-09:
right, that's sorted that. nice one. moon mode working perfectly so far. cheers. :-)

I've been tinkering... and have found out that you don't need to add a configuration for CCD_VIDEO_FORMAT at all, to resolve the 16bit issue. It's just this in my INDI config now...

{
"PROPERTIES": {},
"SWITCHES": {
"CCD_CAPTURE_FORMAT": {
"on": ["INDI_RGB"],
"off": ["INDI_RAW"]
}
}
}

This puts it in 16bits per pixel mode and keeps it there during day/night transitions while switching to HCG and the gain I've chosen for both night and moon mode. 

I'm going to carry on tinkering with HCG/LCG settings and will let you know how I get on.

This initial crash thing is still going on, if you get it figured out please let me know. It feels like a timeout thing to me as it always takes around 1min 20s to fail upon initial startup. dunno if that helps at all. If you want any more logs just let me know. on that... how do I clear the old logs please ?

I've learned a bit while doing this so thanks for that and for figuring this out. 
Jack :-)

aaronwmorris, 2022-05-10:
The timeout does not make a lot of sense to me.  The timeout is not even at the point where it changes the setting, it is just looking up the setting to be changed and the lookup times out.  The timeout matches the timeout in my code of 65 seconds.

You might try increasing the timeout in indi-allsky/indi.py on line 85.  Change the timeout to something like 650.0.

You can clear the log with the following command:
```
print | sudo tee /var/log/indi-allsky/indi-allsky.log

# or as root

> /var/log/indi-allsky/indi-allsky.log
```

sharpie78, 2022-05-10:
thanks for that. log cleared successfully and here's the new clean log from reboot, through the issue and on to 5 mins worth of it being stable and happy, doing exactly what it should do. :-)

[indi-allsky.log](https://github.com/aaronwmorris/indi-allsky/files/8656445/indi-allsky.log)

I'll try the timeout increase tomorrow and get back to you... getting late.

sharpie78, 2022-05-10:
update :
All changing the timeout did was extend the time it took to fail so I've reduced it back to 65. 

during testing I have found that...

Indi-allsky fails on boot everytime... I think because it is starting before the indiserver.

Indi-allsky fails eveytime you start the server and then start indi-allsky. Once this failure happens, the next time you start indi-allsky everything works with no probems. 

If you start the indiserver and then connect the camera through a 3rd party (kstars in my case) before starting indi-allsky, it works straight away and starts capturing.

It's repeatable and there's a link here somewhere I think.

aaronwmorris, 2022-05-10:
The indiserver is definitely started, but its possible the devices are not ready yet.

Can you try extending the timeout in `indi_allsky/allsky.py`.  There is an 8 second sleep on line 721 to give the devices time to initialize.  Try extending that to 80 seconds.

sharpie78, 2022-05-10:
unfortunately that hasn't worked either. I'm happy to keep trying different things if you think of any. If it's just something I have to live with for a while though... fair enough. It's only a small bug.
should I be putting the settings back to original each time btw ?
supplementary questions... what does the HUP indi-allsky button do ? ... and in the loop section, is it possible to stop "rock" being on by default please ?

aaronwmorris, 2022-05-11:
Yeah, revert the settings to the original.

HUP allows you to change some settings without doing a full restart.

I will look into making the setting permanent.

sharpie78, 2022-05-11:
cool. cheers for that. handy to know.

I'd suggest making the"rock" checkbox "sticky" rather than adding any permanent state to it. Sometimes I like it to rock back and forth but sometimes I do like it to play from the start each time. 

...by "sticky" I mean that whatever state you last had it in, that's the state it reverts to when you go into the loop section. make sense ?

I had a poke about in your code to see if I could find a simple switch for it that sets it to on/off by default so I could set it to off but I suspect it's more complicated than that. lol. After getting the 16bit issue sorted and poking about in the software it's made me want to learn python so I understand what I'm doing better.

aaronwmorris, 2022-05-11:
The way the rock functionality is implemented, it is a purely local javascript function.  The page gets a list of images (from the server) and the browser loops the images based on the javascript variable.  Currently, the rock setting is not configured from the server side.

You can change the default setting for rock in the flask template `indi_allsky/flask/templates/loop.html` line 161.

Comment out this line:
```
    //rock = $('#ROCK_CHECKBOX').prop('checked');
```

I do not think there is a big problem with using indi web manager and having multiple devices there.  indi-allsky will only connect to the first ccd class device it finds on the indi server and it should ignore the rest of the devices.  There is always a chance there might be some other unexpected interaction between devices, but I think it will be unlikely.

sharpie78, 2022-05-11:
I know even less about editing html and javascript than my limited knowledge of linux. lol.
I've searched around the subject of "commenting out" and tried multiple things but none of them worked. I assumed it was html I needed to research so I ended up putting  around the text on that line but it didn't seem to do anything apart from make it turn orange. 
Do I need to do some kind of update to the file or run setup.sh again or am I barking up completely the wrong tree with what I've done?
Sorry I'm such a noob and if you can't be bothered to explain it, no worries :) It's such a minor thing and you've sorted the important 16bit issue out nicely. I've had a day of trying to get indi to work on an upgrade of Kubuntu to 22.04 (dependency hell for a noob) but amazingly the clouds have cleared and everything is working...touch wood. indi-allsky is running smoothly, kstars is imaging. happy days

aaronwmorris, 2022-05-12:
Its no big deal.  With javascript, comments are like C comments, two forward slashes // for a single line comment and /* */ for multi-line comments.

I did not make it clear, but the example above has the comment included.

If you screw up the file, you can just delete it and re-checkout from the repository.

```
rm indi_allsky/flask/templates/loop.html
git checkout indi_allsky/flask/templates/loop.html
```

aaronwmorris, 2022-05-12:
Forget what I said about commenting out that line.  Now that look more closely, I do not think it would work.

I have merged #197 to use HTML5 storage to save the settings related to the loop page.

sharpie78, 2022-05-13:
I've been tinkering and reading and trying things to come to some understanding of what is going on with this failure of indi-allsky on its first attempt to start after the indiserver has just been started. 

things I do know... 
I do not understand programming languages ;-) 

indiserver and indi-allsky work perfectly from boot, fully automated, no issues... when you **_don't_** give it a custom configuration like we have for the 16bit issue. 

when I stop indi-allsky but not indiserver... next time I start indi-allsky, it works perfectly, no failure.

if i start indiserver and connect to it with kstars, then connect the camera and then start indi-allsky... I get no failure. off it goes... capturing away. I have observed that the camera is in the default 8bit mode when first connected in kstars but once indi-allsky starts, it switches to the settings from indi-allsky. 

The issue defintiely lies somewhere in the connection to the server and application of the settings in indi config. I can see why you're going down the timeout route. It seems like there needs to be a pause between starting the server, connecting the camera and applying the configuration to me. something like... start server, wait 10 seconds, connect camera, wait 10 seconds, apply indi-allsky camera settings, wait 10 seconds, start indi-allsky. 

dunno if that helps at all but figured I'd share what replicates the problem and what doesn't.

aaronwmorris, 2022-05-13:
Yes, it helps.  It is possible that kstars is setting certain parameters in a specific order which is not subject to the timeout.  Maybe on purpose... maybe by accident.

Let's try a few more things... there were a couple of settings that were set by kstars, but they did not look all that important to me, but who knows.

```
{
    "PROPERTIES" : {
	"CCD_CONTROLS" : {
            "Speed" : 0
        }
    },
    "SWITCHES" : {
        "CCD_CAPTURE_FORMAT" : {
            "on"  : ["INDI_RAW"],
            "off" : ["INDI_RGB"]
        }
    }
}
```

sharpie78, 2022-05-13:
think that's cracked it...
it works without failing when i stop both indiserver and indi-allsky, restart indiserver and then restart indi-allsky. no need for the intermediary of kstars anymore.
just going to do a full reboot of the pi to see if it fails or works on boot.

sharpie78, 2022-05-13:
failed on first boot I'm afraid. may not be indi though. currently got the image save location at an nfs mount point that isn't mounting on boot. I'll resolve that by changing it to a directory on the rpi itself and try again.

sharpie78, 2022-05-13:
oh dear. failed again even after making sure it definitely had somewhere to save the file. both on boot and just restarting both server and allsky. 
thought we'd cracked it then. ah well... next. lol

sharpie78, 2022-05-13:
this works....
start indiserver... wait 10 seconds... start indi-allsky and wait 20 seconds... query the camera (I'm running $ indi_getprop -t 2 "Altair ALTAIR224C3.CCD_INFO.CCD_BITSPERPIXEL.*" and getting an answer of 8bits) ...wait 5-10 seconds and query it again... result is 16 bit.
pretty sure that just asking the camera for it's information at the correct point is what is enabling the custom config to be loaded and for everything to work as it should.
If you query the server too early it doesn't work btw

sharpie78, 2022-05-13:
right...I've narrowed it down. if i query the camera between 10-20 seconds after starting indi-allsky everything works. it works on boot of the rpi or on restarting server and indi-allsky manually. luckily that's just enough time to ssh into the rpi after it boots and input the getprop command. 
If you can figure out how to get the software to behave like that... problem sorted I reckon. :-)

aaronwmorris, 2022-05-24:
Okay, there has been a new release of the pyindi-client python module that fixes a bug that prevented enumerating the camera device properties like ccd_getprop does.

Pull down the latest code and see if the latest changes help.

sharpie78, 2022-05-30:
No dice I'm afraid mate. Hasn't made any difference. All still running fine though... once ya get past the startup issue.

aaronwmorris, 2022-05-31:
Without a camera in hand, I am going to have a lot of difficulty tracking down this problem.

Is your camera a GPCAM2 or GPCAM3 model?

sharpie78, 2022-06-01:
It's a GPCam3 224c. just to check... pulling down the latest code... I did the same as I did before.... "git pull origin master" from inside the indi-allsky folder.
I'm assuming that's what you meant ?

aaronwmorris, 2022-07-05:
I have a bit of an unusual proposal for you...

Would you be willing to ship the camera to me so I can run more tests?  I would be willing to pay for the shipping myself, I just cannot justify $300 on a new camera.

sharpie78, 2022-07-24:
it's currently in use almost constantly I'm afraid. The software has been stable for weeks at a time so I'm inclined to just leave the issue and put up with the very small amount of hassle it causes. sorry man. 
Altair Astro are based very near me and I know they're really decent guys so my suggestion would be to ask them directly. They are always happy to help me. They might be able to give you more detailed information about the camera or maybe even lend you one for testing. You might suggest adding some branding/advertisement/promotion to your software to get a test camera ? I would expect a lot of the cameras have similar code runnng their cameras. I know the open source community doesn't generally encourage that but maybe frame it as collaboration instead of branding.

aaronwmorris, 2022-07-24:
I knew it was a long shot, but I am glad to hear the camera is still working.

aaronwmorris, 2022-10-26:
I have been working with @vastmeridian in #378 on what is likely the same issue.  I may have a new lead on what the problem might be.

sharpie78, 2022-10-26:
That's good to hear.  I've switched to using an ASI120MC-S for my allsky camera as it was a better shape for the enclosure but I am thinking of making another setup to watch the inside of my freshly rebuilt observatory. I'd use the Altair camera for that so rectifying this minor issue would be handy. Nice one.  :-)

aaronwmorris, 2022-10-27:
I have submitted an order for an Altair GPCAM3 224C and it should arrive in a week or two.

aaronwmorris, 2022-11-02:
Opened https://github.com/indilib/indi-3rdparty/issues/681

sharpie78, 2022-11-03:
Cool. I'm using the ASI120MC-s at the moment as I'm waiting for a cradle to attach the Altair 224 cam to my allsky box. Keen to get the altair cam going though....the asi120 is a lot noisier than the 224c.

[14 more replies]

closed 2023-01-27 by aaronwmorris