aaronwmorris/indi-allsky #193: Screen is blank but logs appear to be capturing

maphilli14, 2022-05-09:
I had a working allsky.git on a Pi3 with ASI290mm.  Upgrade script ran 2x times to resolve https://github.com/aaronwmorris/indi-allsky/issues/192
Now I see this
[image]

here are some possibly relevant logs
[image]

`May  9 16:00:05 allsky [INFO] MainProcess allsky.detectMoonMode() #817: Moon altitide: 30:48:49.1, phase 58.1%
May  9 16:00:05 allsky [INFO] MainProcess indi.getCcdTemperature() #345: Sensor temperature: 53.6
May  9 16:00:15 allsky [INFO] MainProcess allsky.run() #571: Camera last ready: 62.8s
May  9 16:00:15 allsky [INFO] MainProcess allsky.run() #572: Exposure state: ALERT
May  9 16:00:15 allsky [INFO] MainProcess allsky.detectNight() #800: Sun altitude: 48:07:10.0
May  9 16:00:15 allsky [INFO] MainProcess allsky.detectMoonMode() #817: Moon altitide: 30:50:48.7, phase 58.1%
May  9 16:00:15 allsky [INFO] MainProcess indi.getCcdTemperature() #345: Sensor temperature: 53.1
May  9 16:00:25 allsky [INFO] MainProcess allsky.run() #571: Camera last ready: 72.9s
May  9 16:00:25 allsky [INFO] MainProcess allsky.run() #572: Exposure state: ALERT
May  9 16:00:25 allsky [INFO] MainProcess allsky.detectNight() #800: Sun altitude: 48:05:09.8
May  9 16:00:25 allsky [INFO] MainProcess allsky.detectMoonMode() #817: Moon altitide: 30:52:48.4, phase 58.1%
May  9 16:00:25 allsky [INFO] MainProcess indi.getCcdTemperature() #345: Sensor temperature: 52.7
May  9 16:00:36 allsky [INFO] MainProcess allsky.run() #571: Camera last ready: 83.0s
May  9 16:00:36 allsky [INFO] MainProcess allsky.run() #572: Exposure state: ALERT
May  9 16:00:36 allsky [INFO] MainProcess allsky.detectNight() #800: Sun altitude: 48:03:09.3
May  9 16:00:36 allsky [INFO] MainProcess allsky.detectMoonMode() #817: Moon altitide: 30:54:48.3, phase 58.1%
May  9 16:00:36 allsky [INFO] MainProcess indi.getCcdTemperature() #345: Sensor temperature: 52.5
May  9 16:00:46 allsky [INFO] MainProcess allsky.run() #571: Camera last ready: 93.0s
May  9 16:00:46 allsky [INFO] MainProcess allsky.run() #572: Exposure state: ALERT
May  9 16:00:46 allsky [INFO] MainProcess allsky.detectNight() #800: Sun altitude: 48:01:09.1
May  9 16:00:46 allsky [INFO] MainProcess allsky.detectMoonMode() #817: Moon altitide: 30:56:48.0, phase 58.1%
May  9 16:00:46 allsky [INFO] MainProcess indi.getCcdTemperature() #345: Sensor temperature: 52.3
May  9 16:00:56 allsky [INFO] MainProcess allsky.run() #571: Camera last ready: 103.1s
May  9 16:00:56 allsky [INFO] MainProcess allsky.run() #572: Exposure state: ALERT
May  9 16:00:56 allsky [INFO] MainProcess allsky.detectNight() #800: Sun altitude: 47:59:08.8
May  9 16:00:56 allsky [INFO] MainProcess allsky.detectMoonMode() #817: Moon altitide: 30:58:47.7, phase 58.1%
May  9 16:00:56 allsky [INFO] MainProcess indi.getCcdTemperature() #345: Sensor temperature: 53.1
pi@allsky:~ $ htop
pi@allsky:~ $ pi@allsky:~ $`

`2, in __init__#012    self.camera_id = self.getLatestCamera()#012  File "/home/pi/indi-allsky/indi_allsky/flask/views.py", line 244, in getLatestCamera#012    return latest_camera.id#012AttributeError: 'NoneType' object has no attribute 'id'
May  9 15:38:12 allsky [INFO] MainProcess glogging.info() #264: Worker reloading: config.json modified
May  9 15:38:12 allsky [INFO] MainProcess views.dispatch_request() #1070: Wrote new config.json
May  9 15:38:13 allsky [INFO] MainProcess glogging.info() #264: Worker exiting (pid: 13068)
May  9 15:38:15 allsky gunicorn.indi_allsky.wsgi.error: [13811] Booting worker with pid: 13811
May  9 15:46:10 allsky gunicorn.indi_allsky.wsgi.error: [712] Starting gunicorn 20.1.0
May  9 15:46:10 allsky gunicorn.indi_allsky.wsgi.error: [712] Listening at: unix:/var/lib/indi-allsky/gunicorn-indi-allsky.sock (712)
May  9 15:46:10 allsky gunicorn.indi_allsky.wsgi.error: [712] Using worker: gthread
May  9 15:46:10 allsky gunicorn.indi_allsky.wsgi.error: [994] Booting worker with pid: 994
May  9 15:49:35 allsky [INFO] MainProcess glogging.info() #264: Worker reloading: config.json modified
May  9 15:49:35 allsky [INFO] MainProcess views.dispatch_request() #1070: Wrote new config.json
May  9 15:49:36 allsky [INFO] MainProcess glogging.info() #264: Worker exiting (pid: 994)
May  9 15:49:38 allsky gunicorn.indi_allsky.wsgi.error: [2120] Booting worker with pid: 2120
May  9 15:50:20 allsky [WARNING] MainProcess forms.__init__() #944: No images found in DB
May  9 15:56:29 allsky [INFO] MainProcess views.dispatch_request() #1490: Command return: None
May  9 15:56:29 allsky gunicorn.indi_allsky.wsgi.error: [712] Handling signal: term
May  9 15:56:30 allsky [INFO] MainProcess glogging.info() #264: Worker exiting (pid: 2120)
May  9 15:56:32 allsky gunicorn.indi_allsky.wsgi.error: [712] Shutting down: Master
May  9 15:58:30 allsky gunicorn.indi_allsky.wsgi.error: [713] Starting gunicorn 20.1.0
May  9 15:58:30 allsky gunicorn.indi_allsky.wsgi.error: [713] Listening at: unix:/var/lib/indi-allsky/gunicorn-indi-allsky.sock (713)
May  9 15:58:30 allsky gunicorn.indi_allsky.wsgi.error: [713] Using worker: gthread
May  9 15:58:30 allsky gunicorn.indi_allsky.wsgi.error: [947] Booting worker with pid: 947
May  9 16:03:37 allsky [INFO] MainProcess glogging.info() #264: Worker reloading: config.json modified
May  9 16:03:37 allsky [INFO] MainProcess views.dispatch_request() #1070: Wrote new config.json
May  9 16:03:38 allsky [INFO] MainProcess glogging.info() #264: Worker exiting (pid: 947)
May  9 16:03:40 allsky gunicorn.indi_allsky.wsgi.error: [2733] Booting worker with pid: 2733`

aaronwmorris, 2022-05-09:
You probably need to restart the indi-allsky processes.  I would just do a full reboot.

aaronwmorris, 2022-05-10:
In the v2.0 release, I made some changes to the database schema and this is probably what caused issues for you.  Re-running the setup script will perform the necessary database migrations.

maphilli14, 2022-05-10:
Should I re clone?  I ran setup 2x times and rebooted.  I rm -rf ‘ed the whole clone and re-git’ed it and the script ended as: 1) indi_altair_ccd        9) indi_fishcamp_ccd    17) indi_nikon_ccd       25) indi_simulator_ccd2) indi_apogee_ccd       10) indi_fli_ccd         18) indi_nncam_ccd       26) indi_sony_ccd3) indi_asi_ccd          11) indi_fuji_ccd        19) indi_orion_ssg3_ccd  27) indi_starshootg_ccd4) indi_asi_single_ccd   12) indi_gphoto_ccd      20) indi_pentax_ccd      28) indi_sv305_ccd5) indi_atik_ccd         13) indi_inovaplx_ccd    21) indi_playerone_ccd   29) indi_sx_ccd6) indi_canon_ccd        14) indi_mallincam_ccd   22) indi_qhy_ccd         30) indi_toupcam_ccd7) indi_dsi_ccd          15) indi_mi_ccd          23) indi_qsi_ccd         31) indi_v4l2_ccd8) indi_ffmv_ccd         16) indi_nightscape_ccd  24) indi_sbig_ccd        32) indi_webcam_ccdSelect an INDI driver: 3Select an INDI driver: 3**** Setting up indiserver service ******** Setting up indi-allsky service ******** Setting up gunicorn ******** Enabling services ******** Setup policy kit permissions ******** Ensure user is a member of the systemd-journal group ******** Setup rsyslog logging ******** Indi-allsky config ****Detected image folder: /var/www/html/allsky/images**** Flask config ******** Start apache2 service ****Updating certificates in /etc/ssl/certs...0 added, 0 removed; done.Running hooks in /etc/ca-certificates/update.d...done.Module rewrite already enabledModule headers already enabledConsidering dependency setenvif for ssl:Module setenvif already enabledConsidering dependency mime for ssl:Module mime already enabledConsidering dependency socache_shmcb for ssl:Module socache_shmcb already enabledModule ssl already enabledModule proxy already enabledConsidering dependency proxy for proxy_http:Module proxy already enabledModule proxy_http already enabledSite 000-default already disabledSite default-ssl already disabledSite indi-allsky already enabledSynchronizing state of apache2.service with SysV service script with /lib/systemd/systemd-sysv-install.Executing: /lib/systemd/systemd-sysv-install enable apache2**** Setup HTDOCS folder ******** Setup image folder ******** Setup DB ******** Backup DB prior to migration ****/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/paramiko/transport.py:236: CryptographyDeprecationWarning: Blowfish has been deprecated  "class": algorithms.Blowfish,ERROR [flask_migrate] Error: Can't locate revision identified by ***@***.***:~/indi-allsky $ Gi1/0/13  Thanks, Mike From: Aaron W MorrisSent: Monday, May 9, 2022 8:59 PMTo: aaronwmorris/indi-allskyCc: maphilli14; AuthorSubject: Re: [aaronwmorris/indi-allsky] Screen is blank but logs appear to be capturing (Issue #193) In the v2.0 release, I made some changes to the database schema and this is probably what caused issues for you. Re-running the setup script will perform the necessary database migrations.

maphilli14, 2022-05-10:
This is confusing.  It seems allsky is still running can controlling the camera vs indi-allsky:
[image]

see it's updating timestamp, all my end of night ran just fine with legacy allsky as well...

[image]

maphilli14, 2022-05-10:
Seems old allsky service needed to be disabled, all my scripts ran despite multiple upgrades to indi-allsky

[image]

maphilli14, 2022-05-10:
It's working now, do you want to close this?

maphilli14, 2022-05-11:
> If everything is working, we can close this.
> 
> If you do not mind, how did you end up resolving this?

I disabled the allsky service and rebooted.  It seems to have fixed the camera 'ownership' which gave initial positive results but now it hangs despite trying to do the fix listed here:

> If you deleted your initial checkout, then you probably need to clear the flag in the DB related to migrations.
> 
> ```
> sqlite3 /var/lib/indi-allsky/indi-allsky.sqlite "DELETE FROM alembic_version;"
> ```
> 
> And re-run the setup.sh script.

I'm attaching some logs in hope it helps but here's some interesting console outputs after a fresh power up:
```

pi@allsky:~ $
Message from syslogd@allsky at May 11 13:33:01 ...
 kernel:[ 1140.621332] Internal error: Oops: 5 [#1] SMP ARM

Message from syslogd@allsky at May 11 13:33:01 ...
 kernel:[ 1140.621946] Process (lighttpd) (pid: 4735, stack limit = 0x72dc4d3a)

Message from syslogd@allsky at May 11 13:33:01 ...
 kernel:[ 1140.621960] Stack: (0x963fdde0 to 0x963fe000)

Message from syslogd@allsky at May 11 13:33:01 ...
 kernel:[ 1140.621979] dde0: b08ffcc0 00000000 803bb91c 00000000 00000000 00000000 00000000 80f05008

Message from syslogd@allsky at May 11 13:33:01 ...
 kernel:[ 1140.621999] de00: 000000e0 9175d800 9175d800 b08ffcc0 00000001 80f05008 963fde5c 963fde28

Message from syslogd@allsky at May 11 13:33:01 ...
 kernel:[ 1140.622018] de20: 803bb91c 80319b64 b08ffcc0 80f05008 963fde5c 11bc50bf 80333d84 87504a00
```

[indi-allsky.log-20220511.txt](https://github.com/aaronwmorris/indi-allsky/files/8672646/indi-allsky.log-20220511.txt)
[console-logs-20220511.txt](https://github.com/aaronwmorris/indi-allsky/files/8672650/console-logs-20220511.txt)
[indi-allsky.log](https://github.com/aaronwmorris/indi-allsky/files/8672651/indi-allsky.log)
[indi-allsky.log-20220511.txt](https://github.com/aaronwmorris/indi-allsky/files/8672652/indi-allsky.log-20220511.txt)
[webapp-indi-allsky.log-20220511.txt](https://github.com/aaronwmorris/indi-allsky/files/8672656/webapp-indi-allsky.log-20220511.txt)

maphilli14, 2022-05-11:
The whole Pi seems to hang regularly after being online for ~5min or so.  I'm not sure that leaves enough time to complete the upgrade:

```
2.8)
Requirement already satisfied: PyWavelets>=1.1.1 in ./virtualenv/indi-allsky/lib/python3.9/site-packages (from scikit-image->ccdproc->-r /home/pi/indi-allsky/requirements_debian11.txt (line 10)) (1.3.0)
1) indi_altair_ccd        7) indi_dsi_ccd         13) indi_inovaplx_ccd    19) indi_orion_ssg3_ccd  25) indi_simulator_ccd   31) indi_v4l2_ccd
2) indi_apogee_ccd        8) indi_ffmv_ccd        14) indi_mallincam_ccd   20) indi_pentax_ccd      26) indi_sony_ccd        32) indi_webcam_ccd
3) indi_asi_ccd           9) indi_fishcamp_ccd    15) indi_mi_ccd          21) indi_playerone_ccd   27) indi_starshootg_ccd
4) indi_asi_single_ccd   10) indi_fli_ccd         16) indi_nightscape_ccd  22) indi_qhy_ccd         28) indi_sv305_ccd
5) indi_atik_ccd         11) indi_fuji_ccd        17) indi_nikon_ccd       23) indi_qsi_ccd         29) indi_sx_ccd
6) indi_canon_ccd        12) indi_gphoto_ccd      18) indi_nncam_ccd       24) indi_sbig_ccd        30) indi_toupcam_ccd
Select an INDI driver: 3
**** Setting up indiserver service ****
**** Setting up indi-allsky service ****
**** Setting up gunicorn ****
**** Enabling services ****
**** Setup policy kit permissions ****
**** Ensure user is a member of the systemd-journal group ****
**** Setup rsyslog logging ****
**** Indi-allsky config ****
Detected image folder: /var/www/html/allsky/images
**** Flask config ****
!!! WARNING - lighttpd is active - This might interfere with apache !!!
**** Start apache2 service ****
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
Module rewrite already enabled
Module headers already enabled
Considering dependency setenvif for ssl:
Module setenvif already enabled
Considering dependency mime for ssl:
Module mime already enabled
Considering dependency socache_shmcb for ssl:
Module socache_shmcb already enabled
Module ssl already enabled
Module proxy already enabled
Considering dependency proxy for proxy_http:
Module proxy already enabled
Module proxy_http already enabled
Site 000-default already disabled
Site default-ssl already disabled
Site indi-allsky already enabled
Synchronizing state of apache2.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable apache2
**** Setup HTDOCS folder ****
**** Setup image folder ****
**** Setup DB ****
**** Backup DB prior to migration ****
/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/paramiko/transport.py:236: CryptographyDeprecationWarning: Blowfish has been deprecated
  "class": algorithms.Blowfish,
  Generating /home/pi/indi-allsky/migrations/versions/38c161267f41_.py ...  done
/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/paramiko/transport.py:236: CryptographyDeprecationWarning: Blowfish has been deprecated
  "class": algorithms.Blowfish,

```

I might want to try a clean raspbian install if you are not sure what to do.

aaronwmorris, 2022-05-11:
I think you are just running out of memory.  This is a common issue on the RPi 3 with 1GB of memory.  If there is a GUI running, you can disable [the GUI] with this command and reboot.

```
sudo systemctl set-default multi-user.target
```

This can restore the GUI.
```
sudo systemctl set-default graphical.target
```

maphilli14, 2022-05-12:
That seems to have fixed it but the end of night might have crushed it.  Is there a toggle to turn off timelapse?

aaronwmorris, 2022-05-12:
Yes, there is an option "Enable Timelapse Creation" that is normally enabled.  You can disable this and restart indi-allsky.

I also have a wiki page of some memory optimizations you can perform.
https://github.com/aaronwmorris/indi-allsky/wiki/Memory-Optimizations

maphilli14, 2022-05-12:
Will try those tweaks!  Does the timelapse flag only kill the timelapse or keogram as well?  Thanks much for all this!

maphilli14, 2022-05-20:
I'm in the process of swapping out the Pi3 with a Pi4 as the 3 is just giving me fits no matter what!

aaronwmorris, 2022-05-20:
That is probably not a bad move.  1GB of memory is frankly just barely enough in most cases and it does not take much to push it over the edge.

My Rpi 3 devices also occasionally lock up for unexplained reasons.  I always enable the watchdog to ensure they get automatically rebooted.

https://github.com/aaronwmorris/indi-allsky/wiki/Raspberry-Pi-Watchdog