aaronwmorris/indi-allsky #195: Wont run a full day.

Dentonknifeworks, 2022-05-10:
The setup is a PI4 4 GB HQ, Cam 
Have set up a swap file  site of 1024
Installed your softare
Installed INDI Library including all 3rd party drivers:
Set config to camera-specific settings the main one being 50% on image size.
What is happening is the Indi-all-sky shuts down. This morning the only way I could get it going was to disable the moon with 91 in the moon Altitude. Most of the time I just reboot but I am trying to figure out what the issues are.
I attached the log file
Help
[indi-allsky.log](https://github.com/aaronwmorris/indi-allsky/files/8661011/indi-allsky.log)

aaronwmorris, 2022-05-11:
That is very strange.  According to the log, indi-allsky is receiving a TERM signal to shutdown which is normal when stopped from systemd.

Is it possible you had two instances of indi-allsky running simultaneously?

Dentonknifeworks, 2022-05-11:
Well, I don't think so. I did the install fresh and then a reboot.
This morning when I found Indi not running I disabled the Moon item and then could start Indi using the web interface. It been running all day and I have got a Kellogram but not a timelapse yet. Its been about 1 hour since the transition from day to night. I do have the WEB gui up on two computers at times but cant see that doing it. I will check it in the morning and report what i find.

Dentonknifeworks, 2022-05-11:
Ok got a timelapse. I will see if I get the night one in the morning. So Not sure why None shows up when I click on the Kellogram the timelapse opens. 
[image]

Dentonknifeworks, 2022-05-11:
Ok got the keg and timelapse from last night. I am going to turn on the moon mode and sell how things go.
Still not getting a thumbnail on the WEB page for timelapse
I am thinking that the thumbnail should be in the same directory as the rest.
Is this something I should see an error in the log?
Not seen a Star Trail yet but I figure that's because it's inside the office.[image]
?

Dentonknifeworks, 2022-05-12:
Well. it failed with moonset.  Logg attached I disabled it for today's run to see what happens.
Bra
[error.txt](https://github.com/aaronwmorris/indi-allsky/files/8677549/error.txt)
d

Dentonknifeworks, 2022-05-12:
Larger part of the log

[error2 copy.txt](https://github.com/aaronwmorris/indi-allsky/files/8678709/error2.copy.txt)

aaronwmorris, 2022-05-12:
Here are the key lines...

May  9 23:03:46 raspberrCAM [INFO] MainProcess indi.serverDisconnected() #175: Server disconnected (exit code = -1, localhost, 7624
May  9 23:03:47 raspberrCAM [INFO] MainProcess allsky.shoot() #994: Taking 0.66148210 s exposure (gain 16)
May  9 23:03:47 raspberrCAM [WARNING] ImageWorker001 allsky.sigterm_handler() #206: Caught TERM signal, shutting down
May  9 23:03:47 raspberrCAM [WARNING] FileUploader001 allsky.sigterm_handler() #206: Caught TERM signal, shutting down
May  9 23:03:47 raspberrCAM [WARNING] VideoWorker001 allsky.sigterm_handler() #206: Caught TERM signal, shutting down

What this tells me is something is stopping the indiserver first and then something is sending the TERM signal to indi-allsky to stop.  It looks like something on your images is stopping services.

Dentonknifeworks, 2022-05-12:
This is a clean install with nothing else running.
I see what you are saying.
Let's see what we get tonight so far I have found turning off Moon mode works It doesn't add up.
I will entertain anything you would like me to check

aaronwmorris, 2022-05-12:
With 4GB of RAM, it is unlikely you are running out of memory.

Just to double check, you are running vanilla Raspbian Buster (Legacy)?

Dentonknifeworks, 2022-05-12:
pi@raspberrCAM:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrCAM:~ $

aaronwmorris, 2022-05-12:
Is it possible your Raspberry PI is rebooting?

I tested with one of my test systems and if I kill the indiserver process, it will produce the disconnect error message.  However, in your logs, the indi-allsky process is getting the SIGTERM signal, as well.  It appears that something is stopping both services, first indiserver, then indi-allsky.  A SIGTERM signal is the default signal systemd uses to stop child processes.

Dentonknifeworks, 2022-05-12:
Hmm, let's see where can I check that? maybe system logs?

aaronwmorris, 2022-05-12:
If you have not manually rebooted, just check the system `uptime`

Dentonknifeworks, 2022-05-12:
pi@raspberrCAM:~ $ uptime
 13:32:51 up 1 day,  7:05,  2 users,  load average: 1.30, 1.31, 1.13
pi@raspberrCAM:~ $

aaronwmorris, 2022-05-12:
Keep an eye on it... if it happens again, check the uptime to see if the system rebooted.

Dentonknifeworks, 2022-05-12:
I have done some rebooting I will not and will see if that is the case

Dentonknifeworks, 2022-05-13:
Well, ok I have now disproven Moon mode doing it. Something is sending the Term. This log shows it doing it a lot last night. The pi did not reboot. It did take a reboot this morning to get INDI server back running.
I can redo the system but this would be the 3rd time.

Dentonknifeworks, 2022-05-13:
[allsky.log.zip](https://github.com/aaronwmorris/indi-allsky/files/8686471/allsky.log.zip)

aaronwmorris, 2022-05-13:
Try updating the rpi firmwares....

```
sudo rpi-update
```

aaronwmorris, 2022-05-13:
I am going to setup my Raspi 4 with my HQ camera and re-test things for a while.

Dentonknifeworks, 2022-05-13:
I put together a second Pi 4 2GB today I will let it run also.

Dentonknifeworks, 2022-05-14:
rpi-update did not do it got the term again tonight.

aaronwmorris, 2022-05-14:
My Rpi with the HQ cam ran all night without a problem.

Do you run the desktop image or the lite Raspbian image?

Dentonknifeworks, 2022-05-14:
I run a Desktop
Mine crashed again.
The second one worked all night the same setup Desktop, Buster.
Am I running the wrong image?

aaronwmorris, 2022-05-14:
No, you are not running the wrong image, the desktop image should be fine, I am just looking for differences.

Have you double checked the physical wiring and the camera cable?

Dentonknifeworks, 2022-05-14:
Well yes I even switched out the ribbon cable.

Dentonknifeworks, 2022-05-16:
Well, the second Pi is running with no issues. I put TJ's software back in the first Pi and it's running with no issues.
So we're going to run with that. Thanks for your help not sure what the issue was but looks to be a one-off. Feel free to close this.
Brad

Dentonknifeworks, 2022-05-18:
Well, I took the second pi outside last night and got the term signal. Everything on this is different hardware. The same method of installation. 
[indi-allsky.log.zip](https://github.com/aaronwmorris/indi-allsky/files/8715861/indi-allsky.log.zip)

aaronwmorris, 2022-05-18:
I just re-deployed my test camera using the full buster image with GUI and running as the pi user.  I will let this run for a while.

Dentonknifeworks, 2022-05-23:
Any issues on your end?

aaronwmorris, 2022-05-23:
My camera ran for 4 days straight until it looks like something related to the camera crashed (completely different error).

aaronwmorris, 2022-06-04:
I just checked my test system again.... 10 days of continuous uptime.

Dentonknifeworks, 2022-06-04:
Starting a new install today
Going to install indi driver from https://www.indilib.org/get-indi/download-rpi.html
Then your install it and see what happens.

Dentonknifeworks, 2022-06-05:
Ran yesterday and overnight but no time-lapses.
I have the Scaling at 50 are there other settings I need to set in the Config file?

aaronwmorris, 2022-06-05:
The only difference in my case is I normally set the scaling to 75%.

Dentonknifeworks, 2022-06-08:
Well, 3 days no issues.
What is HUP indi-allsky?

aaronwmorris, 2022-06-08:
That is good news!

HUP causes indi-allsky to re-read the config file and apply/change most of the settings in realtime without a full process restart.  Camera gain and bin settings will not change, but almost all of the other settings take effect immediately.

Dentonknifeworks, 2022-06-09:
Lets close this. Thanks for the help.

closed 2022-06-09 by Dentonknifeworks