aaronwmorris/indi-allsky #205: Remote cam usage

jkoenig72, 2022-05-25:
Hi,

very interesting project. Will try it asap.

I see you mention remote cam usage. That would bring lots of boards with small RAM into consideration, right? Like older orange pi with 512mb ram.

I will give it a try over the weekend…

Again congratulations for this project!

aaronwmorris, 2022-05-26:
That is correct!  indi-allsky is just an indi network client, so it can connect to a camera plugged into a different system running an indiserver.

jkoenig72, 2022-05-26:
Did a quick test. Using an old nanopi m1. 

Changed from localhost to that IP. Within indiweb on that machine I selected Telescope Simulator and cam in Web Manager.

Looks good... ;-)

This could be an interesting case - then the cam only needs a very small SMB, 256mg should do it. All the heavy processing can be done in some VM.

aaronwmorris, 2022-06-27:
Closing this issue for now.  Let me know if there is anything else I can do.

closed 2022-06-27 by aaronwmorris