aaronwmorris/indi-allsky #208: Feature request: planetarium overlay with constellations and planetary objects.

maphilli14, 2022-05-28:
This is now defunct and broken of my own doing, but is something like this in the works natively in the package?

https://maphilli14.github.io/allskyplanetarium/

TIA

Mike

aaronwmorris, 2022-05-28:
This is definitely something I have been interested in doing.  I will add it to my list.

maphilli14, 2022-05-28:
Should I leave this open?

aaronwmorris, 2022-05-28:
Yeah, let's leave this open for now.

aaronwmorris, 2022-05-30:
Well, my first pass at trying to integrate VirtualSky failed miserably.  I think part of the problem I am having is I am already using a canvas element to show images (instead of using an <img> tag).  The canvas is how I update the image without any noticeable flicker.

evripidis, 2024-02-27:
I would like just to bump it up, if anyone came up with an idea :-)

clambert24, 2024-05-29:
I also like this idea if possible

RainerMinixhofer, 2025-07-10:
I like it very much as well. Automatic plate solving and extracting the camera orientation from it, would be needed for a good overlay and is an interesting problem on it's own. Furthermore a good idea about the image distortion due to the dome may be needed to get a good overlay as well...