aaronwmorris/indi-allsky #209: Set Dat/Time Feature

HxPI, 2022-05-29:
Can a feature to set date/time be added? For some reason the RPi4 doesn’t save the date/time when power is removed, which is how I normally store it. I was unable to set the time and the Timelapse changed the gain in the middle of the night. Lost about a third of the exposures. Thanks.

aaronwmorris, 2022-05-29:
I will work on this.

HxPI, 2022-05-29:
It would be really useful for the workflow to be the following:

1. Power on RPi
2. Connect tablet to hotspot and login through web interface
3. Goto Config tab
4. Select “Flush image database”, I manually delete images after every session
5. Select “Set Date”, perhaps date info can be retrieved from tablet?
6. Select “Set Time”, perhaps time info can be retrieved from tablet?
7. Let session run
8. At the end of the session, select “Shutdown System”

That’s it. Image capture is running properly and images are retrieved at the end of the session. Day, night, and moon gain changes will occur and the proper times and labels will be correct!

aaronwmorris, 2022-05-30:
I merged #210 to update the system clock via the web interface.  You will have to re-run setup.sh to grant the permissions for this action.

I will work on the flush action soon.

HxPI, 2022-06-18:
Ok thanks. This feature is working as expected. Will close this issue.

aaronwmorris, 2022-06-18:
I have just merged #224 to allow you to flush all of the images from the system.  This will only delete individual frames, but will not touch timelapses, keograms, and star trails.

HxPI, 2022-06-18:
Does it flush raw images also? If not no big deal. I assume raw images are not tracked in the database? Also, what will happen if the command is executed when the images are already deleted from the storage? Will only the database be flushed or will it cause an error? I usually delete images from the storage after I transfer them to my computer but the database flush requires CLI action, which I don’t normally do in the field. Thanks.

aaronwmorris, 2022-06-18:
The raw images would not be deleted since they are not tracked in the DB.

If you pre-delete the images, the flush images should still function properly.  It only deletes the files if the exist on the filesystem and will skip them if not found.  Finally, the DB entries would be removed.

I have a CLI script for validating the DB entries, I can look at implementing the same function in the web interface.

HxPI, 2022-06-28:
This would be great. Thanks.

aaronwmorris, 2022-07-01:
Merged #247.  Added functions to flush timelapses, keograms, and startrails.  Added new function to validate DB entries from web interface.

aaronwmorris, 2022-08-21:
Closing this issue for now.  Reopen if you need anything.

closed 2022-08-21 by aaronwmorris