aaronwmorris/indi-allsky #214: Overlay External Information

beckrd, 2022-06-02:
I would like to overlay information from a python script I have running on the pi.  The script monitors the CPU and enclosure temperature (using a DHT22) and runs a fan if temperature thresholds are met.  With Thomas Alquin's Allsky, I put the information in an html file and used a php include to add this information to his "public.php" page.  

I could also envisage grabbing ambient conditions from my Davis weather gauge (already captured for INDI use).  Any recommendations or feature enhancement would be appreciated.

aaronwmorris, 2022-06-02:
I will take a look to see how they do that and see if I can do something comparable.

aaronwmorris, 2022-06-03:
Can you post the example code or a screen shot to see how it is presented?

beckrd, 2022-06-03:
Here is my added line to public.php:

` <?php include("/home/pi/allsky/temp+fan.html"); ?>`

It's placed between the </script> and the </body> lines near the end of the file.

With this in place, the public.php displays the image with the text of the file displayed as follows

Date
CPU Temperature = xx.x C
Case Temperature = xx.x C
Fan is (on or off, depending on the temperatures)

aaronwmorris, 2022-06-04:
What do you think about having the data in a text file and including each line in the status under the info at the bottom right?
[image]

beckrd, 2022-06-04:
[AllSky_Public_Page.pdf](https://github.com/aaronwmorris/indi-allsky/files/8837569/AllSky_Public_Page.pdf)
I restarted TA's Allsky and printed out the page displayed  by my custom public.php.

Your suggestion of adding user content to the info at the bottom would likely be a good idea for other users as well.

aaronwmorris, 2022-06-04:
Merged #216 to include data in the web view.  The config option is the config form.

The text file can contain HTML tags if you desire.

```
CPU Temp: 20.1C
Case Temp: <span style="color:#ff0000;">800C</span>
Fan: Off
```

[image]

closed 2022-06-04 by beckrd