aaronwmorris/indi-allsky #215: Day/Night Timelapse Enable and Image Capture

HxPI, 2022-06-04:
The day/night timelapse generation mode switch setting doesn’t seem to be functioning as expected. When day timelapse is disable, no day images are saved, even when enable day capture is selected. I am assuming this is the case because they don’t show up in the Loop or Image tabs. When I reenable the day timelapse generation, the images would show up in the Loop and Image tabs. The Latest tab shows latest image and works normally in either case.
 
Seems images should still be saved if capture is enabled, even if timelapse generation is disabled. Please take a look at this and see if this is an issue. Check for night image capture also.

aaronwmorris, 2022-06-05:
Just copying my response from CN...

 The way this works is basically the same way TJ's Allsky software works... at least the way I remember (which could be wrong).

With "Daytime Capture" enabled and "Daytime Timelapse" disabled, an image is captured, but only saved as the latest image.  Historical images are not saved and there is no metadata to display in the graphs.

However, if you disable "Enable Timelapse Creation" and enable both "Daytime Capture" and "Daytime Timelapse" it should save the images without actually generating the timelapse.

HxPI, 2022-06-18:
OK thanks for the clarification. Will close this issue.

closed 2022-06-18 by HxPI