aaronwmorris/indi-allsky #217: Customise text overlay

StarGeezerPhil, 2022-06-12:
No issue, but I'd love to be able to disable the overlay for some clean images to stack/star trail/etc.

Is it possible for us to customise the text overlay that's added to the images.

It'd be nice to be able to: 
- disable it completely - or only have it apply to e.g. FTP uploaded image, not timelapses and startrails.
- change the date/time format
- add custom text - maybe use templates for the different standard variables (e.g. `%datetime%\nCPU Temperature: %CPUtemp%\nLocation: Somewhere over the rainbow`
- include an image overlay (e.g. transparent png)

Great work - loving this so far. Still weather-testing my build at the moment, but getting some great results with the ASI 224MC.

aaronwmorris, 2022-06-13:
You can disable the Text Overlay completely by selecting "Disabled" for the font.

Changing the date/time format is a good idea, I will look in to that.

Custom text can be added via the "Extra Image Text File" option using an external text file.  I could add support for templating variables, but there are not too many standard variables I could define.  Things like "CPU Temperature" are not standard across different system/SoC types.

As for overlaying an image or the planetarium, I am having some difficulty with this in my testing.  The images in the web interface are rendered using <canvas> element and not a standard <img> tag.  Using the canvas, it allows much smoother transitions between images.  Unfortunately, overlaying canvas elements appears to be a little complicated.  (Front-end design is not really my forte)

StarGeezerPhil, 2022-06-13:
Wow, thanks for coming back to me so quickly! 🔥

Brilliant - sorry, I'd missed the _Font Face: disabled_ setting! 👍

I usually work in PHP, but have some code snippets I'll dig out for capturing a canvas element to an image in case that helps.

Alternatively, if you can point me to which file/where the image overlay part is done I'm happy to have a tinker here too.

aaronwmorris, 2022-06-13:
In order to use the Text Overlay, you just use the "Extra Image Text File" option in the config.  You can point this to a text file on the file system which can be generated by an external process.  Each line in this file will be added to the meta-data in the upper left hand part of the image.

The index page template for the home view is located at https://github.com/aaronwmorris/indi-allsky/blob/master/indi_allsky/flask/templates/index.html .

aaronwmorris, 2022-06-16:
Merged #222 to customize datetime format.

StarGeezerPhil, 2022-06-19:
Thanks for coming back again.

Unfortunately disabling overlay through the font also removes the x-axis labels on the keogram - be great if there were separate options to disable overlay on images vs. keogram.

aaronwmorris, 2022-06-19:
I will take a look at this.

aaronwmorris, 2022-06-25:
Merged #227 to disable image label and keogram labels separately.  I also remove the "Disabled" option in the font face settings since it no longer really made sense.

aaronwmorris, 2022-08-21:
Closing this issue for now.  Let me know if I can help with anything else.

closed 2022-08-21 by aaronwmorris