aaronwmorris/indi-allsky #225: Night imaging stopped

HxPI, 2022-06-19:
Tried the update merge #224 but there was an issue. Day images saved but as soon as night began imaging stopped. There was only one image saved for night and no further images were taken, including the latest image. Something has caused the system to stop taking night images when going from day to night. Please look into this. Thanks.

aaronwmorris, 2022-06-19:
Restart the software and let it run.  Let me know if it happens again.

aaronwmorris, 2022-06-20:
I have experienced this issue, as well.

Sometimes it happens when new code is applied without restarting the processes.

More often, this type of failure can happen every few months where changing the gain on the camera (like between day and night) will fail.  I do not know if the failure happens in the indi code or if it is within the driver for the camera.  Unfortunately, the only way to recover is a restart.  In some cases, the entire system might have to be rebooted.

HxPI, 2022-06-21:
I’ll test again and report back. Thanks.

HxPI, 2022-06-25:
It did the same thing again. Something is wrong with the system causing it to stop imaging transitioning from day to night. Luckily this time I was able to restart and not lose anything.

aaronwmorris, 2022-06-25:
Can you send me the contents of your config file at `/etc/indi-allsky/config.json`.  Perhaps the issue is related to a setting you are using.

If you could post the log around the time of the transition from day to night, that might help.

HxPI, 2022-06-25:
Ok. Where are the logs located?

aaronwmorris, 2022-06-25:
The logs are located at `/var/log/indi-allsky/indi-allsky.log`  There are also rotated and compressed logs in `/var/log/indi-allsky/`

HxPI, 2022-06-25:
Here are the files:
[Archive.zip](https://github.com/aaronwmorris/indi-allsky/files/8985367/Archive.zip)

aaronwmorris, 2022-06-25:
This is the error:

> Jun 24 20:56:02 raspberrypi [INFO] MainProcess indi.newMessage() #210: new Message 2022-06-25T00:56:02: [ERROR] Exposure failed after 3 attempts.

This is a failure in the camera.

You may want to do try a full package upgrade and see if that helps.  `apt-get update` and `apt-get upgrade` and see if there are any indi package updates available.  Sometimes when you run setup.sh for an indi-allsky upgrade, it will upgrade some of the indi packages but not all of them.  I would suggest a full reboot if you find any indi package upgrades being applied.

aaronwmorris, 2022-06-27:
I merged #230 to keep the indi camera related packages in sync with the main indi packages.

HxPI, 2022-06-28:
The update did find some changes. Still to test again. Thanks for updating the script to handle this.

aaronwmorris, 2022-08-21:
Closing this issue for now.  Let me know if I can help with anything else.

closed 2022-08-21 by aaronwmorris

pterodyne, 2022-09-30:
Im not sure whether or not to reopen this or create a new issue.  I am using an ASI ZWO 178mm/pi4, and have this issue with transitioning to night.  Strangely I had the same issue with TJacquin's allsky. In both cases turning off daytime imaging solves the problem permanently. BUT of course no daytime images.  Is this an SDK issue? Seems strange that it would be present in indi and a SDK native application. In the other project people suggested a hardware problem, but this setup will operate for months, night after night without issue as long as I'm not transitioning from day to night.  Seems to point to software, libraries, sdk or algorithms.  At any rate, I don't really need daytime images. It would be nice if it displayed a NO IMAGES DAYTIME or something during the day if disabled. Mine just makes an over exposed image all day, maybe not good for the sensor, but who knows. It's been deployed for years.

aaronwmorris, 2022-09-30:
@pterodyne  That is very strange.  Can you open a new issue and post your config file?