aaronwmorris/indi-allsky #228: [ERROR] MainProcess uploader.unhandled_exception() #32: Value: 'task_id'

ASUK-Development, 2022-06-26:
I just got the following error when trying to capture darks for the first time.

Build: Current
OS: Raspberry Pi OS
SBC: Pi 3B
Camera: SV305

`2022-06-26 11:02:52,853 [ERROR] MainProcess uploader.unhandled_exception() #32: Value: 'task_id'
[2022-06-26 11:02:52,853] ERROR in uploader: Value: 'task_id'
2022-06-26 11:02:52,857 [ERROR] MainProcess uploader.unhandled_exception() #37: '  File "/home/pi/indi-allsky/./darks.py", line 87, in <module>\n    action_func()\n'
[2022-06-26 11:02:52,857] ERROR in uploader: '  File "/home/pi/indi-allsky/./darks.py", line 87, in <module>\n    action_func()\n'
2022-06-26 11:02:52,858 [ERROR] MainProcess uploader.unhandled_exception() #37: '  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 305, in sigmaclip\n    self._run(IndiAllSkyDarksSigmaClip)\n'
[2022-06-26 11:02:52,858] ERROR in uploader: '  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 305, in sigmaclip\n    self._run(IndiAllSkyDarksSigmaClip)\n'
2022-06-26 11:02:52,859 [ERROR] MainProcess uploader.unhandled_exception() #37: '  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 430, in _run\n    self._take_exposures(exposure, dark_filename_t, bpm_filename_t, ccd_bits, stacking_class)\n'
[2022-06-26 11:02:52,859] ERROR in uploader: '  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 430, in _run\n    self._take_exposures(exposure, dark_filename_t, bpm_filename_t, ccd_bits, stacking_class)\n'
2022-06-26 11:02:52,860 [ERROR] MainProcess uploader.unhandled_exception() #37: '  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 490, in _take_exposures\n    hdulist = self._wait_for_image()\n'
[2022-06-26 11:02:52,860] ERROR in uploader: '  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 490, in _take_exposures\n    hdulist = self._wait_for_image()\n'
2022-06-26 11:02:52,861 [ERROR] MainProcess uploader.unhandled_exception() #37: '  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 234, in _wait_for_image\n    task_id = i_dict[\'task_id\']\n'
[2022-06-26 11:02:52,861] ERROR in uploader: '  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 234, in _wait_for_image\n    task_id = i_dict[\'task_id\']\n'
`

I'm not sure if this is common or just me doing something daft, but I wanted to report it.

aaronwmorris, 2022-06-26:
This is my fault.  I updated some code last week and I forgot to update the darks processing.  I will get this fixed today.

aaronwmorris, 2022-06-26:
Merged #229 .  Sorry about this.

aaronwmorris, 2022-07-26:
I am going to close this issue for now.  Let me know if there is anything else I can do for you.

closed 2022-07-26 by aaronwmorris