aaronwmorris/indi-allsky #234: Experience with ASI224mc

ennui2342, 2022-06-28:
Aaron,

I've moved over from Thomas' allsky since I'm an indi man for the rest of my astro setup and I'm amazed by what you've done to get to parity so fast. This really is a great piece of work. Here's some feedback points from a fresh setup with ASI224mc:

* The ASI224mc works, but has the same problem it seems you see with other cameras where if the default and min exposure are zero it starts off with a flat return which meets the ADU criteria and keeps the system at lowest exposure. I'm finding that I need to have some value in default is helping getting it started properly.
* Your colours for text and the orbs are stated to be r,g,b but seem to be b,g,r
* It took me a while to realise that saving config also then needs a HUP to be applied. I would suggest this should be done automatically
* I always have had max exposure on 60s (not sure why anyone would have it on anything else?). However the period between exposures is automatically lengthened to the max exposure if the latter is bigger. (ie I have 15s as period, but this is reset to 60 as that is my max exposure). This new period applies even if the exposure isn't taking the max amount, which I didn't expect - I thought the behaviour would be to use whichever was larger would be the time between the start of exposures. I can see keeping this fixed is useful though - on my old setup timelapses would speed up or slow down based on varying exposure as clouds went past.
* It would be useful to have the equivalent of Thomas' public.php available somewhere so an external system can iframe in the current image with a javascript reloader built in - I have personally used this for iframing an updating image into grafana

A final thought - with respect to your jSQM ADU count. It might be possible to get the lux of an image as described in https://stackoverflow.com/questions/4876315/determining-image-luminance-brightness/4876528#4876528 and from the lux you can then get to the sky quality. This is some python I wrote for weatherradio to use a light (lux) sensor to measure sky quality and integrate the result into a weewx weatherstation converting between lux and SQM:

`# Sky quality formula based on http://unihedron.com/projects/darksky/magconv.php
def luxtomag(x): return math.log10(x/108000)/-0.4
weewx.units.default_unit_format_dict['sky_quality'] = '%.1f'
weewx.units.default_unit_label_dict['sky_quality'] = u" visual mag/arcsec²"
weewx.units.conversionDict['sky_quality'] = {'lux': lambda x : 10.8e4 * 10**(-0.4*x) }
weewx.units.conversionDict['lux'] = {'sky_quality': luxtomag }`

The nice thing about having SQM in mag/arcsec^2 is that then it can be pumped back into indi as WEATHER_LIGHT like with https://github.com/rkaczorek/indi-weather-mqtt/blob/main/indi-weather-mqtt.cpp so it could be used in applications like kstars for observatory control.

-Mark

aaronwmorris, 2022-06-28:
Thank you for the feedback!  I am happy that the software is useful to people.

> The ASI224mc works, but has the same problem it seems you see with other cameras where if the default and min exposure are zero it starts off with a flat return which meets the ADU criteria and [...]

I have seen this happen on one of my other cameras.  I have some theories as to why this happens, but I have found that taking dark frames helps.  Have you taken any dark frames?

> Your colours for text and the orbs are stated to be r,g,b but seem to be b,g,r

<sigh> Yeah, that looks to be the case.  I fed the color array directly into CV2.  If you know anything about OpenCV, it processes colors as BGR by default.  I will fix this.

> I always have had max exposure on 60s (not sure why anyone would have it on anything else?). However the period between exposures is automatically lengthened to the max exposure if the latter is [...]

I had also done this with TJs software, but I never liked the result.  Let me take a look if I can change my logic to allow a shorter period than the exposure so the exposures start immediately.  

> It would be useful to have the equivalent of Thomas' public.php available somewhere so an external system can iframe in the current image with a javascript reloader built in - I have personally used this for iframing an updating image into grafana

Let me see what I can do about this.

> It would be nice to have a setting to control the speed of the exposure correction to match ADU. I'm finding in the day with clouds moving overhead it's taking too long to ramp up and down to meet the changing light.

You can probably achieve this by reducing the `Target ADU Deviation` setting.  indi-allsky decides to adjust the exposure when the average ADU of 5-6 frames exceeds the target brightness by +/- the deviation.  If you reduce the deviation threshold, it should adjust faster (increasing should cause it to happen less often).

> A final thought - with respect to your jSQM ADU count. It might be possible to get the lux of an image as described in [...]

This is really cool.  I have been looking for a starting point for doing this conversion to a somewhat real magnitude, but I could not quite wrap my head around the math (and I am a math guy).  I even had some conversations with Frank Freestar8n (author of Metaguide) about this same topic, but I could just not find a good starting point.  Maybe this is it, I will take a look at this!

aaronwmorris, 2022-06-28:
Merged #236 to fix the RGB/BGR color issue

aaronwmorris, 2022-06-29:
Merged #237 with new /public URL that does not require authentication

ennui2342, 2022-06-29:
> Thank you for the feedback! I am happy that the software is useful to people.
>

No problem at all - can't believe you're faster with the pull requests than I am with a comment reply!

> 
> I have seen this happen on one of my other cameras. I have some theories as to why this happens, but I have found that taking dark frames helps. Have you taken any dark frames?

No dark frames yet which I do need to do - after a couple of years outside and constantly on my little camera is looking worse for wear! I'll have a play when I've done that and see if it makes a difference.  I set the default to 0.000033 which seems to get me started so no problems - just wasn't initially obvious that should be the fix.

> 
> I had also done this with TJs software, but I never liked the result. Let me take a look if I can change my logic to allow a shorter period than the exposure so the exposures start immediately.
>

I think your way of doing it is the better way, now I'm getting used to it - the speeding and slowing of timelapses isn't really desirable. The main problem is really one of expectations on this field since its overridden if you enter a value less than the max exposure. It would be more obvious to the user if that was validated in the form, or that this field became time between frames (rather than start to start) which is what I initially assumed it was.
 
> You can probably achieve this by reducing the `Target ADU Deviation` setting. indi-allsky decides to adjust the exposure when the average ADU of 5-6 frames exceeds the target brightness by +/- the [...]

Thanks, this does seem to be improving things, thanks.
 
> This is really cool. I have been looking for a starting point for doing this conversion to a somewhat real magnitude, but I could not quite wrap my head around the math (and I am a math guy). I even [...]

Great. The uncertain bit is getting lux from an image. I'm surprised by the suggestion that it could be done in opencv without specific camera information - I would have thought it depended on things like the QE. But I haven't done any experimenting.

Also, I've noticed your star trails seems to perform much better - I could never get this to reliably work in my old setup. I've always wondered though whether an exposure threshold would produce better results than an ADU threshold. Given autoexposure, that means the ADU is potentially a moving target, but if you could assume any exposure above x seconds is clear then that might be interesting to see if the results are better.

-Mark

aaronwmorris, 2022-06-29:
> I think your way of doing it is the better way, now I'm getting used to it - the speeding and slowing of timelapses isn't really desirable. The main problem is really one of expectations on this [...]

In this case, I think I found a bug in my code.  This might have prevented the setting from being changed.  I was using the wrong variable name when populating the web field and this probably always reset the value back to 15.0.  I will be merging another PR shortly with these changes.

> Great. The uncertain bit is getting lux from an image. I'm surprised by the suggestion that it could be done in opencv without specific camera information - I would have thought it depended on things like the QE. But I haven't done any experimenting.

Not only QE, but also the spectrum sensitivity for the camera.  If you want to use a camera to get an absolute SQM value, you really want an IR-Cut filter to block invisible spectrum, that way you only have to deal with QE for the visible range.  Some newer cameras have increased response well into the invisible IR range (ASI462mc, et al) which would make this value pretty wacky.  The best we can hope for is a relative SQM value.

> Also, I've noticed your star trails seems to perform much better - I could never get this to reliably work in my old setup. I've always wondered though whether an exposure threshold would produce [...]

That is really nice to hear.  The star trails and keogram functionality were things I had to completely reimplement in Python.  Even though TJ's implementations are written in OpenCV, they are C based and the C version of OpenCV works differently from the Python version.  The differences are mainly in the array handling, C uses cv::Mat and Python uses Numpy.

I have a few enhancements from TJ's star trail generator.  Mainly that I do not use the whole image for ADU calculations, but only the central part of the image.  Very few people have a full 180 degree view.  There is some additional thresholds with the number of individual pixels exceeding a certain ADU.

The star trails processing is basically implementing a programmatic "guess" at which frames should be used.

I could also change the star trails to base using images on star count, which would be interesting if you only wanted true "star trails", but some of my most interesting star trail images had no stars at all.

I call this one the "q-tip".  The full lunar eclipse left a "moon trail" in the image with flares on each side.
[image]

aaronwmorris, 2022-06-29:
Merged #238 to enhance exposure period configuration.

aaronwmorris, 2022-06-29:
Update to the RGB/BGR issue, I merged #239 due to a bug I introduced

aaronwmorris, 2022-06-29:
And merged #241 because I did not test things properly.  :-)

aaronwmorris, 2022-07-01:
> I was pondering the star count for things like trails and measuring sky quality, and was wondering how much latitude washes out the signal with the fact that different parts of the sky were moving through the (generally central) window with different star densities. Ideally a circular windows around the pole star would provide a fixed star count to work from, but like in your picture that isn't necessarily in view for many setups.

Even my star counts are not perfect.  The algorithm I am using does a pretty good job of detecting actual stars, but it is not perfect.  My color cameras always detect 25% more stars than my monochrome camera which I attribute to the algorithm detecting some of the chrominance noise from the debayering as stars.

Also, one of the reasons I only perform star detection in the central region of the image--gaps in the tree leaves where the sky/background shows through will be detected as "stars".

Even with these problems, the star detection is a very accurate indicator of sky quality and cloudiness.  I went into detail comparing two different methods against some test images ranging from completely clear to complete cloud cover in my indilib post.  https://indilib.org/forum/general/10619-new-all-sky-camera-management-software-indi-allsky.html?start=12

The star count is meant more of an indicator of sky quality rather than a true count of the stars.  While the number of stars will vary over the night and from frame to frame, sharp/sudden changes of star counts are clear indication of cloud cover.

aaronwmorris, 2022-08-21:
Closing this issue for now.  Let me know if you need anything else.

closed 2022-08-21 by aaronwmorris