aaronwmorris/indi-allsky #244: Feature request: meteor detection

ennui2342, 2022-06-30:
It would be nice to automatically detect meteors as part of an allsky camera image pipeline. I've heard of this being done from a separate windows machine running https://allskeye.com/. I also had a play with the https://globalmeteornetwork.org/ software to process a stack of fits images with their RMS software, but the way this works is designed around video capture so was a dead end (https://globalmeteornetwork.groups.io/g/main/topic/79381433).

The most promising project for an allsky camera that might be integratable is https://github.com/dwkim78/ASTRiDE - I haven't played with this at all, but you could imagine it being part of the pipeline. Another route would be to train an ML model, there are a lot of published research papers and some public code around like https://github.com/satmonkey/MeteorDL but again, I haven't had time to play to see how well they perform with allsky data.

aaronwmorris, 2022-07-01:
I have been planning on implementing something like this for a while.  I already have a field in the database to record detected events.... I just need to detect events now.

My current plan is just to implement some simple line detection.  Time will tell if that is sufficient.

aaronwmorris, 2022-07-02:
I started testing some meteor detection code which is just simple line detection like I mentioned previously.  I do not know if I will stick with this code, but I managed to detect a meteor on the first night!

Frankly, it is probably better at detecting planes than meteors, but this is pretty cool.

The number beside the time is the number of lines detected in the central part of the frame.  (The algorithm will return multiple lines for each detection.)

[image]

ennui2342, 2022-07-02:
Ooh that's looking pretty good! I did a bit of playing last night with the Astride library (https://github.com/dwkim78/ASTRiDE) which works by boundary tracing. I got it up and running, but I don't have enough space on my card to switch on saving a nights fits so I'll have to wait for a bigger card to properly give it a run through and try tweaking the settings. The main thing I wanted to get running was to see if I could add windowing so I could remove false positives from the scenery.

aaronwmorris, 2022-07-06:
I have merged #248 with the meteor detection code.  It is disabled by default and must be explicitly enabled in the config.

ennui2342, 2022-07-07:
Upgraded without problems - just waiting for a clear night...

aaronwmorris, 2022-07-08:
I just merged #250  to filter images that have active detections in the image viewer.

aaronwmorris, 2022-08-21:
Closing this issue for now.  Let me know if you need anything else.

closed 2022-08-21 by aaronwmorris