aaronwmorris/indi-allsky #245: Feature request: Expiry of content

ennui2342, 2022-07-01:
My raspberry pi is very light on storage so with my TJ setup I kept only a few days on it and primarily used the scp to push everything to a nas for long term archiving. I assumed the expire parameter in the indi-allsky config set the same behaviour, but I now realise this is for images written out not the days stored in sqllite. Would it be possible to add a parameter that limits the number of days of data stored in sqllite. Thanks.

aaronwmorris, 2022-07-01:
I will work on this.

aaronwmorris, 2022-07-01:
Merged #246 to allow expiring old timelapses, keograms, and star trails.  The default age is 365 days.

ennui2342, 2022-07-01:
Perfect, thanks for adding this - it'll tide me over until I picked up a new big SD card on prime day :)

closed 2022-07-01 by ennui2342