aaronwmorris/indi-allsky #252: Tuning questions

ennui2342, 2022-07-11:
Hi Aaron,

Below is a nice daytime video in relatively clear skies. For reference my ADU window is 475,275 - 575,375 a patch in the top left of the view. The minimum exposure time of the camera is 0.000032s according to the tech specs. Target ADU is 150, deviation 5.

As you can see once the sun rises the system is managing exposure well, then it hits a phase where the exposure time is bouncing around causing flickering. Then it stabilises but over the rest of the day it steps up in brightness and drifts away from the nice exposure value at the start of the day. This is somewhat counterintuitive as the ADU window is top left so as the sun moves closer and finally over the window I would have expected the exposure to ramp down not up.

Do you have any tips for improving the behaviours here - overall this is much smoother than I had with the TJ setup, but if there's optimisation I can do I thought it worth asking. Thanks.

-Mark

https://www.dropbox.com/s/v3f66zuchv5y1ws/allsky-timelapse_ccd2_20220710_day.mp4?dl=0

aaronwmorris, 2022-07-11:
Yeah, I call that "exposure flapping".  I occasionally see the same thing.  The main cause is tiny exposure changes during the day time can cause significant changes in brightness.

I recall I recommended lowering the deviation in your other issue.  Unfortunately, lowering the deviation will exacerbate the tendency to exposure flap during the day.

I have some ideas, let me think about it for a few days and play with some settings.

ennui2342, 2022-07-11:
I did have a thought about the exposure getting brighter when intuition says it should dim as the sun moves into the adu window. The timelapse has flip horizontally and vertically set, so does that mean the adu window is actually bottom right?

aaronwmorris, 2022-07-11:
Yes, that is correct.  I need to mention this in the config form.

ennui2342, 2022-07-11:
> I could also move the flip code above the line and star detection code to ensure the ROI matches what the user is seeing.  I kind of like this idea.

Yeah that would make sense since most users will want to use the latest image to measure where the ROI should be.

-Mark

aaronwmorris, 2022-07-12:
Merged #253 to change the image processing order of operations.

I also created a wiki page to document the order of operations.
https://github.com/aaronwmorris/indi-allsky/wiki/Image-order-of-operations

aaronwmorris, 2022-07-13:
Merged #257 to add a new option to have separate daytime deviation settings.

aaronwmorris, 2022-07-15:
@StarGeezerPhil That was before I changed the order of operations.  The image is now flipped prior to any processing or detection.  If you upgrade to the latest code, you can use the output image as a reference for ROI.

aaronwmorris, 2022-07-16:
Merged #266 to draw the detections and ROI regions in the image for calibration.  Below is an example.

The outer box is the detection region for SQM, stars, and lines.  The inner box is the ADU detection region.

Circles are detected stars and lines are draw on top of detected lines (airplane trail in this case).

[image]

aaronwmorris, 2022-08-21:
Closing this issue for now.  Let me know if you need anything else.

closed 2022-08-21 by aaronwmorris