aaronwmorris/indi-allsky #259: You have to install this via the GUI / desktop

petely, 2022-07-13:
I was building this on a RPi 4. 
One thing I discovered is that you actually have to be logged in to the GUI to get this code to build successfully.
At first I was just SSH'd in and running from the command line.
It built (once all dependencies were satisfied, some were a bit tricky) and got as far as asking me to
**Select an INDI driver**
After that the setup.sh script ran a bit more then terminated as follows:
```
**** Setting up indiserver service ****
**** Setting up indi-allsky service ****
**** Setting up gunicorn ****
**** Enabling services ****
Failed to connect to bus: No such file or directory

The script exited abnormally, please try to run again...
```

However if I used VNC / Nomachine to connect to the RPI desktop and ran the same setup.sh script, it ran through to completion.
Hopefully this tip will help others.

aaronwmorris, 2022-07-14:
Merged #263 to detect the dbus user session and indicate to the user if they need to re-login.

linuxkidd, 2022-07-17:
I had this same issue.. turns out, if you `sudo` to the user account, you don't get dbus setup.  I had to SSH in as the actual user, not root... then run the installer.

aaronwmorris, 2022-07-17:
When you sudo, did you leave out the `-`?  If so, this could be the cause.

When you sudo, you should `sudo su - username`.  The dash simulates a full login of the user with a unique session.

linuxkidd, 2022-07-17:
sorry, I should have said that I `su - pi` to get into the account, and that was preventing dbus from working.

At least in my instance.

aaronwmorris, 2022-07-26:
I am going to close this issue for now.  Let me know if there is anything else I can do for you.

closed 2022-07-26 by aaronwmorris