aaronwmorris/indi-allsky #260: Error generating star trail

Dentonknifeworks, 2022-07-13:
This is on a new install.
The error code is #408 Not enough images found to build star trail.
Timelaps is good and so is the Kegogram.

Dentonknifeworks, 2022-07-13:
Sample image
[image]

[image]
e

aaronwmorris, 2022-07-14:
This is not unusual for full moon nights.  Most (or all of the frames) end up being excluded because they are too bright.  The exclusion algorithm is pretty aggressive to ensure only the darkest images are used.  You could try raising the `Star Trails Max ADU` setting to something like 80 to try to get more images included.

In the situation above, you may also considering setting the `SQM/Star Region of Interest` setting to only encompass open sky in the image.  The brightness in the trees could cause all of the images to be excluded.

In fairness, in cases where no images are selected, I should add at least one image so an image is generated for startrails.

aaronwmorris, 2022-07-14:
Merged #261 so that there is always at least one image used for star trails.

Dentonknifeworks, 2022-07-14:
That works!
Looks much better than a broken Icon image.
Thanks

closed 2022-07-14 by Dentonknifeworks