aaronwmorris/indi-allsky #277: Issues with remote indiserver

jkoenig72, 2022-07-30:
Hi,

I love this project! Great work! 

I like the possibility of having those small older boards - for here its 3 orangepizeros - as a remote indiserver and connecting from a more powerful - in my case a NUC with more mem - server machine.  Then use LXD containers to run multiple instances. 

Works  nicely - but a few questions/issues:

- I'm using a Zwo ASI 178 MC, an older T7 (ASI120MC clone), and a Touptek  G3CMOS06300KPA (those also have an imx178 color inside)
- I have 3 LXD containers running - all on the same host and all point to 3 different orangepizero with the latest indi 1.9.7 with web manager. 2x times  ZWO CCD, one-time Toupcam driver all together with Telescope Simulator. All connections are LAN based, I want to remove any wifi stability issue from the situation for now.

The ASI 178 runs very stable. But I have issues with the t7 and also with the Touptek. After some time I always encounter situations where the cam seems to be busy and not taking any more new pictures. This seems to go endless.  

...
Jul 30 11:23:50 allsky2 [INFO] MainProcess allsky._queueManualTasks() #1435: Checking for manually submitted tasks
Jul 30 11:24:00 allsky2 [INFO] MainProcess allsky.run() #605: Camera last ready: 8775.1s
Jul 30 11:24:00 allsky2 [INFO] MainProcess allsky.run() #606: Exposure state: BUSY
Jul 30 11:24:00 allsky2 [INFO] MainProcess allsky.detectNight() #847: Sun altitude: 47:45:50.3
Jul 30 11:24:00 allsky2 [INFO] MainProcess allsky.detectMoonMode() #864: Moon altitide: 35:23:30.4, phase 2.8%
Jul 30 11:24:00 allsky2 [WARNING] MainProcess indi.getCcdTemperature() #418: Sensor temperature: not supported
Jul 30 11:24:00 allsky2 [INFO] MainProcess allsky._queueManualTasks() #1435: Checking for manually submitted tasks
Jul 30 11:24:10 allsky2 [INFO] MainProcess allsky.run() #605: Camera last ready: 8785.2s
Jul 30 11:24:10 allsky2 [INFO] MainProcess allsky.run() #606: Exposure state: BUSY
Jul 30 11:24:10 allsky2 [INFO] MainProcess allsky.detectNight() #847: Sun altitude: 47:47:04.3
Jul 30 11:24:10 allsky2 [INFO] MainProcess allsky.detectMoonMode() #864: Moon altitide: 35:24:56.8, phase 2.8%
Jul 30 11:24:10 allsky2 [WARNING] MainProcess indi.getCcdTemperature() #418: Sensor temperature: not supported
Jul 30 11:24:10 allsky2 [INFO] MainProcess allsky._queueManualTasks() #1435: Checking for manually submitted tasks
Jul 30 11:24:20 allsky2 [INFO] MainProcess allsky.run() #605: Camera last ready: 8795.2s
Jul 30 11:24:20 allsky2 [INFO] MainProcess allsky.run() #606: Exposure state: BUSY
Jul 30 11:24:20 allsky2 [INFO] MainProcess allsky.detectNight() #847: Sun altitude: 47:48:18.2
Jul 30 11:24:20 allsky2 [INFO] MainProcess allsky.detectMoonMode() #864: Moon altitide: 35:26:23.1, phase 2.8%
Jul 30 11:24:20 allsky2 [WARNING] MainProcess indi.getCcdTemperature() #418: Sensor temperature: not supported
Jul 30 11:24:20 allsky2 [INFO] MainProcess allsky._queueManualTasks() #1435: Checking for manually submitted tasks
Jul 30 11:24:30 allsky2 [INFO] MainProcess allsky.run() #605: Camera last ready: 8805.3s
Jul 30 11:24:30 allsky2 [INFO] MainProcess allsky.run() #606: Exposure state: BUSY
...

When I restart with systemctl --user restart  indi-allsky is runs for some time  then again runs into this busy situation.

My questions / what I tried...

1.) When I run with a remote indiserver, I can disable the local one with something like "systemctl --user disable indiserver.service". After that systemctl --user restart  indi-allsky takes a long time. 

2.) Best color settings for the toupcam - auto white does not really work, still very green pic... ;-)

3.) As the ASI Cam is working stable it must have something to do with the config of cams and/or model? I applied the fix recommended for ASI 120MC:

{
    "PROPERTIES": {},
    "SWITCHES": {
        "CCD_VIDEO_FORMAT": {
            "on": [
                "ASI_IMG_RAW16"
            ],
            "off": [
                "ASI_IMG_RAW8"
            ]
        }
    }
}

sadly no effect.

4.) In general the indiserver on the smaller orangepizeros works stable - I was using EKOS to test:
* shutdown the related container instance for that cam
* connect with ekos to the remote orangepizero (I tested both not working stable with indi-allsky, t7 and Toupcam)
* take a picture, put that into a loop in EKOS - this runs for hours - no issue.

Somehow indi-allsky does something different compared to EKOS to get into this "busy" state. 

Any idea how to overcome this? Did anybody get a T7/Toupcam stable? - and yes, t7 needs darks... ;-).

Thanks!

Regards
Joerg

jkoenig72, 2022-07-30:
One mistake I made and I just detect- when you clone containers in LXD, you have to run 

lxc config unset <containername> volatile.eth0.hwaddr

Otherwise, you get the same IP. I will see if this helps and report back.

Regards
Joerg

aaronwmorris, 2022-07-30:
You would have to go back in the logs to when the cameras first get stuck in the BUSY state.  The cameras are always "busy" when taking an exposure, so it is not an unusual state.

The most common circumstance for being stuck in the BUSY state is the camera throws an exception when trying to take an exposure right after changing the gain between day and night.

For the color balance issues with the Touptek camera, you might be able to enable auto white balance in the camera itself.  The camera may also manual white balance settings for each channel.  I see in my example properties, there are also black balance settings, as well.

You can try this to enable auto white balance in the camera.  I have never tested this before as this is a Touptek specific setting.
```
{
    "PROPERTIES": {},
    "SWITCHES": {
        "CCD_AUTO_CONTROL" : {
            "on" : ["TC_AUTO_WB"]
        }
    }
}
```

For the T7c, I have one of these cameras myself and I never really experienced any problems with it, although I have not done extensive testing with it.  One thing you need to do is apply the Linux compatibility firmware to the camera, you can find this on the ZWO website.

jkoenig72, 2022-07-30:
Thanks Aaron - I will try those Toupcam settings. 

Again congrats on your fantastic work - really love it!

jkoenig72, 2022-07-30:
Btw. is there a way to buy you a coffee? ;-) Paypal or something?

aaronwmorris, 2022-07-30:
Hah, maybe I should set something up, but I am not worried about the money.  I just want to help people see their own sky!

Eventually, I will purchase a Touptek camera (and an Altair) so I can do more testing.  Each vendor has their own quirks.  As you can tell, ZWO's drivers are really, really good (with the exception of the ASI120).

jkoenig72, 2022-07-30:
Set something up… I know it not the money- but to some level it will feel good. If you ever come here to beautiful Germany, to the Duesseldorf/Cologne area - ping me, and I will get you a few beers. I mean it. 👍

You do a useful thing here… 😀

jkoenig72, 2022-07-30:
Zwo does great products, I agree. Ping those vendors- they should sent you some items for your testing. You drive their sales …

aaronwmorris, 2022-09-16:
Merged #355 with some new SCNR algorithms to help remove green bias from images.

aaronwmorris, 2022-12-07:
There was a recent update to the Touptek driver (which is the basis for the Altair driver) which corrected a problem with one shot color cameras.

https://github.com/indilib/indi-3rdparty/issues/681

This fix was included with the recent indi 1.9.9 release.

aaronwmorris, 2023-02-17:
Closing this issue for now.

closed 2023-02-17 by aaronwmorris