aaronwmorris/indi-allsky #278: Extra temperature sensor

jkoenig72, 2022-07-30:
Hi,

I‘m currently using a toupcam model does not contain an internal temperature sensor. Adding something to a raspberry or similar is not an issue. Maybe it would worth thinking about such an reading could be easily included.

Regards 
Jörg

aaronwmorris, 2022-07-30:
If you can get the temperature written to a text file, you can use the `Extra Image Text File` setting to include that text in the image text label.  This is similar to TJ's allsky software.

Similarly, there is a setting `Extra HTML Info File` which will include the text in the information box at the lower right hand side of the indi-allsky interface.

jkoenig72, 2022-08-01:
Does such a value also be included in allsky.py - when selecting darks etc.? Or this is just additional info for the screen?

If the 2nd is true - would it not be good to check that file when no temperature value is coming back from the cam?

aaronwmorris, 2022-08-01:
I see, you are interested in using the temperature for dark frame calibration.  This is something I would have to work on.

jkoenig72, 2022-08-02:
If I get the code right - that would be the place...

    def getCcdTemperature(self, ccdDevice):
        temp = ccdDevice.getNumber("CCD_TEMPERATURE")

        if isinstance(temp, type(None)):
            logger.warning("Sensor temperature: not supported")
            temp_val = -273.15  # absolute zero  :-)
        else:
            temp_val = float(temp[0].value)
            logger.info("Sensor temperature: %0.1f", temp_val)

        with self.sensortemp_v.get_lock():
            self.sensortemp_v.value = temp_val

        return temp_val

Like - if the temperature is not supported look up a file for the value and use that instead

Then use a small python script and reading & writing the value of a simple temp sensor like dallas ds18b20 to that file ever y e.g 10 sec. Stick that sensor close to the camera sensor. Maybe to the lens case. That's metal. 

Then add the location of that file to your config parameters. 

Something like that could work?

aaronwmorris, 2022-08-02:
I was thinking about adding a new config variable for a path to a script that would output a single temperature value.  This script could be used for both the indi-allsky program as well as the darks processing, both of which need access to temperature data.

aaronwmorris, 2022-08-30:
Merged #336 to support an external temperature measurement for cameras.

aaronwmorris, 2022-09-07:
Merged #350 with a new method for communicating the data.  If you need help updating a script, let me know.

https://github.com/aaronwmorris/indi-allsky/wiki/External-Temperature-Sensor

aaronwmorris, 2022-10-22:
Closing this issue for now.  Let me know if there are any problems with this.

closed 2022-10-22 by aaronwmorris