aaronwmorris/indi-allsky #282: Install issue: ../../meson.build:41:0: ERROR: Unknown compiler(s):...

StarGeezerPhil, 2022-08-03:
New install and experiencing new issues:

This time (Pi 4 4gb, Waveshare IMX378-190, Pi PoE hat) I needed to also run the build_Indi script, which seemed all ok.

Have also run apt-get update/upgrade, but setup.py fails each time at the Collecting scipy step.

It installs dependencies ok, but then fails with sub-process error  (Getting requirements to build wheel ... error // error: subprocess-exited-with-error), looks the error step is:
../../meson.build:41:0: ERROR: Unknown compiler(s): [['gfortran'], ['flang'], ['nvfortran'], ['pgfortran'], ['ifort'], ['g95']]

Any encountered/fixed this issue?

`× Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [61 lines of output]
      The Meson build system
      Version: 0.62.2
      Source dir: /tmp/pip-install-day5c5xt/scipy_21c79712eb0641029c877d3a173e0622
      Build dir: /tmp/pip-install-day5c5xt/scipy_21c79712eb0641029c877d3a173e0622/.mesonpy-a5esqvey/build
      Build type: native build
      Project name: SciPy
      Project version: 1.9.0
      C compiler for the host machine: cc (gcc 10.2.1 "cc (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110")
      C linker for the host machine: cc ld.bfd 2.35.2
      C++ compiler for the host machine: c++ (gcc 10.2.1 "c++ (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110")
      C++ linker for the host machine: c++ ld.bfd 2.35.2
      Host machine cpu family: arm
      Host machine cpu: armv7l
      Compiler for C supports arguments -Wno-unused-but-set-variable: YES
      Library m found: YES

      ../../meson.build:41:0: ERROR: Unknown compiler(s): [['gfortran'], ['flang'], ['nvfortran'], ['pgfortran'], ['ifort'], ['g95']]
      The following exception(s) were encountered:
      Running "gfortran --version" gave "[Errno 2] No such file or directory: 'gfortran'"
      Running "gfortran -V" gave "[Errno 2] No such file or directory: 'gfortran'"
      Running "flang --version" gave "[Errno 2] No such file or directory: 'flang'"
      Running "flang -V" gave "[Errno 2] No such file or directory: 'flang'"
      Running "nvfortran --version" gave "[Errno 2] No such file or directory: 'nvfortran'"
      Running "nvfortran -V" gave "[Errno 2] No such file or directory: 'nvfortran'"
      Running "pgfortran --version" gave "[Errno 2] No such file or directory: 'pgfortran'"
      Running "pgfortran -V" gave "[Errno 2] No such file or directory: 'pgfortran'"
      Running "ifort --version" gave "[Errno 2] No such file or directory: 'ifort'"
      Running "ifort -V" gave "[Errno 2] No such file or directory: 'ifort'"
      Running "g95 --version" gave "[Errno 2] No such file or directory: 'g95'"
      Running "g95 -V" gave "[Errno 2] No such file or directory: 'g95'"

      A full log can be found at /tmp/pip-install-day5c5xt/scipy_21c79712eb0641029c877d3a173e0622/.mesonpy-a5esqvey/build/meson-logs/meson-log.txt
      + meson setup --native-file=/tmp/pip-install-day5c5xt/scipy_21c79712eb0641029c877d3a173e0622/.mesonpy-native-file.ini -Ddebug=false -Doptimization=2 --prefix=/usr /tmp/pip-install-day5c5xt/scipy_21c79712eb0641029c877d3a173e0622 /tmp/pip-install-day5c5xt/scipy_21c79712eb0641029c877d3a173e0622/.mesonpy-a5esqvey/build
      Traceback (most recent call last):
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
          main()
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 130, in get_requires_for_build_wheel
          return hook(config_settings)
        File "/tmp/pip-build-env-xyv95hr3/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 909, in get_requires_for_build_wheel
          with _project(config_settings) as project:
        File "/usr/lib/python3.9/contextlib.py", line 117, in __enter__
          return next(self.gen)
        File "/tmp/pip-build-env-xyv95hr3/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 888, in _project
          with Project.with_temp_working_dir(
        File "/usr/lib/python3.9/contextlib.py", line 117, in __enter__
          return next(self.gen)
        File "/tmp/pip-build-env-xyv95hr3/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 547, in with_temp_working_dir
          yield cls(source_dir, tmpdir, build_dir)
        File "/tmp/pip-build-env-xyv95hr3/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 463, in __init__
          self._configure(reconfigure=bool(build_dir) and not native_file_mismatch)
        File "/tmp/pip-build-env-xyv95hr3/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 494, in _configure
          self._meson(
        File "/tmp/pip-build-env-xyv95hr3/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 477, in _meson
          return self._proc('meson', *args)
        File "/tmp/pip-build-env-xyv95hr3/overlay/lib/python3.9/site-packages/mesonpy/__init__.py", line 472, in _proc
          subprocess.check_call(list(args))
        File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['meson', 'setup', '--native-file=/tmp/pip-install-day5c5xt/scipy_21c79712eb0641029c877d3a173e0622/.mesonpy-native-file.ini', '-Ddebug=false', '-Doptimization=2', '--prefix=/usr', '/tmp/pip-install-day5c5xt/scipy_21c79712eb0641029c877d3a173e0622', '/tmp/pip-install-day5c5xt/scipy_21c79712eb0641029c877d3a173e0622/.mesonpy-a5esqvey/build']' returned non-zero exit status 1.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

The script exited abnormally, please try to run again...`

aaronwmorris, 2022-08-03:
I have been working with @RCompassi in #184 on this same issue.

I just merged #283 to see if this is sufficient to build scipy.

StarGeezerPhil, 2022-08-03:
Ah brilliant -sorry, hadn't spotted the other issue and didn't come up with keywords I searched 🤣

I'll take a look - is there a fix/workaround?

aaronwmorris, 2022-08-03:
Try pulling down the latest update and re-run the setup.sh script again.  It has additional packages it will install.

aaronwmorris, 2022-08-03:
Let's keep this issue here.  The other issue is more about issues on Orange Pi/Armbian.

StarGeezerPhil, 2022-08-03:
Can confirm it's progressed past the errors... thanks for the incredibly quick response/update!

...which leads me to a query for the waveshare imx378 camera - do you know if this should be a v4l2 driver?

aaronwmorris, 2022-08-03:
> ...which leads me to a query for the waveshare imx378 camera - do you know if this should be a v4l2 driver?

That is a really good question.  Please try the v4l2 driver.  It might even work with the indi_rpicam driver, but I have my doubts.

If neither work, I have been considering my options for supporting a non-indi compatible camera.  I have a few ideas, but it may take a while for me to work something out.

StarGeezerPhil, 2022-08-03:
Have tried indi_v4l2_ccd, but service has failed numerous times. I'll try rpi next, but which option is this - can't see it in the list?

aaronwmorris, 2022-08-03:
If you installed the 64-bit version of buster, the rapcam driver is not built as it is not compatible with 64-bit ARM.

StarGeezerPhil, 2022-08-03:
No, it's raspberry pi os 32-bit and I'm not seeing the option:
[image]

aaronwmorris, 2022-08-03:
Did you install the overlay for the camera?  (I am assuming it is needed)

Interesting.  Maybe I should be passing another option to the indi build to build the rpicam driver.  I have only ever tested it on bullseye (Legacy).

aaronwmorris, 2022-08-03:
I was referring to the dtoverlay in the waveshare instructions.
https://www.waveshare.com/wiki/IMX378-190_12.3MP_Camera

The camera port may need to be enabled too.  The setup.sh script does this automatically when you select indi_rpicam (but you cannot do that, obviously)

StarGeezerPhil, 2022-08-03:
hmm... following their instructions, but no joy yet. Running headless so may need a screen or vlc to experiment with live previews to test.

aaronwmorris, 2022-08-03:
You may need to ensure your user is a member of the video group.  You can run `id` to verify.

If not:
```
sudo usermod -a -G video USERNAME
```

StarGeezerPhil, 2022-08-04:
Thank you again Aaron for all the support and quick responses.

Not joy yet, but setup.sh no longer completes - think I'll start with a fresh image again.

Don't know if you recognise this error or why it's arisen, but setup script doesn't complete with the following error:

```
**** Setup image folder ****
**** Setup DB ****
**** Backup DB prior to migration ****
ERROR [flask_migrate] Error: Could not determine revision id from filename a0b143c668c9_.py. Be sure the 'revision' variable is declared inside the script (please see 'Upgrading from Alembic 0.1 to 0.2' in the documentation).

The script exited abnormally, please try to run again...
```

aaronwmorris, 2022-08-04:
Your database has gotten out of sync with the migrations.  I added some instructions to this page:  https://github.com/aaronwmorris/indi-allsky/wiki/Database-migrations

StarGeezerPhil, 2022-08-05:
Brill - yes that sorted the dB issue thank you.

I've tried a variety of drivers, but no joy with the camera and raspi cam hasn't appeared as an option.

I tested with the original (non-Indi) allsky and all works OK - think that uses raspihq camera mode with libcamera.

I'm at a dark location, so will use that temporarily, but would prefer to use your version if we can get the camera to work.

Thank you again for all the support - and the rapid features and updates you're continuing to release.

aaronwmorris, 2022-08-05:
I appreciate the feedback.  Other than problems, I do not get a lot of feedback on who is actually using indi-allsky and if they even like it.  :-)

As far as uninstalling, you can disable the services and delete the checkout, but disabling the services is enough to prevent a conflict.

```
systemctl --user disable indiserver
systemctl --user disable indi-allsky
```

aaronwmorris, 2022-08-05:
I went ahead and created a wiki topic for the uninstall.

https://github.com/aaronwmorris/indi-allsky/wiki/Uninstall-indi-allsky

aaronwmorris, 2022-08-18:
Hey @StarGeezerPhil I just merged #299 to [hopefully] support the imx378 camera.  If you would like to test this, I would appreciate the feedback.

Fulhair, 2022-08-22:
Sorry to jump of this thread, but I have just installed indi-allsky and trying out with the waveshare 378-190. Can get the server up and running but no image is displayed. Any idea on where I might have gone wrong or can fix it. 
Using the libcamera IMX-477 camera interface

Fulhair, 2022-08-22:
Update :  Have it working now. Solution was on the Waveshare set up page 
sudo nano /boot/config.txt
Find "camera-auto-detect=1" and modify it to "camera_auto_detect=0" 
At the end of the file, add the following setting statements 'dtoverlay=imx378'
The image has a weird green tinge but does work!

aaronwmorris, 2022-08-22:
The green bias is common in new cameras.  My guess is libcamera does not have the correct parameters to balance the colors in /usr/share/libcamera/ipa/raspberrypi/imx378.json

My imx477 will also do this with artificial lighting, sunlight tends to do a little better.  I have removed the ir cut filter on my sensor, which throws things out of wack too.

Fulhair, 2022-08-23:
Was wondering that but the basic libcam test on the pi desktop ( "libcamerahello -t 0") gives a normal colored image video image.
I tried the './allsky.sh cameraReport' command but it didn't work

aaronwmorris, 2022-08-23:
It's also possible that the green bias exists due to the image being the raw bayer info.  There is no such thing as "white balancing" when you get the raw image.  I will play with my imx477 and see if I can add the white balance settings in libcamera to see if they can make a difference.

aaronwmorris, 2022-08-23:
The white balance options did not have any effect when using the DNG raw data.  Those settings only affect jpeg and png output.

There are some post-processing white balancing options in indi-allsky that can be used to mitigate the green bias.

aaronwmorris, 2022-10-22:
I mentioned this in a couple of other threads, but I missed this one.  I merged #355 to remove the green bias from the images.  This immensely helps with the IMX477 and IMX378.

closed 2022-12-07 by aaronwmorris