aaronwmorris/indi-allsky #284: SV305Pro and ASI120mcs both not displaying images

laterrible, 2022-08-06:
Hey - thanks for an awesome bit of software. I've just finished setting up a new Pi and a fresh installation of this today, however I cannot get an image to display in the webserver. I'm met instead with a grey screen with both my 305 and 120, with different USB cables and USB ports tried. 

lsusb shows the pi recognises a camera is plugged in.

Both the server and indisky are running/active. I've run setup.sh twice (first time to fix indi-allsky service failing). What else can I try?

Huge thanks!

laterrible, 2022-08-06:
Upon further tinkering, mainly through various Indi sudo updates and upgrade commands, it seems that the Svbony driver package wasn't installed. Now working through methods to resolve that

laterrible, 2022-08-06:
> Upon further tinkering, mainly through various Indi sudo updates and upgrade commands, it seems that the Svbony driver package wasn't installed. Now working through methods to resolve that

Sadly couldn't get anywhere. Ended up with a stack of dependencies that I couldn't force install my way through. Back to square one. A fresh install of raspbian and try again.

I gave Astroberry a go, but couldn't get it to headless install on my network, so skipped that too.

laterrible, 2022-08-06:
Gosh, bit going on here. Upon reinstallation, I was given many, many more options for camera driver installation, including sv305. THis was desipte the installation hanging midway through and requiring a disconnect and restart of ./setup.sh

Now to start the services and see if the camera is functioning.

/edit: great success! Everything working as it should be. Obv first installation was broken.

closed 2022-08-06 by laterrible

aaronwmorris, 2022-08-06:
Very strange you had the initial issues.  I am glad you got it working!