aaronwmorris/indi-allsky #286: ASI120MC-S dark capture 8-bit revert issue

JamesHarrison, 2022-08-09:
Hi,

The ASI120MC-S obviously has a quirk where it falls back to 8-bit mode on gain changes.

I found that when running the darks script, this was happening and the resultant dark files would not be matched in use.

The fix I had for this was to reconfigure the device each shot, i.e.

```python
    def shoot(self, ccdDevice, exposure, sync=True, timeout=None):
        logger.info('Taking %0.8f s exposure (gain %d)', exposure, self.gain_v.value)
        self.indiclient.configureDevice(self.ccdDevice, self.config['INDI_CONFIG_DEFAULTS'])
        self.indiclient.setFrameType(self.ccdDevice, 'FRAME_DARK')

        ctl = self.indiclient.setCcdExposure(ccdDevice, exposure, sync=sync, timeout=timeout)

        return ctl
```

This then produces dark frames correctly. The config snippet from the wiki was applied but didn't fix the issue in the darks script, just the main imaging loop.

May be worth just reconfiguring before each frame by default or based on the device name? In any case thought it worth recording in case anyone else ran into it. Thanks for a fantastic implementation of an ASC!

aaronwmorris, 2022-08-10:
Thanks for catching this!  I will look at the dark frame processing for this issue.

aaronwmorris, 2022-08-10:
Did the frame type get reset to FRAME_LIGHT when the gain value change?

JamesHarrison, 2022-08-11:
Not sure how to spot that - I have noticed that even with the switches set I still get errors in normal operation. Logs look like this:

```
Aug 11 22:20:57 unreliable-witness python3[13269]: [INFO] MainProcess allsky.run() #607: Camera last ready: 25.9s
Aug 11 22:20:57 unreliable-witness python3[13269]: [INFO] MainProcess allsky.run() #608: Exposure state: BUSY
Aug 11 22:20:57 unreliable-witness python3[13269]: [INFO] MainProcess allsky.detectNight() #853: Sun altitude: -13:43:48.1
Aug 11 22:20:57 unreliable-witness python3[13269]: [INFO] MainProcess allsky.detectMoonMode() #870: Moon altitide: 8:10:01.4, phase 99.8%
Aug 11 22:20:57 unreliable-witness python3[13269]: [INFO] MainProcess allsky.detectMoonMode() #874: Moon Mode conditions detected
Aug 11 22:20:57 unreliable-witness python3[13269]: [INFO] MainProcess indi.getCcdTemperature() #422: Sensor temperature: 24.5
Aug 11 22:20:57 unreliable-witness python3[13269]: [INFO] MainProcess allsky._queueManualTasks() #1441: Checking for manually submitted tasks
Aug 11 22:21:02 unreliable-witness python3[13269]: [INFO] MainProcess indi.newMessage() #218: new Message 2022-08-11T21:21:02: [INFO] Switching to 8-bit video.
Aug 11 22:21:02 unreliable-witness python3[13269]: [INFO] MainProcess indi.newSwitch() #205: new Switch CCD_VIDEO_FORMAT for device ZWO CCD ASI120MC-S
Aug 11 22:21:02 unreliable-witness python3[13269]: [INFO] MainProcess indi.newMessage() #218: new Message 2022-08-11T21:21:02: [INFO] Taking a 30 seconds frame...
Aug 11 22:21:02 unreliable-witness kernel: usb 2-1: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
Aug 11 22:21:07 unreliable-witness python3[13269]: [INFO] MainProcess allsky.run() #607: Camera last ready: 36.0s
Aug 11 22:21:07 unreliable-witness python3[13269]: [INFO] MainProcess allsky.run() #608: Exposure state: BUSY
```

aaronwmorris, 2022-08-13:
Merged #292 to fix the issue with ASI120 and dark calibration frame creation.

aaronwmorris, 2022-10-22:
Closing this for now.  Let me know if there is anything else I can do.

closed 2022-10-22 by aaronwmorris