aaronwmorris/indi-allsky #297: SV305 not detected, allsky service failing

laterrible, 2022-08-16:
Hello Aaron it's me again. I've had great success these last few days, capturing lots of wonderful timelapses, thank you for this bit of software. I think I've engineered myself into breaking either my pi, or the software, however.

As background, incase this is causal: I wanted to setup the pi4 to boot from USB, so that I could use a 64gb USB stick I had lying around. So I set about figuring out how to do that, and had some success. However I had a stack of trouble getting the indi-allsky package to properly install (until I landed on using the OS Lite Legacy no desktop OS, which has worked flawlessly).

This at least got allsky to install and run, like normal. However the allsky _service_ keeps failing. 

Running lsusb does not show the camera as plugged in (I've tried several cables and tested that my PC can detect the camera OK), and systemctl status --user indi-allsky shows the following:

indi-allsky.service - Indi AllSky Service
   Loaded: loaded (/home/pi/.config/systemd/user/indi-allsky.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Tue 2022-08-16 12:25:24 AEST; 11s ago
  Process: 2266 ExecStart=/home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --config /etc/indi-allsky/config.json --log
 Main PID: 2266 (code=killed, signal=SEGV)

Aug 16 12:25:22 test python3[2266]: [INFO] MainProcess indi._findCcds() #404: Found device SVBONY SV305PRO 0
Aug 16 12:25:22 test python3[2266]: [INFO] MainProcess indi._findCcds() #409:  Detected ccd
Aug 16 12:25:22 test python3[2266]: [INFO] MainProcess indi._findCcds() #409:  Detected guider
Aug 16 12:25:22 test python3[2266]: [INFO] MainProcess indi.findCcd() #419: Found 1 CCDs
Aug 16 12:25:22 test python3[2266]: [WARNING] MainProcess allsky._initialize() #338: Connecting to device SVBONY SV305PRO 0
Aug 16 12:25:22 test python3[2266]: [INFO] MainProcess miscDb.addCamera() #44: Camera DB ID: 1
Aug 16 12:25:22 test python3[2266]: [INFO] MainProcess indi.configureDevice() #434: Setting switch DEBUG
Aug 16 12:25:22 test python3[2266]: [INFO] MainProcess indi.newSwitch() #217: new Switch DEBUG for device SVBONY SV305PRO 0
Aug 16 12:25:24 test systemd[588]: indi-allsky.service: Main process exited, code=killed, status=11/SEGV
Aug 16 12:25:24 test systemd[588]: indi-allsky.service: Failed with result 'signal'.

Which appears to be a memory allocation issue from my limited stackoverflow reading. Re-running ./setup.sh hasn't fixed it. Any tips?

laterrible, 2022-08-16:
Rebooting, upgrading, updating and re-running status gives me a different error, basically showing it hasn't detected a camera:

indi-allsky.service - Indi AllSky Service
   Loaded: loaded (/home/pi/.config/systemd/user/indi-allsky.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2022-08-16 12:48:05 AEST; 38s ago
  Process: 677 ExecStart=/home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --config /etc/indi-allsky/config.json --log syslog run (code=exited, sta
 Main PID: 677 (code=exited, status=1/FAILURE)

Aug 16 12:48:05 test python3[677]: [INFO] MainProcess indi.findCcd() #419: Found 0 CCDs
Aug 16 12:48:05 test python3[677]: [ERROR] MainProcess allsky.unhandled_exception() #36: An uncaught exception occurred:
Aug 16 12:48:05 test python3[677]: [ERROR] MainProcess allsky.unhandled_exception() #37: Type: <class 'IndexError'>
Aug 16 12:48:05 test python3[677]: [ERROR] MainProcess allsky.unhandled_exception() #38: Value: list index out of range
Aug 16 12:48:05 test python3[677]: [ERROR] MainProcess allsky.unhandled_exception() #43: '  File "allsky.py", line 128, in <module>\n    action_func(*args_list, **
Aug 16 12:48:05 test python3[677]: [ERROR] MainProcess allsky.unhandled_exception() #43: '  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 584, in run\n
Aug 16 12:48:05 test python3[677]: [ERROR] MainProcess allsky.unhandled_exception() #43: '  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 330, in _initia
Aug 16 12:48:05 test python3[677]: [ERROR] MainProcess allsky.unhandled_exception() #43: '  File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 421, in fi
Aug 16 12:48:05 test systemd[599]: indi-allsky.service: Main process exited, code=exited, status=1/FAILURE
Aug 16 12:48:05 test systemd[599]: indi-allsky.service: Failed with result 'exit-code'.

laterrible, 2022-08-16:
Final update: created a new boot USB running 64bit  Debian 11, also from imager. Install went well, web server up and running AND it can see the camera is plugged in via lsusb (and I noticed that the 305 is a different driver in this instance, #29 rather than #28)

However the indi-allsky service just won't run. Status command gives a different report this time:

 Loaded: loaded (/home/pi/.config/systemd/user/indi-allsky.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Tue 2022-08-16 15:58:29 AEST; 11s ago
    Process: 783 ExecStart=/home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --config /etc/indi-allsky/config.json --log syslog run (code=exited, status=1/FAILURE)
   Main PID: 783 (code=exited, status=1/FAILURE)
        CPU: 4.560s

Aug 16 15:58:29 usb python3[783]: [2022-08-16 15:58:29,175] ERROR in allsky: '  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 584, in run\n    self._initialize()\n'
Aug 16 15:58:29 usb python3[783]: [2022-08-16 15:58:29,176] ERROR in allsky: '  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 330, in _initialize\n    self.indiclient.findCcd()\n'
Aug 16 15:58:29 usb python3[783]: [ERROR] MainProcess allsky.unhandled_exception() #43: '  File "/home/pi/indi-allsky/allsky.py", line 128, in <module>\n    action_func(*args_list, **kwarg>
Aug 16 15:58:29 usb python3[783]: [2022-08-16 15:58:29,177] ERROR in allsky: '  File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 421, in findCcd\n    self._ccd_device = ccd_lis>
Aug 16 15:58:29 usb python3[783]: [ERROR] MainProcess allsky.unhandled_exception() #43: '  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 584, in run\n    self._initialize()\n'
Aug 16 15:58:29 usb python3[783]: [ERROR] MainProcess allsky.unhandled_exception() #43: '  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 330, in _initialize\n    self.indiclient.>
Aug 16 15:58:29 usb python3[783]: [ERROR] MainProcess allsky.unhandled_exception() #43: '  File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 421, in findCcd\n    self._ccd_devic>
Aug 16 15:58:29 usb systemd[514]: indi-allsky.service: Main process exited, code=exited, status=1/FAILURE
Aug 16 15:58:29 usb systemd[514]: indi-allsky.service: Failed with result 'exit-code'.
Aug 16 15:58:29 usb systemd[514]: indi-allsky.service: Consumed 4.560s CPU time.

aaronwmorris, 2022-08-16:
I just merged a change #295 to build a newer version of indi.  This might help with the issues with the SV305Pro

laterrible, 2022-08-16:
> I just merged a change #295 last night to build a newer version of indi. This might help with the issues with the SV305Pro

Great, thanks. I'll recompile/build indi, and setup again, see if I can get a win. I wish I understood a fraction more so that I could help!

closed 2022-08-16 by laterrible