aaronwmorris/indi-allsky #298: ASI224MC and Chrome give 1e-05 in config

glitchies, 2022-08-16:
When ever I save my config with 0.0 in the Min + Default exposures, they come back as "1e-05" and I have to set them back to 0.0 every time I do anything in the config file, otherwise it doesn't capture anything, even though it says everything is running.

Running:
ASI224MC
Pi 4b with 8GB
Rasberry PI OS - 32bit
everything updated and installed as of writing this.

aaronwmorris, 2022-08-16:
What OS are your running Chrome?  I just want to try to use the same.

glitchies, 2022-08-17:
Ummm, I think this is a "me" issue, not the software.  I just tried it on a different computer and don't have an issue.  It's probably a caching issue with that computer.

Just fyi though, it is Win10.

aaronwmorris, 2022-08-17:
I will run some tests.  It is no problem.

aaronwmorris, 2022-08-18:
I tested Chrome on Windows and did not see any issues.  I also tested Chrome and Chromium on Linux with no problems.

I have seen issues with python returning scientific notation for float values previously, but usually only after you get to 5 or 6 decimal places.

Let me know if you have the issue again.

glitchies, 2022-08-21:
ah, yeah...  the starting min exposure for the asi224mc is 0.0000321 I think...  It is weird that it only happens on my main computer's chrome.  haven't seen it be affected by any other computer/web browser.  I'll just close this for now, can be re-opened if something else pops up.

closed 2022-08-21 by glitchies