aaronwmorris/indi-allsky #300: Support for PlayerOne "Neptune"?

objective-d, 2022-08-19:
Hi - when installing I selected "playerOne" from the CCD list (option 22 if I remember correctly) - however I am getting the following error when attempting to start the service: 

`2022-08-19 11:32:06,742 [INFO] MainProcess indi.serverConnected() #233: Server connected (localhost:7624)
[2022-08-19 11:32:06,742] INFO in indi: Server connected (localhost:7624)
2022-08-19 11:32:06,748 [INFO] MainProcess indi.newDevice() #149: new device Telescope Simulator
[2022-08-19 11:32:06,748] INFO in indi: new device Telescope Simulator
2022-08-19 11:32:14,752 [INFO] MainProcess indi._findCcds() #404: Found device Telescope Simulator
[2022-08-19 11:32:14,752] INFO in indi: Found device Telescope Simulator
2022-08-19 11:32:14,754 [INFO] MainProcess indi._findCcds() #409:  Detected telescope
[2022-08-19 11:32:14,754] INFO in indi:  Detected telescope
2022-08-19 11:32:14,755 [INFO] MainProcess indi._findCcds() #409:  Detected guider
[2022-08-19 11:32:14,755] INFO in indi:  Detected guider
2022-08-19 11:32:14,756 [INFO] MainProcess indi.findCcd() #419: Found 0 CCDs
[2022-08-19 11:32:14,756] INFO in indi: Found 0 CCDs
2022-08-19 11:32:14,758 [ERROR] MainProcess allsky.unhandled_exception() #36: An uncaught exception occurred:
[2022-08-19 11:32:14,758] ERROR in allsky: An uncaught exception occurred:
2022-08-19 11:32:14,760 [ERROR] MainProcess allsky.unhandled_exception() #37: Type: <class 'IndexError'>
[2022-08-19 11:32:14,760] ERROR in allsky: Type: <class 'IndexError'>
2022-08-19 11:32:14,761 [ERROR] MainProcess allsky.unhandled_exception() #38: Value: list index out of range
[2022-08-19 11:32:14,761] ERROR in allsky: Value: list index out of range
2022-08-19 11:32:14,769 [ERROR] MainProcess allsky.unhandled_exception() #43: '  File "/home/pi/indi-allsky/./allsky.py", line 128, in <module>\n    action_func(*args_list, **kwargs_dict)\n'
[2022-08-19 11:32:14,769] ERROR in allsky: '  File "/home/pi/indi-allsky/./allsky.py", line 128, in <module>\n    action_func(*args_list, **kwargs_dict)\n'
2022-08-19 11:32:14,771 [ERROR] MainProcess allsky.unhandled_exception() #43: '  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 584, in run\n    self._initialize()\n'
[2022-08-19 11:32:14,771] ERROR in allsky: '  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 584, in run\n    self._initialize()\n'
2022-08-19 11:32:14,772 [ERROR] MainProcess allsky.unhandled_exception() #43: '  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 330, in _initialize\n    self.indiclient.findCcd()\n'
[2022-08-19 11:32:14,772] ERROR in allsky: '  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 330, in _initialize\n    self.indiclient.findCcd()\n'
2022-08-19 11:32:14,773 [ERROR] MainProcess allsky.unhandled_exception() #43: '  File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 421, in findCcd\n    self._ccd_device = ccd_list[0]\n'
[2022-08-19 11:32:14,773] ERROR in allsky: '  File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 421, in findCcd\n    self._ccd_device = ccd_list[0]\n`

I see "Found 0 CCDs", and so assuming the IndexError is just because there's no camera. Can you confirm is my camera supported?

Cheers.

aaronwmorris, 2022-08-19:
This is a bug in indi.  Upgrade to the latest indi (1.9.7) and it should be fixed.

objective-d, 2022-08-19:
Hmm. things do seem to be almost working now - I see it detected my camera correctly and I now see: 

```
2022-08-19 14:00:52,738 [INFO] MainProcess allsky._queueManualTasks() #1430: Checking for manually submitted tasks
[2022-08-19 14:00:52,738] INFO in allsky: Checking for manually submitted tasks
2022-08-19 14:01:02,774 [INFO] MainProcess allsky.run() #602: Camera last ready: 113.0s
[2022-08-19 14:01:02,774] INFO in allsky: Camera last ready: 113.0s
2022-08-19 14:01:02,775 [INFO] MainProcess allsky.run() #603: Exposure state: BUSY
[2022-08-19 14:01:02,775] INFO in allsky: Exposure state: BUSY
2022-08-19 14:01:02,777 [INFO] MainProcess allsky.detectNight() #844: Sun altitude: -38:30:39.6
[2022-08-19 14:01:02,777] INFO in allsky: Sun altitude: -38:30:39.6
2022-08-19 14:01:02,779 [INFO] MainProcess allsky.detectMoonMode() #861: Moon altitide: -5:05:29.8, phase 50.2%
[2022-08-19 14:01:02,779] INFO in allsky: Moon altitide: -5:05:29.8, phase 50.2%
2022-08-19 14:01:02,781 [INFO] MainProcess indi.getCcdTemperature() #456: Sensor temperature: -300.0
[2022-08-19 14:01:02,781] INFO in indi: Sensor temperature: -300.0
```
however I do not see any images, and the web interface is showing indi-allsky is not running: 

[image]

objective-d, 2022-08-19:
I should also add when I watch the web interface, indi-allsky does become active for a short time, then shows failed - and I am seeing no images anywhere

objective-d, 2022-08-19:
lol. and of course I just rebooted and now its all working ;-)

aaronwmorris, 2022-08-19:
I am glad you got it working.  I was mobile so I could not attach the link, but I suspect this was the bug that affected you.

https://github.com/indilib/indi-3rdparty/issues/591

It also affected my PlayerOne Mar C camera.

aaronwmorris, 2022-08-21:
Closing this issue for now.  Let me know if there is anything I can do for you.

closed 2022-08-21 by aaronwmorris