aaronwmorris/indi-allsky #301: install on rpi3

pierrebilb, 2022-08-19:
Hi,
i tried to install on a rpi3 b with raspbian 11 and i get the following error when running setup.sh:
ERROR: Could not find a version that satisfies the requirement rawpy (from versions: none)
ERROR: No matching distribution found for rawpy
Any clue?
Thanks for your help

aaronwmorris, 2022-08-19:
Are you running the 32 or 64 bit version of Raspbian 11?

pierrebilb, 2022-08-19:
32 bits; should i run 64 bit?

aaronwmorris, 2022-08-19:
Maybe.  The rawpy module requirement is new and I had the same error on Raspian 10 (32bit).  I am starting to suspect that there might be some bug in the rawpy metadata or the module will only work on 64 bit platforms.

aaronwmorris, 2022-08-19:
Merged #303 to skip rawpy for 32bit arm.  This means libcamera will not work on this platform.

pierrebilb, 2022-08-19:
Ok, so i will try with the new merge..
Thanks!

pierrebilb, 2022-08-21:
Install was hanging forever with 32 OS; went with 64 bit OS and install went straightforward; allsky indi is working perfectly!
Thanks

closed 2022-08-21 by pierrebilb