aaronwmorris/indi-allsky #308: What Indi installs is required before your setup. A question

Dentonknifeworks, 2022-08-20:
Every time I try to install your software Indi pre-installs is a moving target.
What is required first?

Thanks
Brad

aaronwmorris, 2022-08-20:
With the latest release of indi-allsky, you can use Raspbian 11 64bit with the libcamera integration for the Rpi HQ camera (libcamera_imx477).

You will have to build indi with the `./misc/build_indi.sh` before you run the setup.sh script.

Dentonknifeworks, 2022-08-20:
[image]
Hangs up here

aaronwmorris, 2022-08-20:
It is probably not hung.  Compiling indi takes a while.

Dentonknifeworks, 2022-08-20:
Well that took all most 2 hours but It completed with no errors.

Dentonknifeworks, 2022-08-20:
This worked took a long time

closed 2022-08-20 by Dentonknifeworks