aaronwmorris/indi-allsky #310: QHY setup does not work

maphilli14, 2022-08-20:
I had an existing ASI290 working and swapped for a QHY.  I reran setup from the original installation and it didn't switch to the new camera.  I did a mv of the indi-allsky directory to indi-allsky.bak and did a fresh git, same result.

Now the indi-allsky service fails and I see these errors:

```

Aug 20 18:25:11 allsky2 [INFO] MainProcess indi.getCcdTemperature() #345: Sensor temperature: 37.7
Aug 20 18:25:21 allsky2 [INFO] MainProcess allsky.run() #571: Camera last ready: 3850.0s
Aug 20 18:25:21 allsky2 [INFO] MainProcess allsky.run() #572: Exposure state: ALERT
Aug 20 18:25:21 allsky2 [INFO] MainProcess allsky.detectNight() #800: Sun altitude: 17:10:11.8
Aug 20 18:25:21 allsky2 [INFO] MainProcess allsky.detectMoonMode() #817: Moon altitide: -24:51:01.9, phase 33.6%
Aug 20 18:25:21 allsky2 [INFO] MainProcess indi.getCcdTemperature() #345: Sensor temperature: 37.7
Aug 20 18:25:31 allsky2 [INFO] MainProcess allsky.run() #571: Camera last ready: 3860.0s
Aug 20 18:25:31 allsky2 [INFO] MainProcess allsky.run() #572: Exposure state: ALERT
Aug 20 18:25:31 allsky2 [INFO] MainProcess allsky.detectNight() #800: Sun altitude: 17:08:09.1
Aug 20 18:25:31 allsky2 [INFO] MainProcess allsky.detectMoonMode() #817: Moon altitide: -24:51:56.4, phase 33.6%
Aug 20 18:25:31 allsky2 [INFO] MainProcess indi.getCcdTemperature() #345: Sensor temperature: 37.7
Aug 20 18:25:41 allsky2 [INFO] MainProcess allsky.run() #571: Camera last ready: 3870.1s
Aug 20 18:25:41 allsky2 [INFO] MainProcess allsky.run() #572: Exposure state: ALERT
Aug 20 18:25:41 allsky2 [INFO] MainProcess allsky.detectNight() #800: Sun altitude: 17:06:06.3
Aug 20 18:25:41 allsky2 [INFO] MainProcess allsky.detectMoonMode() #817: Moon altitide: -24:52:50.9, phase 33.6%
Aug 20 18:25:41 allsky2 [INFO] MainProcess indi.getCcdTemperature() #345: Sensor temperature: 37.7
Aug 20 18:25:51 allsky2 [INFO] MainProcess allsky.run() #571: Camera last ready: 3880.1s
Aug 20 18:25:51 allsky2 [INFO] MainProcess allsky.run() #572: Exposure state: ALERT
Aug 20 18:25:51 allsky2 [INFO] MainProcess allsky.detectNight() #800: Sun altitude: 17:04:03.5
Aug 20 18:25:51 allsky2 [INFO] MainProcess allsky.detectMoonMode() #817: Moon altitide: -24:53:45.2, phase 33.6%
Aug 20 18:25:51 allsky2 [INFO] MainProcess indi.getCcdTemperature() #345: Sensor temperature: 37.7
Aug 20 18:26:01 allsky2 [INFO] MainProcess allsky.run() #571: Camera last ready: 3890.1s
Aug 20 18:26:01 allsky2 [INFO] MainProcess allsky.run() #572: Exposure state: ALERT
Aug 20 18:26:01 allsky2 [INFO] MainProcess allsky.detectNight() #800: Sun altitude: 17:02:00.8
Aug 20 18:26:01 allsky2 [INFO] MainProcess allsky.detectMoonMode() #817: Moon altitide: -24:54:39.5, phase 33.6%
Aug 20 18:26:01 allsky2 [INFO] MainProcess indi.getCcdTemperature() #345: Sensor temperature: 37.7
Aug 20 18:26:11 allsky2 [INFO] MainProcess allsky.run() #571: Camera last ready: 3900.2s
Aug 20 18:26:11 allsky2 [INFO] MainProcess allsky.run() #572: Exposure state: ALERT
Aug 20 18:26:11 allsky2 [INFO] MainProcess allsky.detectNight() #800: Sun altitude: 16:59:58.0
Aug 20 18:26:11 allsky2 [INFO] MainProcess allsky.detectMoonMode() #817: Moon altitide: -24:55:33.8, phase 33.5%
Aug 20 18:26:11 allsky2 [INFO] MainProcess indi.getCcdTemperature() #345: Sensor temperature: 37.7
Aug 20 18:26:21 allsky2 [INFO] MainProcess allsky.run() #571: Camera last ready: 3910.2s
Aug 20 18:26:21 allsky2 [INFO] MainProcess allsky.run() #572: Exposure state: ALERT
Aug 20 18:26:21 allsky2 [INFO] MainProcess allsky.detectNight() #800: Sun altitude: 16:57:55.3
Aug 20 18:26:21 allsky2 [INFO] MainProcess allsky.detectMoonMode() #817: Moon altitide: -24:56:27.9, phase 33.5%
Aug 20 18:26:21 allsky2 [INFO] MainProcess indi.getCcdTemperature() #345: Sensor temperature: 37.7
Aug 20 18:26:31 allsky2 [INFO] MainProcess allsky.run() #571: Camera last ready: 3920.2s
Aug 20 18:26:31 allsky2 [INFO] MainProcess allsky.run() #572: Exposure state: ALERT
Aug 20 18:26:31 allsky2 [INFO] MainProcess allsky.detectNight() #800: Sun altitude: 16:55:52.5
Aug 20 18:26:31 allsky2 [INFO] MainProcess allsky.detectMoonMode() #817: Moon altitide: -24:57:22.1, phase 33.5%
Aug 20 18:26:31 allsky2 [INFO] MainProcess indi.getCcdTemperature() #345: Sensor temperature: 37.7
Aug 20 18:28:27 allsky2 [INFO] MainProcess indi.__init__() #90: creating an instance of IndiClient
Aug 20 18:28:27 allsky2 [INFO] MainProcess allsky._initialize() #291: Connecting to indiserver
Aug 20 18:28:27 allsky2 [INFO] MainProcess indi.serverConnected() #172: Server connected (localhost:7624)
Aug 20 18:28:27 allsky2 [INFO] MainProcess indi.newDevice() #119: new device Telescope Simulator
Aug 20 18:28:35 allsky2 [INFO] MainProcess indi.findCcds() #307: Found device Telescope Simulator
Aug 20 18:28:35 allsky2 [INFO] MainProcess indi.findCcds() #312:  Detected telescope
Aug 20 18:28:35 allsky2 [INFO] MainProcess indi.findCcds() #312:  Detected guider
Aug 20 18:28:35 allsky2 [ERROR] MainProcess allsky._initialize() #304: No CCDs detected
Aug 20 18:31:00 allsky2 [INFO] MainProcess indi.__init__() #90: creating an instance of IndiClient
Aug 20 18:31:00 allsky2 [INFO] MainProcess allsky._initialize() #291: Connecting to indiserver
Aug 20 18:31:00 allsky2 [INFO] MainProcess indi.serverConnected() #172: Server connected (localhost:7624)
Aug 20 18:31:00 allsky2 [INFO] MainProcess indi.newDevice() #119: new device Telescope Simulator
Aug 20 18:31:08 allsky2 [INFO] MainProcess indi.findCcds() #307: Found device Telescope Simulator
Aug 20 18:31:08 allsky2 [INFO] MainProcess indi.findCcds() #312:  Detected telescope
Aug 20 18:31:08 allsky2 [INFO] MainProcess indi.findCcds() #312:  Detected guider
Aug 20 18:31:08 allsky2 [ERROR] MainProcess allsky._initialize() #304: No CCDs detected
Aug 20 19:12:07 allsky2 [ERROR] MainProcess allsky._parseConfig() #259: indi-allsky version does not match config, please rerun setup.sh
Aug 20 19:17:23 allsky2 [ERROR] MainProcess allsky._parseConfig() #259: indi-allsky version does not match config, please rerun setup.sh
maphilli14@allsky2:~ $ /home/maphilli14/indi-allsky/setup.sh
```

I tried this

`
 sudo mv /etc/indi-allsky/config.json /etc/indi-allsky/config.json.old`

Setup failures include DB migration as well as flask

```

Site default-ssl already disabled
Site indi-allsky already enabled
Synchronizing state of apache2.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable apache2
**** Setup HTDOCS folder ****
**** Setup image folder ****
**** Setup DB ****
**** Backup DB prior to migration ****
Usage: flask db revision [OPTIONS]
Try 'flask db revision --help' for help.

Error: Could not locate a Flask application. Use the 'flask --app' option, 'FLASK_APP' environment variable, or a 'wsgi.py' or 'app.py' file in the current directory.

The script exited abnormally, please try to run again...

```

aaronwmorris, 2022-08-20:
You may want to move the DB migrations from the original git folder to the new one...  everything in the `migrations/versions/` folder.

If that does not fix it, there are some commands that may help here:  https://github.com/aaronwmorris/indi-allsky/wiki/Database-migrations#fixing-broken-migrations

As for the camera issue:  I would have to see closer to when the error happened in the log.  Did you build indi or does your system use the indi apt repository?

maphilli14, 2022-08-20:
TYVM, I just realized it's trying to find a DB in the old dir and tried this....

```

mv: cannot move '../indi-allsky.bak/migrations' to './migrations': Directory not empty
mv: cannot move '../indi-allsky.bak/misc' to './misc': Directory not empty
mv: cannot move '../indi-allsky.bak/__pycache__' to './__pycache__': Directory not empty
mv: cannot move '../indi-allsky.bak/service' to './service': Directory not empty
mv: cannot move '../indi-allsky.bak/testing' to './testing': Directory not empty
mv: cannot move '../indi-allsky.bak/virtualenv' to './virtualenv': Directory not empty
maphilli14@allsky2:~/indi-allsky $ rsync -a ../indi-allsky.bak/* .
maphilli14@allsky2:~/indi-allsky $ /home/maphilli14/indi-allsky/setup.sh
###############################################
### Welcome to the indi-allsky setup script ###
###############################################

```

maphilli14, 2022-08-21:
I think I got a solid migration and did try some of the migration suggestions but it seems to still fail:

```

~
maphilli14@allsky2:~/indi-allsky $ systemctl --user status indi-allsky
* indi-allsky.service - Indi AllSky Service
     Loaded: loaded (/home/maphilli14/.config/systemd/user/indi-allsky.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2022-08-20 20:03:51 EDT; 11s ago
    Process: 2910 ExecStart=/home/maphilli14/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --config /etc/indi-allsky/config.json --log syslog run (code=exited, status=1/FAILURE)
   Main PID: 2910 (code=exited, status=1/FAILURE)
        CPU: 5.176s

Aug 20 20:03:49 allsky2 python3[2910]: [INFO] MainProcess indi.findCcds() #312:  Detected telescope
Aug 20 20:03:49 allsky2 python3[2910]: [2022-08-20 20:03:49,909] INFO in indi: Found device Telescope Simulator
Aug 20 20:03:49 allsky2 python3[2910]: [2022-08-20 20:03:49,910] INFO in indi:  Detected telescope
Aug 20 20:03:49 allsky2 python3[2910]: [2022-08-20 20:03:49,911] INFO in indi:  Detected guider
Aug 20 20:03:49 allsky2 python3[2910]: [2022-08-20 20:03:49,911] ERROR in allsky: No CCDs detected
Aug 20 20:03:49 allsky2 python3[2910]: [INFO] MainProcess indi.findCcds() #312:  Detected guider
Aug 20 20:03:49 allsky2 python3[2910]: [ERROR] MainProcess allsky._initialize() #304: No CCDs detected
Aug 20 20:03:51 allsky2 systemd[568]: indi-allsky.service: Main process exited, code=exited, status=1/FAILURE
Aug 20 20:03:51 allsky2 systemd[568]: indi-allsky.service: Failed with result 'exit-code'.
Aug 20 20:03:51 allsky2 systemd[568]: indi-allsky.service: Consumed 5.176s CPU time.
```

maphilli14, 2022-08-21:
> As for the camera issue: I would have to see closer to when the error happened in the log. Did you build indi or does your system use the indi apt repository?

how do I tell you the answer to this?  I did a fresh buster install and then a git clone of your repo, that was it.

TIA

aaronwmorris, 2022-08-21:
If you built indi using the build_indi.sh script, the QHY firmwares are probably located in the wrong folder.  I discovered this with my QHY camera a while back.  Let me write up some commands for you to fix this.

aaronwmorris, 2022-08-21:
Try this and restart indi-allsky.

```
sudo mkdir /lib/firmware/qhy
sudo ln -s /lib/firmware/QHY*.HEX /lib/firmware/qhy
```

maphilli14, 2022-08-21:
```

Enjoy!
maphilli14@allsky2:~/indi-allsky $ ls /lib/firmware/QHY*.HEX /lib/firmware/qhy
/lib/firmware/QHY09000A.HEX    /lib/firmware/QHY21.HEX    /lib/firmware/QHY5LOADER.HEX
/lib/firmware/QHY10.HEX        /lib/firmware/QHY22.HEX    /lib/firmware/QHY695A.HEX
/lib/firmware/QHY11.HEX        /lib/firmware/QHY23.HEX    /lib/firmware/QHY6.HEX
/lib/firmware/QHY12.HEX        /lib/firmware/QHY27.HEX    /lib/firmware/QHY7.HEX
/lib/firmware/QHY15.HEX        /lib/firmware/QHY28.HEX    /lib/firmware/QHY814A.HEX
/lib/firmware/QHY160002AD.HEX  /lib/firmware/QHY29.HEX    /lib/firmware/QHY8.HEX
/lib/firmware/QHY16000.HEX     /lib/firmware/QHY2E.HEX    /lib/firmware/QHY8L.HEX
/lib/firmware/QHY16200A.HEX    /lib/firmware/QHY2.HEX     /lib/firmware/QHY8M.HEX
/lib/firmware/QHY16803A.HEX    /lib/firmware/QHY2PRO.HEX  /lib/firmware/QHY8PRO.HEX
/lib/firmware/QHY16.HEX        /lib/firmware/QHY5.HEX     /lib/firmware/QHY90A.HEX
/lib/firmware/QHY20.HEX        /lib/firmware/QHY5II.HEX   /lib/firmware/QHY9S.HEX

/lib/firmware/qhy:
QHY09000A.HEX  QHY160002AD.HEX  QHY20.HEX  QHY28.HEX    QHY5.HEX        QHY7.HEX     QHY8PRO.HEX
QHY10.HEX      QHY16000.HEX     QHY21.HEX  QHY29.HEX    QHY5II.HEX      QHY814A.HEX  QHY90A.HEX
QHY11.HEX      QHY16200A.HEX    QHY22.HEX  QHY2E.HEX    QHY5LOADER.HEX  QHY8.HEX     QHY9S.HEX
QHY12.HEX      QHY16803A.HEX    QHY23.HEX  QHY2.HEX     QHY695A.HEX     QHY8L.HEX
QHY15.HEX      QHY16.HEX        QHY27.HEX  QHY2PRO.HEX  QHY6.HEX        QHY8M.HEX
maphilli14@allsky2:~/indi-allsky $ watch tail -30 /var/log/indi-allsky/indi-allsky.log
maphilli14@allsky2:~/indi-allsky $ tail -10 /var/log/indi-allsky/indi-allsky.log
Aug 20 21:39:35 allsky2 [INFO] MainProcess indi.findCcds() #312:  Detected guider
Aug 20 21:39:35 allsky2 [ERROR] MainProcess allsky._initialize() #304: No CCDs detected
Aug 20 21:41:09 allsky2 [INFO] MainProcess indi.__init__() #90: creating an instance of IndiClient
Aug 20 21:41:09 allsky2 [INFO] MainProcess allsky._initialize() #291: Connecting to indiserver
Aug 20 21:41:09 allsky2 [INFO] MainProcess indi.serverConnected() #172: Server connected (localhost:7624)
Aug 20 21:41:09 allsky2 [INFO] MainProcess indi.newDevice() #119: new device Telescope Simulator
Aug 20 21:41:17 allsky2 [INFO] MainProcess indi.findCcds() #307: Found device Telescope Simulator
Aug 20 21:41:17 allsky2 [INFO] MainProcess indi.findCcds() #312:  Detected telescope
Aug 20 21:41:17 allsky2 [INFO] MainProcess indi.findCcds() #312:  Detected guider
Aug 20 21:41:17 allsky2 [ERROR] MainProcess allsky._initialize() #304: No CCDs detected
maphilli14@allsky2:~/indi-allsky $
```

aaronwmorris, 2022-08-21:
Sorry, fully reboot your system and see if that helps.

maphilli14, 2022-08-21:
I think a reboot was needed
```

maphilli14@allsky2:~/indi-allsky $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 1618:0920
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
maphilli14@allsky2:~/indi-allsky $ sudo halt
Connection to 192.168.45.49 closed by remote host.
Connection to 192.168.45.49 closed.
maphilli14@Virgo:~$ ssh maphilli14@192.168.45.49
Linux allsky2 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l
maphilli14@allsky2:~ $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 1618:0921 QHY-CCD   QHY5-II
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
maphilli14@allsky2:~ $
```

closed 2022-08-21 by maphilli14