aaronwmorris/indi-allsky #317: URL for latest image

pterodyne, 2022-08-22:
I know there is a public site that doesn't need authentication, but what I really need is the latest image available.  I can't seem to figure out how to get this URL.  TJ version this is easy, but I'm really liking this project with all the other features available.  I embed the latest image into my IP Cam app on my phone, as well as my observatory's ACP webpage.  But having the whole interface is unnecessary.  

From a feature perspective having this have be a jpeg push or mjpeg or even RTSP stream would be ideal, but I can do a browser based refresh a variety of ways, so really just the URL access would be great.

Thanks for this project! I really feel like it is going the direction my needs are vs the TJ project.  I discussed indi support long ago on that project but that wasn't something anyone seemed interested in. (Which makes no sense to me, writing all this specific code for various cameras.)

Thanks for any suggestions.

pterodyne, 2022-08-22:
Ugh.. Ive been looking for days, then I  get an idea and find it right after I post this.  Sorry

anyway, for others looking: baseurl/indi-allsky/images/latest.jpg

So I guess really this is more of a feature request for an official auto push mjpeg url for posting on other systems.

aaronwmorris, 2022-08-22:
For the mjpeg url, I am not exactly familiar with the concept, do you mean something like this:  https://tree.rocks/a-simple-way-for-motion-jpeg-in-flask-806b8bfefa96

pterodyne, 2022-08-22:
Yes, essentially that, but in reality it's overkill for MJPG or any kind of true video stream since it's frames per hour instead of per second :)

Normally what Ive done is take the url of the latest jpeg and do this:

baseurl/indi-allsky/images/latest.jpg?t=randomdigits

In my img tag that Im embedding.  For some reason this isn't working.  But the idea is to make it always refresh when the image changes, but essentially still just be a jpeg url so that ip camera software/apps can just retrieve the jpeg.  So I'll just close it out.  Sorry to bother you.  I do have a different feature idea, but Ill post a new item.  Is it ok to discuss feature ideas using "issues" or is there a preferred way?

Bryan.

closed 2022-08-22 by pterodyne

aaronwmorris, 2022-08-22:
It's no big deal, opening an issue is my preferred way of handling this.

I'm surprised the ?t=randomdigits is not working.  That is pretty much exactly how the image is loaded in the index view.

https://github.com/aaronwmorris/indi-allsky/blob/master/indi_allsky/flask/templates/index.html#L57

aaronwmorris, 2022-08-22:
Feel free to re-open this issue or open another one if you want to discuss.