aaronwmorris/indi-allsky #326: Get only a gray image

topdancer, 2022-08-26:
Hi Aaron,

I installed indi-allsky on bullseye 64-bit per your post in indilib forum (see also my post in your thread) and connected the raspy hq camera with libcamera_IMX477. Both services are running, but I get no image. I checked with raspivid (on a different installation) and also focused the lens with this tool, so this part works. I also did an apt upgrade today.
I then checked with vcgencmd get_camera, this results in
supported=1, detected=1, lib camera interfaces=0

I guess this is the reason, so how do I get libcamera working? Is there anything missing?

Thanks
Kurt

aaronwmorris, 2022-08-26:
Did you reboot after running the setup.sh script?  Enabling the camera interface requires a reboot.

StarGeezerPhil, 2022-08-26:
Apply a minor change to config, apply, revert change and re-apply seems to fix this issue.E.g. when this happens I usually enable auto white balance, apply the config, disable it and apply the config again.Not sure what causes it though.Get Outlook for Android

topdancer, 2022-08-26:
@ Aaron: yes I rebooted several times, also now, same result.

@ StarGeezerPgil: I will try this tomorrow, its now fairly late here. Will report.

aaronwmorris, 2022-08-26:
Are you running a Raspi3 or 4?  I have heard some people have to manually load the dtoverlay for their camera to be detected.

See the Quick Start Guide here:  https://www.arducam.com/docs/cameras-for-raspberry-pi/native-raspberry-pi-cameras/12mp-imx477-camera/

aaronwmorris, 2022-08-27:
I merged #328 to log libcamera errors if they occur.

topdancer, 2022-08-27:
I also found the information on stackoverflow. In addition it seems necessary to disable the legacy camera support in raspy-config. With both changes I got it working on a raspy 4. I still need to find an optimal exposure for day time images, so far they were fairly faint.

aaronwmorris, 2022-10-22:
Closing this for now.  Let me know if there is anything else I can do.

closed 2022-10-22 by aaronwmorris